ifm/ifm3d #112: O3d303 make check error:C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.

hjw1007, 2018-11-14:


hjw1007, 2018-11-14:
I firstly  make check under firmware version (1.6.2114) ,then try update firmware version (1.6.2114) to 1.23.1522 and that problem was same

inbangsa, 2018-11-14:
@hjw1007  This is due to the fact you are not able to connect to camera.  have a [look at troubleshoot guide](https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md#connecting-the-camera). to solve the issue.

hjw1007, 2018-11-14:
@inbangsa  thanks you answer , I make check , get the error:  [  FAILED  ] Image.ComputeCartesian
,I want to know how to deal with it
 make check
[ 16%] Built target ifm3d_camera_shared
[ 22%] Built target ifm3d_pcicclient_shared
[ 27%] Built target gtest
[ 33%] Built target gtest_main
[ 41%] Built target ifm3d-pcicclient-tests
Scanning dependencies of target check_pcicclient
[==========] Running 2 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 2 tests from PCICClientTest
[ RUN      ] PCICClientTest.IncomingResponseMessage
[       OK ] PCICClientTest.IncomingResponseMessage (390 ms)
[ RUN      ] PCICClientTest.InvalidCommandLength
[       OK ] PCICClientTest.InvalidCommandLength (427 ms)
[----------] 2 tests from PCICClientTest (818 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test case ran. (818 ms total)
[  PASSED  ] 2 tests.
[ 41%] Built target check_pcicclient
[ 55%] Built target ifm3d-camera-tests
Scanning dependencies of target check_camera
[==========] Running 19 tests from 3 test cases.
[----------] Global test environment set-up.
[----------] 2 tests from Err
[ RUN      ] Err.LibraryError
[       OK ] Err.LibraryError (0 ms)
[ RUN      ] Err.SystemError
[       OK ] Err.SystemError (0 ms)
[----------] 2 tests from Err (0 ms total)

[----------] 1 test from Version
[ RUN      ] Version.Version
[       OK ] Version.Version (0 ms)
[----------] 1 test from Version (0 ms total)

[----------] 16 tests from CameraTest
[ RUN      ] CameraTest.FactoryDefaults
[       OK ] CameraTest.FactoryDefaults (7132 ms)
[ RUN      ] CameraTest.DefaultCredentials
[       OK ] CameraTest.DefaultCredentials (18 ms)
[ RUN      ] CameraTest.ApplicationList
[       OK ] CameraTest.ApplicationList (57 ms)
[ RUN      ] CameraTest.SessionManagement
[       OK ] CameraTest.SessionManagement (265 ms)
[ RUN      ] CameraTest.CopyDeleteApplication
[       OK ] CameraTest.CopyDeleteApplication (1099 ms)
[ RUN      ] CameraTest.CopyDeleteExceptions
[       OK ] CameraTest.CopyDeleteExceptions (700 ms)
[ RUN      ] CameraTest.CreateDeleteApplication
[       OK ] CameraTest.CreateDeleteApplication (1301 ms)
[ RUN      ] CameraTest.CreateApplicationException
[       OK ] CameraTest.CreateApplicationException (364 ms)
[ RUN      ] CameraTest.ImportExportApplication
[       OK ] CameraTest.ImportExportApplication (1439 ms)
[ RUN      ] CameraTest.ImportExportConfig
[       OK ] CameraTest.ImportExportConfig (1237 ms)
[ RUN      ] CameraTest.ActiveApplication
[       OK ] CameraTest.ActiveApplication (6566 ms)
[ RUN      ] CameraTest.ImagerTypes
[       OK ] CameraTest.ImagerTypes (32717 ms)
[ RUN      ] CameraTest.Filters
[       OK ] CameraTest.Filters (9308 ms)
[ RUN      ] CameraTest.JSON
[       OK ] CameraTest.JSON (5850 ms)
[ RUN      ] CameraTest.Time
[       OK ] CameraTest.Time (967 ms)
[ RUN      ] CameraTest.TemporaryParameters
[... 65 lines cut ...]
Value of: std::equal(x_cam.begin<std::int16_t>(), x_cam.end<std::int16_t>(), x_computed.begin<std::int16_t>(), cmp)
  Actual: false
Expected: true
[  FAILED  ] Image.ComputeCartesian (401 ms)
[ RUN      ] Image.TimeStamp
[       OK ] Image.TimeStamp (1714 ms)
[ RUN      ] Image.IlluTemp
[       OK ] Image.IlluTemp (202 ms)
[----------] 10 tests from Image (3908 ms total)

[----------] Global test environment tear-down
[==========] 10 tests from 1 test case ran. (3909 ms total)
[  PASSED  ] 9 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] Image.ComputeCartesian

 1 FAILED TEST
modules/image/test/CMakeFiles/check_image.dir/build.make:57: recipe for target 'modules/image/test/CMakeFiles/check_image' failed
make[3]: *** [modules/image/test/CMakeFiles/check_image] Error 1
CMakeFiles/Makefile2:743: recipe for target 'modules/image/test/CMakeFiles/check_image.dir/all' failed
make[2]: *** [modules/image/test/CMakeFiles/check_image.dir/all] Error 2
CMakeFiles/Makefile2:78: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
Makefile:184: recipe for target 'check' failed
make: *** [check] Error 2

inbangsa, 2018-11-14:
This is happening due to invalid pixels. we are already working on the pull request for this test failure.  you can find the discussion on this topic [here] (https://github.com/ifm/ifm3d/pull/106)

tpanzarella, 2018-11-14:
For my own curiosity, what version of ifm3d are you running @hjw1007 Any chance you are working with the `dev-0.11.2` branch?

hjw1007, 2018-11-15:
@tpanzarella I run in the firmware version is 1.6.2114 and the sensor is O3D303 and the ifm3d version is 0.11.0 and ubuntu version is 16.04LTS.
  I think @inbangsa said right ,maybe due to invalid pixels ,hope that it can be fixed . I  haven't tried dev-0.11.2 yet ,I can try it.

hjw1007, 2018-11-15:
@tpanzarella  yes, it works with the dev-0.11.2 branch, thanks to all.

closed 2018-11-15 by hjw1007