ifm/ifm3d #118: Why the depth map has the inverted reflection in a concrete ground？

JunLFang, 2018-11-26:


JunLFang, 2018-11-26:
[image]

JunLFang, 2018-11-26:
How to solve the problem or avoid it? Thanks a lot.

graugans, 2018-11-28:
@JunLFang to dig a bit more deeper into this problem we probably need more details. Nevertheless please keep in mind this is a camera with an active illumination. It looks like you are suffering from reflective materials which do act like a mirror.

GarrettPlace, 2018-11-28:
@JunLFang  - thank you for the details provided.  As @graugans mentions, additional detail would be required to best support your questions.  It may also be helpful to understand your primary Use Case as this may help us direct you to the appropriate settings.  Would you like to do this offline or on this thread?

theseankelly, 2019-10-09:
@JunLFang I'm going through and closing old issues -- please feel free to reactivate or reach out to @GarrettPlace directly and take him up on his offer to discuss offline.

closed 2019-10-09 by theseankelly