ifm/ifm3d #119: "Lib: XMLRPC Timeout - can you `ping' the sensor?" with make test even though I can ping the sensor

Vinzz98, 2018-11-26:


tpanzarella, 2018-11-26:
@Vinzz98 Can you run the following command (after you have built the software) from your `build` directory:

```
$ IFM3D_VLOG=20 GTEST_FILTER=CameraTest.FactoryDefaults make check
```

Then, please paste in the output of `/tmp/ifm3d.INFO` into a comment on this issue. It will help us understand the issue better.

Vinzz98, 2018-11-26:
Hello @tpanzarella ,
Thank you for your quick answer. The output when running your command from the ifm3d/build directory is:
CameraTest.FactoryDefaults make check
[ 22%] Built target ifm3d_camera_shared
[ 37%] Built target ifm3d_framegrabber_shared
[ 44%] Built target ifm3d_image_shared
[ 51%] Built target gtest
[ 59%] Built target gtest_main
[ 70%] Built target ifm3d-image-tests
Note: Google Test filter = CameraTest.FactoryDefaults
[==========] Running 0 tests from 0 test cases.
[==========] 0 tests from 0 test cases ran. (0 ms total)
[  PASSED  ] 0 tests.
[ 70%] Built target check_image
[ 88%] Built target ifm3d-camera-tests
Scanning dependencies of target check_camera
Note: Google Test filter = CameraTest.FactoryDefaults
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from CameraTest
[ RUN      ] CameraTest.FactoryDefaults
/home/vinz/catkin_ws/src/ifm3d/modules/camera/test/ifm3d-camera-camera-tests.cpp:28: Failure
Expected: this->cam_->FactoryReset() doesn't throw an exception.
  Actual: it throws.
/home/vinz/catkin_ws/src/ifm3d/modules/camera/test/ifm3d-camera-camera-tests.cpp:30: Failure
Expected: this->cam_->DeviceType() doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] CameraTest.FactoryDefaults (6003 ms)
[----------] 1 test from CameraTest (6003 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (6003 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] CameraTest.FactoryDefaults

 1 FAILED TEST
modules/camera/test/CMakeFiles/check_camera.dir/build.make:57: recipe for target 'modules/camera/test/CMakeFiles/check_camera' failed
make[3]: *** [modules/camera/test/CMakeFiles/check_camera] Error 1
CMakeFiles/Makefile2:330: recipe for target 'modules/camera/test/CMakeFiles/check_camera.dir/all' failed
make[2]: *** [modules/camera/test/CMakeFiles/check_camera.dir/all] Error 2
CMakeFiles/Makefile2:108: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
Makefile:199: recipe for target 'check' failed
make: *** [check] Error 2

tpanzarella, 2018-11-26:
Right. Can you paste in the output of the log file found at: `/tmp/ifm3d.INFO`. Thanks.

Vinzz98, 2018-11-26:
Log file created at: 2018/11/26 15:54:03
Running on machine: Vincent-MacBookPro
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
I1126 15:54:03.564774 11304 camera_impl.hpp:406] Increasing XML-RPC response size limit...
I1126 15:54:03.564962 11304 camera_impl.hpp:410] Initializing Camera: ip=192.168.0.69, xmlrpc_port=80, password=
I1126 15:54:03.564973 11304 camera_impl.hpp:414] XMLRPC URL Prefix=http://192.168.0.69:80
E1126 15:54:03.565232 11304 camera_impl.hpp:259] http://192.168.0.69:80/api/rpc/v1/com.ifm.efector/->getParameter:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP POST transaction, explaining:  Failed to connect to 192.168.0.69 port 80: Connection refused
W1126 15:54:03.565464 11304 camera.cpp:214] Could not probe device type: Lib: XMLRPC Timeout - can you `ping' the sensor?
W1126 15:54:03.565512 11304 camera.cpp:234] Returning instance of base camera class!
E1126 15:54:03.565896 11304 camera_impl.hpp:259] http://192.168.0.69:80/api/rpc/v1/com.ifm.efector/->requestSession:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP POST transaction, explaining:  Failed to connect to 192.168.0.69 port 80: Connection refused
E1126 15:54:03.565959 11304 camera_impl.hpp:193] Lib: XMLRPC Timeout - can you `ping' the sensor?
E1126 15:54:09.566915 11304 camera_impl.hpp:259] http://192.168.0.69:80/api/rpc/v1/com.ifm.efector/->getParameter:Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP POST transaction, explaining:  Failed to connect to 192.168.0.69 port 80: Connection refused
I1126 15:54:09.567260 11304 camera_impl.hpp:419] Dtor...

tpanzarella, 2018-11-26:
> http://192.168.0.69:80/api/rpc/v1/com.ifm.efector/->requestSession:Unable to transport XML to server and get XML response back. libcurl failed to execute the HTTP POST transaction, explaining: Failed to connect to 192.168.0.69 port 80: Connection refused E1126 

@Vinzz98 Do you have a password set on the camera?

@graugans Any insights on what on an O3X100 would cause a "connection refused" to the XML-RPC (HTTP) server?

Vinzz98, 2018-11-26:
> @Vinzz98 Do you have a password set on the camera?
>
> @graugans Any insights on what on an O3X100 would cause a "connection refused" to the XML-RPC (HTTP) server?

@tpanzarella I didn't set one but how could I check?

tpanzarella, 2018-11-26:
To disable the password, you will need the current password. Can you paste in the output of:

```
$ ifm3d ls
```

IIRC, this does not require a session with the camera.

tpanzarella, 2018-11-26:
@Vinzz98 With your camera plugged in, can you paste this URL into your web browser: `http://192.168.0.69/api/rpc/v1/com.ifm.efector/`. What do you get? Maybe you can attache a screen shot from your browser so we can see too?

Also, can you paste in the output of:

```
$ ifconfig
```

and the output of:

```
$ netstat -r
```

Thanks.

Vinzz98, 2018-11-26:
[image]

Vinzz98, 2018-11-26:
The output of $ ifconfig is:
ens9: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.69  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::6db:f994:c2a3:5879  prefixlen 64  scopeid 0x20<link>
        ether 38:c9:86:40:a7:ba  txqueuelen 1000  (Ethernet)
        RX packets 78  bytes 12294 (12.2 KB)
        RX errors 0  dropped 0  overruns 0  frame 1
        TX packets 339  bytes 35979 (35.9 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 19  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 37828  bytes 4287679 (4.2 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 37828  bytes 4287679 (4.2 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp3s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.5.152.142  netmask 255.255.240.0  broadcast 10.5.159.255
        inet6 fe80::d212:37:5204:4bcb  prefixlen 64  scopeid 0x20<link>
        inet6 2001:67c:10ec:574b:8000::155  prefixlen 128  scopeid 0x0<global>
        ether a4:5e:60:d4:09:15  txqueuelen 1000  (Ethernet)
        RX packets 290893  bytes 94232161 (94.2 MB)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 71332  bytes 10938974 (10.9 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
The output of $ netstat -r is:
[image]

tpanzarella, 2018-11-26:
It looks like you have bound the IP address of the camera to your ethernet interface (ens9). `192.168.0.69` is the default IP of the camera. You need to set your address to something different.

Vinzz98, 2018-11-27:
I changed my IP and now everything works fine! Thank you very much for your quick help !

commit a7c92843 (2022-07-18): Merge branch 'bugfix/ods_bugfix' into 'o3r/main-next'

Fix bugs found by CSR team

Closes #119, #120, and #121

See merge request syntron/support/csr/ifm3d/ifm3d!142
closed 2018-11-27 by Vinzz98