ifm/ifm3d #120: g++ 32 bit compile not catching exception?

Postea-CS, 2018-11-26:
Hi i have a simple code as follows
```
#include <iostream>
#include <memory>
#include <ifm3d/camera.h>

int main(int argc, const char **argv)
{
try{ 
	auto cam = ifm3d::Camera::MakeShared("192.168.2.103");
	if (cam)
{
	cam->ToJSON();
}
}catch(...)
{
	std::cout<<"Catch"<<std::endl;
}
std::cout<<"Exiting"<<std::endl;

  return 0;
}
```
my compile command is

g++ -o main -m32 -std=c++11  main.cpp -I./ThirdParty/ifm3d-0.10.0/include/ -L./ThirdParty/ifm3d-0.10.0/lib/linux_x86/gcc-7.3/ -L./ThirdParty/libglog/usr/lib/i386-linux-gnu     -lifm3d_camera -lunwind -lglog -lgflags

This is on a ubuntu bionic 64 bit OS  . When i run it i get

terminate called after throwing an instance of 'ifm3d::error_t'
  what():  Lib: XMLRPC Timeout - can you `ping' the sensor?
Aborted (core dumped)

HOWEVER,
If i run the whole command on a 32 bit bionic OS (without the m32 option of course) , the exe runs fine? 

Any ideas?

tpanzarella, 2018-11-26:
I suspect you are likely linking to two different, ABI-incompatible versions of the dependent libraries. For example, I see that you are explicitly linking to `glog` via a "Third Party" build directory. However, is this the same `glog` that `libifm3d_camera.so` is linked to? 

It is not recommended to specify the compiler command explicitly as you are doing. We use `cmake` and its infrastructure to ensure you are inheriting the proper compiler and linker flags. For example, let's suppose you have created your above C++ program in a file called `ex-foo.cpp`. A minimal `CMakeLists.txt` necessary to build and link that executable would look like:

```
cmake_minimum_required(VERSION 3.5)
project(ex-foo CXX)
set(CMAKE_BUILD_TYPE Release) # Release or Debug

find_package(ifm3d 0.11.2 CONFIG
  REQUIRED COMPONENTS camera
  )

add_executable(ex-foo ex-foo.cpp)
target_link_libraries(ex-foo ifm3d::camera)
```

In this case, the compiler line looks like (on my machine):

```
/usr/bin/c++     -O3 -DNDEBUG   -std=c++14 -o CMakeFiles/ex-foo.dir/ex-foo.cpp.o -c /home/uspanzto/dev/ifm3d-examples/foo/ex-foo.cpp
```

and the link line...

```
/usr/bin/c++   -O3 -DNDEBUG   CMakeFiles/ex-foo.dir/ex-foo.cpp.o  -o ex-foo /usr/lib/libifm3d_camera.so.0.12.0 
```

Maybe you can try this?

tpanzarella, 2018-11-26:
Also, did you compile ifm3d as a 32 bit or 64 bit library? Pasting in the output of `file` would be useful. For example:

```
$ $ file /usr/lib/libifm3d_camera.so.0.12.0 
/usr/lib/libifm3d_camera.so.0.12.0: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=d274bb170e49340df2a1d0f96fdb980e50278613, stripped

```

Postea-CS, 2018-11-26:
Hi Tom,

The reason why this is weird is that, I have 2 VM one 32 bit and on3 64 bit and they are both mount the same folder that contains the dependencies aka the ifm3d_camera n the glog libraries.

The same cpp is copied to both the VM and them compile. 32 bit works but 64 bit generated has that error. Since both uses the same files I would imagine if there are issues with the libraries and dependencies , it would show up on both.

Also main functionality of the library works, its just the exceptions are not caught. I isolate the issues to simplify the issue.

This is a standalone code that I exctracted to demonstrate the problem. And this demonstrate the problem pretty cleanly, I can send you all the libraries that I used if you want.

Ifm3d is built for 32 bit. We don’t have a 64 bit build for that library yet.

Regards,
Chin

tpanzarella, 2018-11-26:
Assuming you have built `ifm3d` and its entire dependency chain as 32-bit, this can work. I.e., you can run your 32-bit program on your 64-bit vm. However, you cannot mix (link) 32-bit apps/libraries with eachother safely. If you have a full 32-bit stack built for ifm3d, this makes me think that potentially it is the runtime linker. I.e., you have one of the libraries (maybe `glog` or something else) that is on your 64-bit vm and at runtime is linked rather than your 32-bit version on your shared partition (??). To that end, can you run `ldd ex-foo` (replace `ex-foo` with your executable program). Look at each one of the libraries you are linking to. Are they all 32-bit libraries? Including the STL. Is your shared partition first in your `LD_LIBRARY_PATH`?

closed 2018-11-26 by Postea-CS