ifm/ifm3d #121: O3D303 Repeatability accuracy

hjw1007, 2018-11-28:
hi ,all
    I test four times in 3m work distance, each line represents the depth data of line 150 of the image,.Why is repeatability accuracy inconsistent at each point?

[150row_data.xlsx](https://github.com/ifm/ifm3d/files/2623509/150row_data.xlsx)

[image]

graugans, 2018-11-28:
@hjw1007 thanks for your contribution, can you please provide a diagram descriptions in the English language. Additionally it would be very nice to have the plotted data in CSV so it is easier to read.

Could you please so kind and provide the following information:

- Recorded amplitude and distance data of the scene you are looking at.
- The firmware version you are running

hjw1007, 2018-11-28:
@graugans Thank you for your quick answer.  I run in the firmware version is 1.6.2114 and the sensor is O3D303 and the ifm3d version is 0.11.1 and ubuntu version is 16.04LTS.
[image]
[image]
[image]

theseankelly, 2020-01-12:
If I'm reading this plot correctly, the jitter observed is within the noise profile of the sensor.

That said, this issue is 1.5 years old so I'm closing it. Don't hesitate to reactivate if further assistance is required.

closed 2020-01-12 by theseankelly