ifm/ifm3d #122: New ifm3d module: Python

tpanzarella, 2018-11-30:
Utilizing [PyBind11](https://github.com/pybind/pybind11) create a new ifm3d module which implements a Python interface to ifm3d. This should expose the full feature set of ifm3d, to include, accessing/mutating camera parameters as well as capturing image data from the live/streaming camera as [numpy](http://www.numpy.org/) arrays (ideally a zero-cost/zero-copy abstraction/projection of the ifm3d [opencv module](https://github.com/ifm/ifm3d/tree/master/modules/opencv) ). This should *not* reinvent the [tools module](https://github.com/ifm/ifm3d/tree/master/modules/tools) i.e., it should not require `tools` to be present on the system.

Let's discuss the "requirements" for this feature on this issue. Feedback and input from the community is welcome. @theseankelly will be the lead developer on this topic.

pfrydlewicz, 2018-12-11:
That's genious, thumbs up.
Good move to use PyBind11 instead of boost

G-Kvist, 2019-01-02:
Awesome! Will the PCL viewer be included as well? Otherwise a point cloud visualizer is welcomed.

G-Kvist, 2019-01-06:
@theseankelly just a thought - https://github.com/IntelVCL/Open3D contains a point cloud visualization feature

Arag24, 2019-01-06:
@theseankelly , I'm following the previous thought of @G-Kvist about [Open3D](https://github.com/IntelVCL/Open3D). Even if I never used it till now _(it' also in my to-do list)_, this lib is becoming more and more popular... 
Some raw benefits I see : 

- Python bindings. Of course, goal of this topic. Working with NumPy looks OK [see here](http://www.open3d.org/docs/tutorial/Basic/working_with_numpy.html)
- Lightweight C++ point cloud visualization possible. For this item, I already know some people using it instead of any PCL viewer.
- Active community. I'm following this repo since its start I'm thinking (beginning of 2018), and content is growing really fast, they are also reactive...
- 

Some disadvantages : 

- Only support C++11.

These are (or were) guys from Intel Intelligent Systems Lab / Intel Visual Computing Lab. (California I think). lib was mainly used for registration at beignning. Now RGB-D camera (Intel Realsense is natively supported 🤔)

So, for me it's a big **YES**!

Hope it helps,

PR #153 (2019-06-24): New module: Python bindings
These commits implement python bindings for the c++ API. I've also created the metadata for a v0.13.0 release.

Check out the commit descriptions and the included doc changes for more details.

Closes #122
closed 2019-06-24 by tpanzarella