ifm/ifm3d #123: Not able to update firmware on o3d313

ShettyHarapanahalli, 2018-12-04:
I am not able to update the firmware on camera, I get the below error - 
Firmware update is not supported with dynamic IP address.Change the device to a static IP address before starting the ..

How do we switch the device to static ip.

ShettyHarapanahalli, 2018-12-04:
I made a make for the ifm3d and after this I tried the make check. I get error  indicating below - 
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.IlluTemp (3004 ms)
[----------] 10 tests from Image (54082 ms total)

[----------] Global test environment tear-down
[==========] 10 tests from 1 test case ran. (54082 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 10 tests, listed below:
[  FAILED  ] Image.MoveCtor
[  FAILED  ] Image.MoveAssignmentOperator
[  FAILED  ] Image.CopyCtor
[  FAILED  ] Image.CopyAssignmentOperator
[  FAILED  ] Image.References
[  FAILED  ] Image.CloudMechanics
[  FAILED  ] Image.XYZImage
[  FAILED  ] Image.ComputeCartesian
[  FAILED  ] Image.TimeStamp
[  FAILED  ] Image.IlluTemp

10 FAILED TESTS
modules/image/test/CMakeFiles/check_image.dir/build.make:57: recipe for target 'modules/image/test/CMakeFiles/check_image' failed
make[3]: *** [modules/image/test/CMakeFiles/check_image] Error 1
CMakeFiles/Makefile2:741: recipe for target 'modules/image/test/CMakeFiles/check_image.dir/all' failed

inbangsa, 2018-12-05:
For changing to static Ip use ifm3d tooling as follow 

Assuming Device ip as "192.168.0.68" and changing it to "192.168.0.61" and also making dhcp off.
```
> echo  {"ifm3d":{"Net": {"StaticIPv4Address": "192.168.0.61","StaticIPv4Gateway": "192.168.0.201","StaticIPv4SubNetMask": "255.255.255.0","UseDHCP": "false"}}} | ifm3d --ip=192.168.0.68 config
```
but for above command you must know the Ip Address of the device.  you can use [troubleshoot guide](https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md#dhcp-enabled-device-on-dhcp-network) 

 Testcases: By default ifm3d assumes device ip as 192.168.0.69 you can change this by setting env variable IFM3D_IP=<device IP> and for other connection issues follow steps in [troubleshoot guide](https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md)

ShettyHarapanahalli, 2018-12-05:
I have updated the software on camera but still make check fails, I have camera IP as 192.168.178.36
I am able to ping the camera as well. 
I have cleaned the make and rebuild it, but still the make check fails. I get the error as - 
make check
[ 22%] Built target ifm3d_camera_shared
[ 37%] Built target ifm3d_framegrabber_shared
[ 44%] Built target ifm3d_image_shared
[ 51%] Built target gtest
[ 59%] Built target gtest_main
[ 70%] Built target ifm3d-image-tests
Scanning dependencies of target check_image
[==========] Running 10 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 10 tests from Image
[ RUN      ] Image.MoveCtor
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.MoveCtor (6009 ms)
[ RUN      ] Image.MoveAssignmentOperator
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.MoveAssignmentOperator (6008 ms)
[ RUN      ] Image.CopyCtor
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.CopyCtor (6008 ms)
[ RUN      ] Image.CopyAssignmentOperator
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.CopyAssignmentOperator (6008 ms)
[ RUN      ] Image.References
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.References (6008 ms)
[ RUN      ] Image.CloudMechanics
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.CloudMechanics (6007 ms)
[ RUN      ] Image.XYZImage
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.XYZImage (6008 ms)
[ RUN      ] Image.ComputeCartesian
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.ComputeCartesian (6009 ms)
[ RUN      ] Image.TimeStamp
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.TimeStamp (3004 ms)
[ RUN      ] Image.IlluTemp
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] Image.IlluTemp (3005 ms)
[----------] 10 tests from Image (54074 ms total)

[----------] Global test environment tear-down
[==========] 10 tests from 1 test case ran. (54074 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 10 tests, listed below:
[  FAILED  ] Image.MoveCtor
[  FAILED  ] Image.MoveAssignmentOperator
[  FAILED  ] Image.CopyCtor
[  FAILED  ] Image.CopyAssignmentOperator
[  FAILED  ] Image.References
[  FAILED  ] Image.CloudMechanics
[  FAILED  ] Image.XYZImage
[  FAILED  ] Image.ComputeCartesian
[  FAILED  ] Image.TimeStamp
[  FAILED  ] Image.IlluTemp

10 FAILED TESTS
modules/image/test/CMakeFiles/check_image.dir/build.make:57: recipe for target 'modules/image/test/CMakeFiles/check_image' failed
make[3]: *** [modules/image/test/CMakeFiles/check_image] Error 1
CMakeFiles/Makefile2:741: recipe for target 'modules/image/test/CMakeFiles/check_image.dir/all' failed
make[2]: *** [modules/image/test/CMakeFiles/check_image.dir/all] Error 2
CMakeFiles/Makefile2:76: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
Makefile:184: recipe for target 'check' failed
make: *** [check] Error 2

How do I fix this.

ShettyHarapanahalli, 2018-12-05:
ping -c 5 192.168.178.36
PING 192.168.178.36 (192.168.178.36) 56(84) bytes of data.
64 bytes from 192.168.178.36: icmp_seq=1 ttl=64 time=0.436 ms
64 bytes from 192.168.178.36: icmp_seq=2 ttl=64 time=0.405 ms
64 bytes from 192.168.178.36: icmp_seq=3 ttl=64 time=0.404 ms
64 bytes from 192.168.178.36: icmp_seq=4 ttl=64 time=0.435 ms
64 bytes from 192.168.178.36: icmp_seq=5 ttl=64 time=0.422 ms

--- 192.168.178.36 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4102ms
rtt min/avg/max/mdev = 0.404/0.420/0.436/0.023 ms

ShettyHarapanahalli, 2018-12-05:
ifm2d is not recognized as the make install if not done - 

ifm3d ls
ifm3d: command not found

inbangsa, 2018-12-05:
For installing the ifm3d please do a 

sudo make install 

From build directory.

ShettyHarapanahalli, 2018-12-05:
It's working after setting the env variable as indicated below all is good. 

$ export IFM3D_IP=192.168.178.36
$ export no_proxy=192.168.178.36

ShettyHarapanahalli, 2018-12-05:
how do we set the ip for ros node ?

inbangsa, 2018-12-05:
I am not an expert on ros but as per my knowledge its the same way we provide IP to ifm3d by setting env variables. Behind the ros camera node its ifm3d acting as a driver for camera. you can check our ros installation [guide](https://github.com/ifm/ifm3d-ros/blob/master/doc/building.md)

ShettyHarapanahalli, 2018-12-05:
I get this error when I run catkin_make run_tests for ros node
Do we again set any other env variable 
[ WARN] [1544022509.685131958]: -100001: Lib: XMLRPC Timeout - can you `ping' the sensor?
[ WARN] [1544022509.685518916]: Could not initialize pixel stream!

inbangsa, 2018-12-05:
Please set the env variables in the same command window where you are launching the camera node.

inbangsa, 2018-12-05:
I am very new to ros,  @tpanzarella @graugans need your comment on this issue.

tpanzarella, 2018-12-05:
The ip address of the camera is passed to `ifm3d` via the ROS launch file. See [here](https://github.com/ifm/ifm3d-ros/blob/master/launch/camera.launch#L6). It is propogated from the `ip:=<...>` command line argument.

ShettyHarapanahalli, 2018-12-05:
Thats in the launch file, i have to make the source first. 
when I run the "catkin_make run_tests", the "rostest-test_ifm3d.xml" is having the default ip. This file gets dynamic created on every make.

tpanzarella, 2018-12-05:
You can modify [this file](https://github.com/ifm/ifm3d-ros/blob/master/test/ifm3d.test) to look like:

```
<?xml version="1.0"?>
<launch>
  <include file="$(find ifm3d)/launch/camera.launch">
     <arg name="ip" value="XXX.XXX.XXX.XXX"/>
  </include>
  <test pkg="ifm3d" test-name="test_camera" type="test_camera.py"/>
</launch>
```
Replace `XXX.XXX.XXX.XXX` with your camera's ip address.

The other option, is to use the default IP when building/running tests, then setting your non-default ip for runtime purposes.

graugans, 2018-12-06:
@ShettyHarapanahalli it looks like your problem is solved. Can you please close the issue in this case.