ifm/ifm3d #124: software trigger mode

hjw1007, 2018-12-06:
I run O3D303 in ubuntu 16.04 by software trigger mode,but I get only 3 fps.How can I get much frame image?
How to use sw for example,.This is my code.Thanks!

`auto cam = ifm3d::Camera::MakeShared();
auto fg = std::make_shared<ifm3d::FrameGrabber(cam,ifm3d::IMG_AMP|ifm3d::IMG_CART|ifm3d::IMG_RDIS);
 auto img = std::make_shared<ifm3d::ImageBuffer>();
 while (true)
    {
      fg->SWTrigger();
      if (! fg->WaitForFrame(img.get(), 1000))
      {
        std::cerr << "Timeout waiting for camera!" << std::endl;
        return -1;
      }
    rdis = img->DistanceImage();
............
}
`

graugans, 2018-12-06:
@hjw1007 this depends on your configuration.

```
ifm3d@7bc03a89a811:~$ ifm3d --ip=192.168.0.102 hz --sw --nframes=100
FrameGrabber running at: 10.7118 Hz
100 frames captured, over 1 runs
```

hjw1007, 2018-12-07:
@graugans   
This is my configuration
./ifm3d dump
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "No Description",
        "Id": "729243532",
        "Imager": {
          "Channel": "0",
          "ClippingBottom": "263",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
          "ClippingRight": "351",
          "ClippingTop": "0",
          "ContinuousAutoExposure": "false",
          "EnableAmplitudeCorrection": "true",
          "EnableFastFrequency": "false",
          "EnableFilterAmplitudeImage": "false",
          "EnableFilterDistanceImage": "true",
          "EnableRectificationAmplitudeImage": "false",
          "EnableRectificationDistanceImage": "false",
          "ExposureTime": "2600",
          "ExposureTimeList": "2600",
          "FrameRate": "25",
          "MaxAllowedLEDFrameRate": "22.9",
          "MinimumAmplitude": "42",
          "Resolution": "1",
          "SpatialFilter": {},
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {},
          "TemporalFilterType": "0",
          "ThreeFreqMax2FLineDistPercentage": "80",
          "ThreeFreqMax3FLineDistPercentage": "80",
          "TwoFreqMaxLineDistPercentage": "80",
          "Type": "upto30m_low",
          "UseSimpleBinning": "false"
        },
        "Index": "1",
        "LogicGraph": "{\"IOMap\": {\"OUT1\": \"RFT\",\"OUT2\": \"AQUFIN\"},\"blocks\": {\"B00001\": {\"pos\": {\"x\": 200,\"y\": 200},\"properties\": {},\"type\": \"PIN_EVENT_IMAGE_ACQUISITION_FINISHED\"},\"B00002\": {\"pos\": {\"x\": 200,\"y\": 75},\"properties\": {},\"type\": \"PIN_EVENT_READY_FOR_TRIGGER\"},\"B00003\": {\"pos\": {\"x\": 600,\"y\": 75},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT1\"},\"B00005\": {\"pos\": {\"x\": 600,\"y\": 200},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT2\"}},\"connectors\": {\"C00000\": {\"dst\": \"B00003\",\"dstEP\": 0,\"src\": \"B00002\",\"srcEP\": 0},\"C00001\": {\"dst\": \"B00005\",\"dstEP\": 0,\"src\": \"B00001\",\"srcEP\": 0}}}",
        "Name": "Application",
        "PcicEipResultSchema": "{\"layouter\" : \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicPnioResultSchema": "{\"layouter\" : \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicTcpResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"ascii\" }, \"elements\": [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"blob\", \"id\": \"normalized_amplitude_image\" }, { \"type\": \"blob\", \"id\": \"distance_image\" }, { \"type\": \"blob\", \"id\": \"x_image\" }, { \"type\": \"blob\", \"id\": \"y_image\" }, { \"type\": \"blob\", \"id\": \"z_image\" }, { \"type\": \"blob\", \"id\": \"confidence_image\" }, { \"type\": \"blob\", \"id\": \"diagnostic_data\" }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "TemplateInfo": "{\n}\n",
        "TriggerMode": "2",
        "Type": "Camera"
      }
    ],
    "Device": {
      "ActiveApplication": "1",
      "ArticleNumber": "O3D303",
      "ArticleStatus": "AD",
      "Description": "",
      "DeviceType": "1:2",
      "EthernetFieldBus": "1",
      "EthernetFieldBusEndianness": "0",
      "EvaluationFinishedMinHoldTime": "10",
      "ExtrinsicCalibRotX": "0",
      "ExtrinsicCalibRotY": "0",
[... cut ...]
        "ELDK": "GOLDENEYE_ELDK-2016-07-14_09-36-02-2e9219cce224332ecbe2f36934d2c3c7aa738c78",
        "IFM_Software": "1.6.2114",
        "Linux": "Linux version 3.11.0-00002-gd1bbe0b (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Fri Jun 24 16:34:26 CEST 2016",
        "Main_Application": "0.4.222"
      },
      "ifm3d_version": 1102
    }
  }
}

`$ ./ifm3d --ip=192.168.0.69 hz --sw --nframes=100
FrameGrabber running at: 3.79523 Hz
`
I try this code get FrameGrabber running at: 3.79523 Hz. I want to use 100k get much frame, how to deal with it? Thanks!

graugans, 2018-12-07:
@hjw1007 unfortunately in the 100k mode you can not get very high framerates. Depending on your application you can try to switch to ``under5m_low`` mode. Another thing you might consider is the fact to use the Continuous trigger instead of the Software trigger. So my suggestion in #115 was not to use the software trigger in a permanent way but to _stop_ the camera temporary switch to software trigger mode. In the software trigger mode you always dealing with latency of the network.

In general the less data to transfer the better the frame rate is. So transfering Cartesian coordinates and distance data is typically not useful:

```
(cam,ifm3d::IMG_AMP|ifm3d::IMG_CART|ifm3d::IMG_RDIS);
```

Depending on your compute I recommend to use the RDIS image and calculate the cartesian data by yourself. This saves some bandwidth. And of course switching to firmware 1.23.1522 also can give you some performance improvements.

graugans, 2019-02-05:
@hjw1007 do you need any further information, or can we close this?

hjw1007, 2019-02-11:
@graugans Thanks. Closing isue.

closed 2019-02-11 by hjw1007