ifm/ifm3d #126: How to change the IP address of camera properly?

asaini12, 2018-12-20:
Hello,
I need to change the IP address of camera from 192.168.0.69 to the network we are using on the robot. 

I am having issues. Just to test, I changed the IP address from 192.168.0.69 to 192.168.0.5. I did that using "ifm3d config". It finished successfully. But if I run "ifm3d dump" after that I get the error "XMLRPC Timeout - can you 'ping' the sensor?" I can ping the camera using "ping 192.168.0.5". And if I run ifm3d commands using "ifm3d --ip=192.168.0.5 ..... " they run properly. 

So my question is how can I change the IP address of the camera so that everything runs the way it runs with camera ip set to 192.168.0.69?

inbangsa, 2018-12-21:
Hello @asaini12  
if you don't specify flag --ip then ifm3d takes the IP value from env variable IFM3D_IP or if thats is also not available it assumes ip as=192.168.0.69 

So In your case the IP is changes to 192.168.0.5 but  when you don't provide  --ip switch so it tries to connect to default IP 192,168.0.69.  so for you please set the env Variable IFM3D_IP=192.168.0.5 and then everything should work.  All this steps are very well explained in our [troubleshoot](https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md#provide-the-device-ip-address-to-the-ifm3d) guide.

asaini12, 2018-12-21:
Thank you! That works.

Is there any documentation on how to setup a UDP connection for the camera?

asaini12, 2018-12-26:
We have a jetson board that communicates with rgb cameras over UDP through a switch. So we want to setup the IFM ToF camera the same way. 

In this PDF https://www.ifm.com/mounting/706447UK.pdf , on page 5, section 3, it's mentioned that a UDP port must be enabled for vision assistant program. So I was wondering if there is a way to setup UDP connection for the camera.

graugans, 2019-01-08:
@asaini12 this is only used for the device discovery. I do not see any limitations which prevents you in the described scenario to use a TCP/IP communication.

theseankelly, 2019-10-09:
Hey @asaini12 is this issue resolved?

theseankelly, 2020-01-12:
Closing due to issue age and lack of activity.

commit 8fce7043 (2022-08-24): Merge branch 'bugfix/timestamp_in_python' into 'o3r/main-next'

bugfix timestamp in python

Closes #126

See merge request syntron/support/csr/ifm3d/ifm3d!149
closed 2020-01-12 by theseankelly