ifm/ifm3d #127: XMLRPC was not found: libcurl4-gnutls-dev should be a dependency

Alabate, 2018-12-21:
I tried to install ifm3d into a brand new docker container and it fails with the error:

```
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  XMLRPC was not found.  Make sure the entries XMLRPC_* are set.  (missing:
  XMLRPC_C_FOUND XMLRPC_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindXMLRPC.cmake:125 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  modules/camera/src/libifm3d_camera/CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/ifm3d/build/CMakeFiles/CMakeOutput.log".
See also "/ifm3d/build/CMakeFiles/CMakeError.log".
```

After looking at the implementation of [FindXMLRPC.cmake](https://github.com/Kitware/CMake/blob/master/Modules/FindXMLRPC.cmake#L112), I find out that it fails because it couldn't find `XMLRPC_curl_LIBRARY`. (Same conclusion as this issue: https://github.com/ifm/libo3d3xx/issues/105 )

Installing `libcurl4-gnutls-dev` with apt fixed the build, so I suggest to add this dependency to your readme.

inbangsa, 2019-06-04:
@Alabate  Can we close this issue ?

Alabate, 2019-06-08:
Yes, thanks for the explanation

closed 2019-06-08 by Alabate