ifm/ifm3d #129: Robo Pick&Place

ShettyHarapanahalli, 2019-01-08:
Hi, 

I see the changelog in firmware has instances of robotic pick and place, can get more information about this application. I am using the Robot arm to do something similar application.

G-Kvist, 2019-01-17:
Hi!

This is a ready-made application that runs on the ifm O3D smart sensor (O3D300/2/4): https://www.ifm.com/us/en/shared/product-news/2018/hmi/3d-sensors-for-flexible-automation-of-robot-grippers 

I hope this helps, please reach out if you need any more support.

ShettyHarapanahalli, 2019-02-05:
Thanks for sharing this information. 
I was able to read more information in the below link - 
https://www.ifm.com/binaries/content/assets/pdf-files/en/product-news/2018/hmi/ifm-o3d300-3d-sensors-en-18.pdf

closed 2019-02-05 by ShettyHarapanahalli