ifm/ifm3d #130: arm compilation

jChaudet, 2019-01-18:
Hi,

I tested the v0.11.2 library on arm and modified part of the CmakeLists for my purpose.
This allow me to find the boost package.

```
---
 modules/framegrabber/src/libifm3d_framegrabber/CMakeLists.txt | 2 +-
 modules/pcicclient/src/libifm3d_pcicclient/CMakeLists.txt     | 2 +-
 modules/tools/src/libifm3d_tools/CMakeLists.txt               | 2 +-
 3 files changed, 3 insertions(+), 3 deletions(-)

diff --git a/modules/framegrabber/src/libifm3d_framegrabber/CMakeLists.txt b/modules/framegrabber/src/libifm3d_framegrabber/CMakeLists.txt
index a982e77..90cd674 100644
--- a/modules/framegrabber/src/libifm3d_framegrabber/CMakeLists.txt
+++ b/modules/framegrabber/src/libifm3d_framegrabber/CMakeLists.txt
@@ -47,7 +47,7 @@ target_link_libraries(
     ${LIB_glog}
   PRIVATE
     ifm3d_camera_shared
-    Boost::system
+    ${Boost_SYSTEM_LIBRARY}
     Threads::Threads
   )
 
diff --git a/modules/pcicclient/src/libifm3d_pcicclient/CMakeLists.txt b/modules/pcicclient/src/libifm3d_pcicclient/CMakeLists.txt
index 3a3e270..f1e7737 100644
--- a/modules/pcicclient/src/libifm3d_pcicclient/CMakeLists.txt
+++ b/modules/pcicclient/src/libifm3d_pcicclient/CMakeLists.txt
@@ -50,7 +50,7 @@ target_link_libraries(
   PRIVATE
     ifm3d_camera_shared
     ${LIB_glog}
-    Boost::system
+    ${Boost_SYSTEM}
     Threads::Threads
   )
 
diff --git a/modules/tools/src/libifm3d_tools/CMakeLists.txt b/modules/tools/src/libifm3d_tools/CMakeLists.txt
index a7dde05..202d312 100644
--- a/modules/tools/src/libifm3d_tools/CMakeLists.txt
+++ b/modules/tools/src/libifm3d_tools/CMakeLists.txt
@@ -71,7 +71,7 @@ target_link_libraries(
   ifm3d_tools_shared
   ifm3d_camera_shared
   ${LIB_glog}
-  Boost::program_options
+  ${Boost_PROGRAM_OPTIONS_LIBRARY}
   ${CURL_LIBRARIES}
   )
 
-- 
```

graugans, 2019-01-22:
@jChaudet as far as I know you are using buildroot isn't it? May I ask what version of Boost and CMake is provided by your buildroot version.

jChaudet, 2019-01-22:
@graugans 
Boost 1.64
CMake 3.8.2

graugans, 2019-02-05:
@jChaudet I have not used ``buildroot`` for years. I do expect it is possible to add patches there like we do within Yocto.

jChaudet, 2019-02-20:
@graugans, sure, I already applied this in my buildroot with a patch. 
This is ok for me, this message is to help other users.

closed 2019-02-20 by jChaudet