ifm/ifm3d #131: Question: Coordinate system and confidence information

katdti, 2019-02-05:
Hallo

I have build ifm3d and ifm3d-examples on a windows pc.
I am now trying to access the confidence information for each point in the point cloud from a O3D303 camera using factory settings.

Questions:
- If I look in the ifmVisionAssistant - it look like x,y,z is not the same as  x,y,z in the PCL point cloud structure. How is x,y,z defined in the point cloud? 
- There is a ConfidenceImage - but how do I  get/lookup the confidence bit for a `cloud->points[i]` in the point cloud?

*This is my code logging the point cloud:*
```c++
#include <iostream>
#include <memory>
#include <opencv2/opencv.hpp>
#include <ifm3d/camera.h>
#include <ifm3d/fg.h>
#include <ifm3d/image.h>

int main(int argc, const char **argv)
{
  auto cam = ifm3d::Camera::MakeShared("192.0.2.2");
  

  ifm3d::ImageBuffer::Ptr img = std::make_shared<ifm3d::ImageBuffer>();
  ifm3d::FrameGrabber::Ptr fg =
    std::make_shared<ifm3d::FrameGrabber>(
      cam, ifm3d::IMG_AMP|ifm3d::IMG_CART|ifm3d::IMG_RDIS);

  if (! fg->WaitForFrame(img.get(), 1000))
    {
      std::cerr << "Timeout waiting for camera!" << std::endl;
      return -1;
    }
  auto cloud = img->Cloud();
  for (int i = 0; i < cloud->points.size(); i++) {
	  std::cout << "- XYZI(" << i << "):\t" << cloud->points[i].x << "\t" << cloud->points[i].y << "\t" << cloud->points[i].z << "\t" << cloud->points[i].intensity << std::endl;
  }
  return 0;
}
```

*The output of the program:*
`
- XYZI(0):	0.956	0.61	0.428	1443
- XYZI(1):	0.947	0.595	0.422	1531
- XYZI(2):	0.944	0.585	0.419	1489
- XYZI(3):	0.99	0.604	0.438	1436
`
Comparing with below - it seems the order is not x,y,z but z,x,y and some values is negated.

*The same in ifmVisionAssistant *
[image]

tpanzarella, 2019-02-05:
> * If I look in the ifmVisionAssistant - it look like x,y,z is not the same as  x,y,z in the PCL point cloud structure. How is x,y,z defined in the point cloud?

That is correct. The coordinate frame you see in Vision Assistant is what we refer to as the `optical frame`. Given that the provenance of this library is rooted in mobile robotics our `camera frame` follows a standard right-hand coordinate system with x forward, y left, and z up. While `ifm3d` is independent of ROS, my original use case for writing the library was (and remains today) for usage within the ROS framework. To that end, notes on coordinate frame conventions used in ROS (and the wider robotics community) can be referenced in [REP 0103](http://www.ros.org/reps/rep-0103.html#coordinate-frame-conventions). I hope this helps.

> * There is a ConfidenceImage - but how do I  get/lookup the confidence bit for a `cloud->points[i]` in the point cloud?

Sure, both the point cloud index `i` and the `row`/`col` indices into the confidence (and other) images are zero based. So, to compute `row` and `col` from `i`:

```
int row = i / ncols
int col = i % ncols
```

Where `ncols` is the number of columns in the image/point cloud. Of course, the above is integer arithmetic. I hope this helps.

For deeper reference, the construction of the point cloud in the frame grabber code happens [here](https://github.com/ifm/ifm3d/blob/master/modules/image/src/libifm3d_image/image_buffer_impl.hpp#L186) This should elucidate both the coord frame transformation done in-line with the pixel parsing as well as the indexing scheme.

katdti, 2019-02-06:
Thanks for very good links. Closing isue.

closed 2019-02-06 by katdti