ifm/ifm3d #133: make check failed due to camera settings

wixiw, 2019-02-18:
When trying to build the instructions defined here : 
https://github.com/ifm/ifm3d-ros/blob/master/doc/kinetic.md

When fall into a trap because an error was generated : 
ifm3d-opencv-tests.cpp:33: Failure Value of: fg->WaitForFrame(im.get(), 1000)

It happened that a camera reset to factory settings solved the issue. Hence I share my experience with you as it's highly possible that either : 
- the timeout should be adapted to an internal camera settings
- a log should be added to warn the user about the faulty parameter

graugans, 2019-02-22:
@tpanzarella what do you think about adding the factory reset to the [Global Set-Up](https://github.com/google/googletest/blob/master/googletest/docs/advanced.md#global-set-up-and-tear-down) by sub classing the ``::testing::Environment`` class?

theseankelly, 2020-01-12:
Handling arbitrary camera settings is outside the scope of unit testing -- the camera must start from a known good state (factory reset)

closed 2020-01-12 by theseankelly