ifm/ifm3d #134: Question about images related to the mask applied

jChaudet, 2019-02-20:
Hi,
I try to reduce latency from pmd chip to `m_fg->WaitForFrame(m_im.get(), m_timeout_msecs ); `

Ntp is activated.
I compute the delay between :
 `m_im->TimeStamp() and getRealTime() right after m_fg->WaitForFrame`

**Example 1:**
- O3D303
- 12 fps
- expo = 7000µs - ratio = 20
- double expo
- < 30m_moderate
- mask = 3
- container: ImageBuffer

```
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "984411720",
        "Imager": {
          "AutoExposureMaxExposureTime": "10000",
          "AutoExposureReferencePointX": "88",
          "AutoExposureReferencePointY": "66",
          "AutoExposureReferenceROI": "{\"ROIs\":[{\"id\":0,\"group\":0,\"type\":\"Rect\",\"width\":130,\"height\":100,\"angle\":0,\"center_x\":88,\"center_y\":66}]}",
          "AutoExposureReferenceType": "0",
          "Channel": "3",
          "ClippingBottom": "131",
          "ClippingCuboid": "{\"XMax\":1.2,\"XMin\":-1.2,\"YMax\":100,\"YMin\":-100,\"ZMax\":8,\"ZMin\":0}",
          "ClippingLeft": "0",
          "ClippingRight": "175",
          "ClippingTop": "0",
          "ContinuousAutoExposure": "false",
          "ContinuousUserFrameCalibration": "false",
          "EnableAmplitudeCorrection": "true",
          "EnableFastFrequency": "true",
          "EnableFilterAmplitudeImage": "false",
          "EnableFilterDistanceImage": "true",
          "EnableRectificationAmplitudeImage": "false",
          "EnableRectificationDistanceImage": "false",
          "ExposureTime": "7000",
          "ExposureTimeList": "350;7000",
          "ExposureTimeRatio": "20",
          "FrameRate": "12",
          "MaxAllowedLEDFrameRate": "16.4",
          "MinimumAmplitude": "42",
          "Resolution": "0",
          "SpatialFilter": {},
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {},
          "TemporalFilterType": "0",
          "ThreeFreqMax2FLineDistPercentage": "80",
          "ThreeFreqMax3FLineDistPercentage": "80",
          "TwoFreqMaxLineDistPercentage": "80",
          "Type": "upto30m_moderate",
          "UseSimpleBinning": "false"
        },
        "Index": "1",
        "LogicGraph": "{\"IOMap\": {\"OUT1\": \"RFT\",\"OUT2\": \"AQUFIN\"},\"blocks\": {\"B00001\": {\"pos\": {\"x\": 200,\"y\": 200},\"properties\": {},\"type\": \"PIN_EVENT_IMAGE_ACQUISITION_FINISHED\"},\"B00002\": {\"pos\": {\"x\": 200,\"y\": 75},\"properties\": {},\"type\": \"PIN_EVENT_READY_FOR_TRIGGER\"},\"B00003\": {\"pos\": {\"x\": 600,\"y\": 75},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT1\"},\"B00005\": {\"pos\": {\"x\": 600,\"y\": 200},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT2\"}},\"connectors\": {\"C00000\": {\"dst\": \"B00003\",\"dstEP\": 0,\"src\": \"B00002\",\"srcEP\": 0},\"C00001\": {\"dst\": \"B00005\",\"dstEP\": 0,\"src\": \"B00001\",\"srcEP\": 0}}}",
        "Name": "New application",
        "PcicEipResultSchema": "{\"layouter\" : \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicPnioResultSchema": "{\"layouter\" : \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicTcpResultSchema": "{\n    \"elements\": [\n        {\n            \"id\": \"start_string\",\n            \"type\": \"string\",\n            \"value\": \"star\"\n        },\n        {\n            \"id\": \"normalized_amplitude_image\",\n            \"type\": \"blob\"\n        },\n        {\n            \"id\": \"distance_image\",\n            \"type\": \"blob\"\n        },\n        {\n            \"id\": \"confidence_image\",\n            \"type\": \"blob\"\n        },\n        {\n            \"id\": \"end_string\",\n            \"type\": \"string\",\n            \"value\": \"stop\"\n        }\n    ],\n    \"format\": {\n        \"dataencoding\": \"ascii\"\n    },\n    \"layouter\": \"flexible\"\n}\n",
        "RtspOverlayStyle": "{\"ROI\": {\"default\": {\"visible\": true, \"filled\": false, \"use_symbol\": false, \"label_alignment\": \"top\", \"label_content\": \"\", \"label_background\": \"black\", \"label_fontsize\": 8, \"label_failonly\": false}, \"model_defaults\": {}, \"specific\": {} } }",
        "TemplateInfo": "",
        "TriggerMode": "1",
[... 104 lines cut ...]
[image]

**Example 2**
- O3D303
- 12 fps
- expo = 7000µs - ratio = 20
- double expo
- < 30m_moderate
- mask = 10
- container: ImageBuffer

I retrieve the cloud :
```
        pcl::copyPointCloud(*(m_im->Cloud().get()), *m_cloudForCopy);
```

With this, I have no trouble but a latency > to 120 ms

[image]

So my question is:
Why are the amplitude and the distance matrices empty with mask 3 ?
I also use this method to compute Ifm O3X100 sensor and I have no trouble with it.

Thanks.

theseankelly, 2019-10-09:
Hi @jChaudet, please help me make sure I understand what you're asking. You're saying when you use mask = 0x3 the data inside the amplitude and distance buffers is all zero?

jChaudet, 2019-10-17:
Hi,
exactly.
I spoke about this with @Arag24 and he didn't succeed to reproduce.
Since then, I use the mask 10.
I should need to mask 3 in two months and so I will test again to verify this behavior at this moment.

jChaudet, 2019-12-23:
Hi @theseankelly, 
problem was I use float for both o3x100 and o3d303 instead of float for o3x100 and uint16_t for o3d303.

```
float* dist_pixel = distance_img.ptr<float>(i);
float* amp_pixel = amplitude_img.ptr<float>(i);

uint16_t* dist_pixel = distance_img.ptr<uint16_t>(i);
uint16_t* amp_pixel = amplitude_img.ptr<uint16_t>(i);
```

That was an overflow, not zero value.

Thanks.

commit 04e617b4 (2022-09-08): Merge branch 'bugfix/debs_issues' into 'o3r/main-next'

Bugfix/debs issues

Closes #134

See merge request syntron/support/csr/ifm3d/ifm3d!152
closed 2019-12-23 by jChaudet