ifm/ifm3d #135: I can't obtain the point cloud of a cup two meters away with detailed surface geometry

joggingsong, 2019-02-27:
Hi, all
I try to capture the geometry of one cup 2m away using O3D303.However the result of point cloud is not satisfactory. 
The setting is as follows

> {
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "1851897342",
        "Imager": {
          "Channel": "0",
          "ClippingBottom": "263",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
          "ClippingRight": "351",
          "ClippingTop": "0",
          "ContinuousAutoExposure": "false",
          "EnableAmplitudeCorrection": "true",
          "EnableFastFrequency": "false",
          "EnableFilterAmplitudeImage": "true",
          "EnableFilterDistanceImage": "true",
          "EnableRectificationAmplitudeImage": "true",
          "EnableRectificationDistanceImage": "true",
          "ExposureTime": "1000",
          "ExposureTimeList": "1000",
          "FrameRate": "25",
          "MaxAllowedLEDFrameRate": "100",
          "MinimumAmplitude": "42",
          "Resolution": "1",
          "SpatialFilter": {},
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {},
          "TemporalFilterType": "0",
          "ThreeFreqMax2FLineDistPercentage": "80",
          "ThreeFreqMax3FLineDistPercentage": "80",
          "TwoFreqMaxLineDistPercentage": "80",
          "Type": "under5m_low",
          "UseSimpleBinning": "false"
        },
        "Index": "1",
        "LogicGraph": "{\"IOMap\": {\"OUT1\": \"RFT\",\"OUT2\": \"AQUFIN\"},\"blocks\": {\"B00001\": {\"pos\": {\"x\": 200,\"y\": 200},\"properties\": {},\"type\": \"PIN_EVENT_IMAGE_ACQUISITION_FINISHED\"},\"B00002\": {\"pos\": {\"x\": 200,\"y\": 75},\"properties\": {},\"type\": \"PIN_EVENT_READY_FOR_TRIGGER\"},\"B00003\": {\"pos\": {\"x\": 600,\"y\": 75},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT1\"},\"B00005\": {\"pos\": {\"x\": 600,\"y\": 200},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT2\"}},\"connectors\": {\"C00000\": {\"dst\": \"B00003\",\"dstEP\": 0,\"src\": \"B00002\",\"srcEP\": 0},\"C00001\": {\"dst\": \"B00005\",\"dstEP\": 0,\"src\": \"B00001\",\"srcEP\": 0}}}",
        "Name": "Sample Application",
        "PcicEipResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicPnioResultSchema": "{\"layouter\" : \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicTcpResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"ascii\" }, \"elements\": [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"blob\", \"id\": \"normalized_amplitude_image\" }, { \"type\": \"blob\", \"id\": \"distance_image\" }, { \"type\": \"blob\", \"id\": \"x_image\" }, { \"type\": \"blob\", \"id\": \"y_image\" }, { \"type\": \"blob\", \"id\": \"z_image\" }, { \"type\": \"blob\", \"id\": \"confidence_image\" }, { \"type\": \"blob\", \"id\": \"diagnostic_data\" }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "TemplateInfo": "",
        "TriggerMode": "2",
        "Type": "Camera"
      }
    ],
    "Device": {
      "ActiveApplication": "1",
      "ArticleNumber": "O3D303",
      "ArticleStatus": "AD",
      "Description": "",
      "DeviceType": "1:2",
      "EthernetFieldBus": "1",
      "EthernetFieldBusEndianness": "0",
      "EvaluationFinishedMinHoldTime": "10",
      "ExtrinsicCalibRotX": "0",
      "ExtrinsicCalibRotY": "0",
      "ExtrinsicCalibRotZ": "0",
      "ExtrinsicCalibTransX": "0",
      "ExtrinsicCalibTransY": "0",
      "ExtrinsicCalibTransZ": "0",
      "IODebouncing": "true",
      "IOExternApplicationSwitch": "0",
      "IOLogicType": "1",
      "IPAddressConfig": "0",
      "ImageTimestampReference": "0",
      "Name": "New sensor",
      "OperatingMode": "0",
      "PNIODeviceName": "",
      "PasswordActivated": "false",
      "PcicProtocolVersion": "3",
      "PcicTcpPort": "50010",
      "SaveRestoreStatsOnApplSwitch": "true",
      "ServiceReportFailedBuffer": "15",
      "ServiceReportPassedBuffer": "15",
      "SessionTimeout": "30",
      "TemperatureFront1": "3276.7",
      "TemperatureFront2": "3276.7",
      "TemperatureIMX6": "24.375",
      "TemperatureIllu": "30.2",
      "UpTime": "0.369444444444444"
    },
    "Net": {
      "MACAddress": "00:02:01:40:D1:05",
      "NetworkSpeed": "0",
      "StaticIPv4Address": "192.168.0.69",
      "StaticIPv4Gateway": "192.168.0.201",
      "StaticIPv4SubNetMask": "255.255.255.0",
      "UseDHCP": "false"
    },
    "Time": {},
    "_": {
      "Date": "Wed Feb 27 13:38:33 2019",
      "HWInfo": {
        "Connector": "#!02_A300_C40_03610607_008023176",
        "Diagnose": "#!02_D322_C34_03760170_008026824",
        "Frontend": "#!02_F342_C34_18_00116_008023607",
        "Illumination": "#!02_I300_001_03543852_008001175",
        "MACAddress": "00:02:01:40:D1:05",
        "Mainboard": "#!02_M381_C41_03805968_008023690",
        "MiraSerial": "8230-59f3-0bec-0848"
      },
      "SWVersion": {
        "Algorithm_Version": "1.2.19",
        "Calibration_Device": "00:02:01:40:d1:05",
        "Calibration_Version": "0.9.0",
        "Diagnostic_Controller": "0.3.207",
        "ELDK": "GOLDENEYE_ELDK-2016-07-14_09-36-02-2e9219cce224332ecbe2f36934d2c3c7aa738c78",
        "IFM_Software": "1.6.2114",
        "Linux": "Linux version 3.11.0-00002-gd1bbe0b (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Fri Jun 24 16:34:26 CEST 2016",
        "Main_Application": "0.4.222"
      },
      "ifm3d_version": 1102
    }
  }
}

I will show one example. 
The amplitude image:
[image]
The distance image:
[image]
Two views of the point cloud from two different viewpoints and the cup is circled in red:
[image]

[image]

The cup in distance image looks pretty good. However after the distance image is  converted to point cloud, the cup is not like a 3D cup. Can O3D303 capture the geometrical detail? What can I do to improve the result?

pfrydlewicz, 2019-05-07:
@joggingsong, even though it's a late response, let me give you some hints.
1) The noise might be reduced when you use longer exposure times. I suggest you using triple exposure time mode, you won't need to set any exposure time. It's like HDR with exposure bracketing.
2) Use a temporal filter or a lateral filter (e.g. bilateral filter)
3) when this is your main target and distance, better use the O3D300, it has a 40° opening angle, thus more details can be seen on smaller objects (like using a telelense). Pixel-size in 2m (with 100k resolution) using 60°: 7x7mm, using 40°: 4x4mm.

theseankelly, 2020-01-12:
Hi @joggingsong. Apologies for the extremely delayed response to your query (and thanks for the help @pfrydlewicz)

Are you still working with the camera and/or interested in resolving your challenges?

theseankelly, 2020-01-31:
Closing due to inactivity. Feel free to reopen if further assistance is needed.

commit 2be86ddd (2022-09-08): Merge branch '135-job-failed-448065' into 'o3r/main-next'

Resolve "Job Failed #448065"

Closes #135

See merge request syntron/support/csr/ifm3d/ifm3d!153
closed 2020-01-31 by theseankelly