ifm/ifm3d #136: Camera works on startup but experiencing XML-RPC timeout using commands and running other programs

asaini12, 2019-03-15:
Hello, I can't seem to resolve this weird issue. 

Background - I have a main computer, jetson and ifm3d tofcamera. They are connected via switch, everything has static ip address, and both  main computer and jetson can ping everything else. Jetson doesn't have a desktop installed and has only command line interface. 

Problem - There is a script that runs a program on startup, to process the ifm3d camera's data and send the results over to the main computer. This program runs fine. I can see the results and data sent on the main computer. 

Now if I log on to the jetson board, either via ssh from main computer or by directly hooking up a monitor and keyboard to the jetson, and try to run any program (even the program that runs without any issues on startup) or the ifm3d commands in terminal, I get "XML-RPC timeout: can you ping the sensor?" error. I can ping the sensor, the environment variable for camera's ip is set properly, and I have gone through steps in the troubleshooting guide. But I can't get any of the ifm3d commands or the programs to work. I keep getting the XML-RPC error. I have even tried connecting the camera directly to the jetson board, and the same issue comes up.

graugans, 2019-03-19:
@asaini12 thanks for checking the troubleshooting guide for creating an issue. I would bet this is related to the environment variable. You can check this with 

```shell
echo $IFM3D_IP
```
Another way to verify if the connection to the camera is working is by using the ``ifm3d --ip=<Your IP Address> dump`` with your specific IP address. 

So typically there is a difference in shells there are ``interactive`` shells and ``none-interactive`` it depends how you set your environment variable if they are available for both environments.

asaini12, 2019-03-19:
@graugans Thank you for the reply. At the time of posting variable IFM3D_IP was set properly. and "echo $IFM3D_IP" would give the correct result. 

As you mention, it ended up being the issue with the shell and they weren't being setup properly. 

Thank you!

commit 1b498089 (2022-09-21): Merge branch 'bugfix/debs_with_local_installation' into 'main'

Bugfix/debs with local installation

Closes #136

See merge request syntron/support/csr/ifm3d/ifm3d!156
closed 2019-03-19 by asaini12