ifm/ifm3d #137: pcl StatisticalOutlierRemoval filter says : "Input dataset is not from a projective device! "

LuffyXie, 2019-04-01:
I'm use the pcl/StatisticalOutlierRemoval on the output of a point cloud from an o3d303 camera. It says this:
`[pcl::OrganizedNeighbor::radiusSearch] Input dataset is not from a projective device!
  Residual (MSE) 0.000133, using 1188 valid points`
I update ifm3d and libs to latest version , but problem not solved.

my code:

```c++
pcl::PointCloud<pcl::PointXYZRGBA>::Ptr cloud_filtered (new pcl::PointCloud<pcl::PointXYZRGBA>);
pcl::StatisticalOutlierRemoval<pcl::PointXYZRGBA> sor;
// cloud from o3d303
sor.setInputCloud (cloud);
sor.setMeanK (10);
sor.setStddevMulThresh (1.0);
sor.setKeepOrganized (true);
sor.filter (*cloud_filtered);
```

tpanzarella, 2019-04-01:
@LuffyXie The point type for the ifm3d point cloud is a `pcl::PointXYZI` where the "intensity" channel is the amplitude value registered to the point. It is not a `pcl::PointXYZRGBA`. You could also just use `ifm3d::PointT` for the point type as well.

LuffyXie, 2019-04-02:
I transfer pcl::PointXYZI to pcl::PointXYZRGBA before filter. may be caused by conversion organized point cloud to organized. 
thanks.

theseankelly, 2019-10-09:
Hey @LuffyXie may we close out this issue?

closed 2019-10-28 by theseankelly