ifm/ifm3d #138: Unable to build on 19.04

kubat, 2019-04-23:
Hello,

I'm attempting to build 0.12.0 on Ubuntu 19.04. The build fails on `make repackage`. Log follows:

```console
/tmp $ wget -nv https://github.com/ifm/ifm3d/archive/v0.12.0.tar.gz && \
  tar xf v0.12.0.tar.gz && \
  rm v0.12.0.tar.gz && \
  cd ifm3d-0.12.0 && \
  mkdir build && cd build && \
  cmake -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_MODULE_OPENCV=ON -DBUILD_MODULE_IMAGE=OFF .. && \
  make -j4 && \
  make package &&  \
  make repackage && \
  dpkg -i ifm3d_*.deb

... <snip> ...

[ 12%] Built target ifm3d_camera_shared
[ 16%] Built target gtest
[ 20%] Built target gtest_main
[ 30%] Built target ifm3d-camera-tests
[ 38%] Built target ifm3d_framegrabber_shared
[ 44%] Built target ifm3d-fg-tests
[ 53%] Built target ifm3d-opencv-tests
[ 95%] Built target ifm3d_tools_shared
[100%] Built target ifm3d
Run CPack packaging tool...
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: IFM3D
CPack: - Install project: IFM3D
CPack: -   Install component: camera
CPack: -   Install component: framegrabber
CPack: -   Install component: opencv
CPack: -   Install component: tools
CPack: Create package
-- CPACK_DEBIAN_PACKAGE_DEPENDS not set, the package will have no dependencies.
-- CPACK_DEBIAN_PACKAGE_DEPENDS not set, the package will have no dependencies.
-- CPACK_DEBIAN_PACKAGE_DEPENDS not set, the package will have no dependencies.
-- CPACK_DEBIAN_PACKAGE_DEPENDS not set, the package will have no dependencies.
CPack: - package: /app/ifm3d-0.12.0/build/ifm3d_0.12.0_amd64-camera.deb generated.
CPack: - package: /app/ifm3d-0.12.0/build/ifm3d_0.12.0_amd64-framegrabber.deb generated.
CPack: - package: /app/ifm3d-0.12.0/build/ifm3d_0.12.0_amd64-opencv.deb generated.
CPack: - package: /app/ifm3d-0.12.0/build/ifm3d_0.12.0_amd64-tools.deb generated.
Processing input file: /app/ifm3d-0.12.0/build/ifm3d_0.12.0_amd64-camera.deb
Working directory is: /tmp/ifm3d-deb-cOrleC
Extracting /app/ifm3d-0.12.0/build/ifm3d_0.12.0_amd64-camera.deb to /tmp/ifm3d-deb-cOrleC
Extracing control info to: /tmp/ifm3d-deb-cOrleC/DEBIAN
Listing .so deps of: /app/ifm3d-0.12.0/build/ifm3d_0.12.0_amd64-camera.deb
Found the following .so files: ['/tmp/ifm3d-deb-cOrleC/usr/lib/libifm3d_camera.so']
.so deps *prior* to full-path resolution:
{'libglog.so.0': 1, 'libstdc++.so.6': 1, 'libxmlrpc++.so.8': 1, 'ld-linux-x86-64.so.2': 1, 'libgcc_s.so.1': 1, 'libxmlrpc_client++.so.8': 1, 'libc.so.6': 1}
.so deps *after* to full-path resolution:
{'/lib/x86_64-linux-gnu/libgcc_s.so.1': 1, '/lib/x86_64-linux-gnu/libstdc++.so.6': 1, '/lib/x86_64-linux-gnu/libxmlrpc_client++.so.8': 1, '/lib/x86_64-linux-gnu/libc.so.6': 1, 'ld-linux-x86-64.so.2': 1, '/lib/x86_64-linux-gnu/libglog.so.0': 1, '/lib/x86_64-linux-gnu/libxmlrpc++.so.8': 1}
Found the following .so deps: ['/lib/x86_64-linux-gnu/libgcc_s.so.1', '/lib/x86_64-linux-gnu/libstdc++.so.6', '/lib/x86_64-linux-gnu/libxmlrpc_client++.so.8', '/lib/x86_64-linux-gnu/libc.so.6', 'ld-linux-x86-64.so.2', '/lib/x86_64-linux-gnu/libglog.so.0', '/lib/x86_64-linux-gnu/libxmlrpc++.so.8']
Computing debian dependencies for: /app/ifm3d-0.12.0/build/ifm3d_0.12.0_amd64-camera.deb
dpkg-query: no path found matching pattern /lib/x86_64-linux-gnu/libstdc++.so.6
Traceback (most recent call last):
  File "./ifm3d-dpkg-deps.py", line 234, in <module>
    sys.exit(main())
  File "./ifm3d-dpkg-deps.py", line 226, in main
    retval = dpkg.run()
  File "./ifm3d-dpkg-deps.py", line 70, in run
    self.deb_deps_ = self.generate_deb_deps()
  File "./ifm3d-dpkg-deps.py", line 139, in generate_deb_deps
    out = subprocess.check_output(["dpkg", "-S", so])
  File "/usr/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['dpkg', '-S', '/lib/x86_64-linux-gnu/libstdc++.so.6']' returned non-zero exit status 1
make[3]: *** [CMakeFiles/repackage.dir/build.make:57: CMakeFiles/repackage] Error 1
make[2]: *** [CMakeFiles/Makefile2:73: CMakeFiles/repackage.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/repackage.dir/rule] Error 2
make: *** [Makefile:186: repackage] Error 2
```

The offending shared lib, `libstdc++.so.6` appears not to be owned by any package. A peer lib (same md5) in `/usr/lib/x86_64-linux-gnu` is owned by `libstdc++6:amd64`, but when I change `LD_LIBRARY_PATH` explicitly to put `/usr/lib` first, I get a similar error when `dpkg` looks for `/usr/lib/x86_64-linux-gnu/libgcc_s.so.1`. Fearing this will go down a rabbit-hole of dependencies, I was hoping someone might have insight.

tpanzarella, 2019-04-24:
Hey @kubat I do not have a 19.04 machine available right now, but thanks for the report. Officially, we are supporting the LTS releases, so, 18.04 would be our latest *officially* supported platform. However, this does not mean we will not help. I’ll see if I can get a 19.04 VM up in the next couple of days to take a look into this. In the meantime, if you are just looking to use the software on your box, I.e., not deploying to a production fleet of 19.04 machines that you need to manage, you can probably just skip the `make repackage` part. Basically, this creates a Debian control file for each deb file so that the proper package dependencies are expressed. It will not preclude you from installing and running the software.

kubat, 2019-04-24:
Thanks @tpanzarella! I've found the easiest proxy is to try to build it in Docker. In the mean time, I'm going without `make repackage` and that seems to be working well.

theseankelly, 2020-01-12:
Closing due to issue age, and that a workaround has been found (Skipping repackage)

closed 2020-01-12 by theseankelly