ifm/ifm3d #139: XMLRPC Timeout

danielm1405, 2019-04-25:
I have O3M151 camera connected via Ethernet. I can ping the camera. `tcpdump` shows traffic between the camera and my computer:

```
12:45:26.121201 IP 10.10.1.11.57345 > 10.10.1.1.42000: UDP, length 1460
```

However, `ifm3d --ip=10.10.1.11  ls` command returns:

```
ifm3d error: -100001
Lib: XMLRPC Timeout - can you 'ping' the sensor?
```

Any help would be appreciated.

graugans, 2019-04-25:
@danielm1405  unfortunatly the O3Mxxx is currently not supported by the ifm3d library. Please get in touch with the ifm's support team to receive the O3M SDK.

danielm1405, 2019-04-25:
That's unfortunate. Anyway, thanks for the quick response.

closed 2019-04-25 by danielm1405

danielm1405, 2019-04-26:
By the way, is there any information about supported models of ifm camera? I couldn't find one in this repository.

graugans, 2019-04-26:
@danielm1405 yes, it is. We have [this compatibility matrix](https://github.com/ifm/ifm3d#current-revision) Which states O3D and O3X with the tested firmware versions.

iliiliiliili, 2021-08-28:
@danielm1405 I have made a [python library](https://github.com/iliiliiliili/ifm-o3m-lidar-python) for channel 8 of O3M Lidar if you need it.

commit 99b8b59a (2022-09-26): Merge branch 'improve_pipelines' into 'main'

Improve pipelines

Closes #139

See merge request syntron/support/csr/ifm3d/ifm3d!166