ifm/ifm3d #140: ifm3d ros - not able to change configurations

philihon, 2019-05-01:
Hello,

I am using an IFM 3D camera O3X1. To change the configs in ROS I use:
ifm3d dump > configs.json
In this file I set the parameters to the desired values and I set most of the outputs to "false": e.g. "OutputAmplitudeImage": false etc.
I do this because I don't want these topics to be published in ROS.
I save the file and then I do:
ifm3d config < configs.json
To check if the changes have persisted I again do:
ifm3d dump > configs.json
The changes are there.
Then I launch the IFM cameras and use them normally.  To check the configurations again I do:
ifm3d dump > configs.json
Now only some of the changes I made previously were carried out. Most of the outputs I set to "false" are now "true" again.
What am I doing wrong? How can I fix this? Or is there no fixing for this? Is it not possible to set most of the outputs to "false"?

Thank you very much for the help! :)

Best regards,

Philipp

tpanzarella, 2019-05-01:
@philihon This should really be asked on the [ifm3d-ros issue tracker](https://github.com/ifm/ifm3d-ros/issues) but since we are here now, I'll give you feedback here.

The issue is that you need to specify the PCIC schema to the ROS launch file. The PCIC schema is a bitmask that determines which images will be published by the camera hardware. You have changed this via the `dump` and `config` commands, but your work is getting overridden by the defaults provided in the launch file. We provide a tool `ifm3d schema` to help figure out what you should set this schema value to per your desired configuration. Here is its help output:

```
$ ifm3d schema --help
usage: ifm3d [<global options>] schema [<schema options>]

global options:
  -h [ --help ]            Produce this help message and exit
  --ip arg (=192.168.0.69) IP address of the sensor
  --xmlrpc-port arg (=80)  XMLRPC port of the sensor
  --password arg           Password for establishing an edit-session with the 
                           sensor

schema options:
  --mask arg (=10)      `mask' used to generate an image acquisition schema
  --str arg (=-)        Mask string: e.g., 'IMG_AMP|IMG_CART'
  --dump                Dump masking options and exit
```

You can see that by default, the launch file that we ship sets the schema mask to `15`. You can see that [here](https://github.com/ifm/ifm3d-ros/blob/master/launch/camera.launch#L9). To understand what that means from a practical perspective, you can run `ifm3d schema` like:

```
$ ifm3d schema --mask=15
mask=15, str=-
---
PCIC (O3D-compatible): 

      {
        "layouter": "flexible",
        "format"  : {"dataencoding":"ascii"},
        "elements":
         [
           {"type":"string", "value":"star", "id":"start_string"},
           {"type":"blob", "id":"distance_image"},
           {"type":"blob", "id":"normalized_amplitude_image"},
           {"type":"blob", "id":"amplitude_image"},
           {"type":"blob", "id":"x_image"},
           {"type":"blob", "id":"y_image"},
           {"type":"blob", "id":"z_image"},
           {"type":"blob", "id":"confidence_image"},
           {"type":"blob", "id":"extrinsic_calibration"},
           {"type":"string", "value":"stop", "id":"end_string"}
         ]
      }
   
---
XML-RPC (O3X-compatible): 

      {
         "Apps":
         [
           {
             "Index":"1",
             "OutputDistanceImage":"true",
             "OutputAmplitudeImage":"true",
             "OutputGrayscaleImage":"false",
             "OutputXYZImage":"true",
             "OutputConfidenceImage":"true"
            }
         ]
      }
```

As you can see, for the O3X, this schema turns on all images except the ambient light gray scale image (unrelated: this is currently unsupported by the O3X hardware). Now, per your original note above, you said you wanted to turn off the Amplitude image. Here is how you can do that.

First, rather than using numerical values for the mask, we can use symbolic names. To see what is available, you can:

```
$ ifm3d schema --dump
Masking options:
	IMG_RDIS: 1
	IMG_AMP:  2
	IMG_RAMP: 4
	IMG_CART: 8
	IMG_UVEC: 16
	EXP_TIME: 32
	IMG_GRAY: 64
	ILLU_TEMP: 128
	INTR_CAL: 256
	INV_INTR_CAL: 512
```

So, to get the numeric value for the O3X mask that will stream the radial distance image and the Cartesian data only but not the amplitude  you can: 

(NOTE: the confidence image is an invariant and will always be streamed back -- this constraint is implemented in the driver and exploited to mask out bad pixels on a per frame basis).

```
$ ifm3d schema --str="IMG_RDIS|IMG_CART"
mask=9, str=IMG_RDIS|IMG_CART
---
PCIC (O3D-compatible): 

      {
        "layouter": "flexible",
        "format"  : {"dataencoding":"ascii"},
        "elements":
         [
           {"type":"string", "value":"star", "id":"start_string"},
           {"type":"blob", "id":"distance_image"},
           {"type":"blob", "id":"x_image"},
           {"type":"blob", "id":"y_image"},
           {"type":"blob", "id":"z_image"},
           {"type":"blob", "id":"confidence_image"},
           {"type":"blob", "id":"extrinsic_calibration"},
           {"type":"string", "value":"stop", "id":"end_string"}
         ]
      }
   
---
XML-RPC (O3X-compatible): 

      {
         "Apps":
         [
           {
             "Index":"1",
             "OutputDistanceImage":"true",
             "OutputAmplitudeImage":"false",
             "OutputGrayscaleImage":"false",
             "OutputXYZImage":"true",
             "OutputConfidenceImage":"true"
            }
         ]
      }
```

If you look at the very first line of the output you will see `mask=9`. This is what you want to pass to the launch file:

```
$ roslaunch ifm3d camera.launch schema_mask:=9
```

This should get you started.

philihon, 2019-05-02:
Dear Mr. Panzarella,

Thank you very much for the thoughtful explanations and help!

It is now working as expected.

Best regards,

Philipp Honegger

closed 2019-05-02 by tpanzarella