ifm/ifm3d #141: Timestamp accuracy with IFM O3X100

NicoSchulthess, 2019-05-03:
Hi all,

We have a rather time critical system and we need the point cloud timestamps to be accurate in the order of +/- 50ms. Currently, we are using the time synchronization provided by ifm3d-ros. However, each time we synchronize the time, the delay of the capture time stamps compared to the arrival time at ifm3d seems to be random up to one second. After looking into the source code, it seems like the synchronization only uses an accuracy of seconds. https://github.com/ifm/ifm3d/blob/9db2bc957a0484c75627b227658310b40e0162f9/modules/camera/src/libifm3d_camera/camera_impl.hpp#L904-L913 Is our assumption correct and is there a possibility to increase the accuracy?

Many thanks in advance :)

Nico

graugans, 2019-05-03:
I am currently not in the office to provide a detailed response. Have you checked to use NTP for synchronization. As you have already noticed the ROS approach is based on a second accuracy. There is also a drift between your host clock and the camera Clock. We do provide a document which describes the NTP set-up. https://github.com/ifm/ifm3d/blob/master/doc/ntp-setup.md

NicoSchulthess, 2019-05-04:
Thank you very much for the quick response. Meanwhile, we set up the NTP server for time synchronization and the result was much more accurate than with the synchronization via ROS.

commit e8776c8a (2022-10-06): Merge branch '141-changes-to-the-github-repo' into 'main'

Changes to Github repo

Closes #141

See merge request syntron/support/csr/ifm3d/ifm3d!169
closed 2019-05-04 by NicoSchulthess