ifm/ifm3d #142: Problem with .exe

clementlermurier, 2019-05-03:
Hi !

I'm currently work with the IFM O3D303 and I installed the ifm3d library.
Every included fonctions likes ifm3d --ip=XXX.XXX.XXX.XXX dump works normally.

But I have a problem with the ifm-pcl-viewer.exe because it doesn't work as you can see.
Do you have a solution to overcome that ?

```
C:\ifm3d\0.12.0>ifm3d --ip=169.254.208.219 ls
[
  {
    "Active": true,
    "Description": "",
    "Id": 917918111,
    "Index": 1,
    "Name": "Sample Application"
  }
]

C:\ifm3d\0.12.0\install\bin>ifm3d-pcl-viewer --ip=169.254.208.219
Lib: XMLRPC Timeout - can you 'ping' the sensor ?

```
And of course I can ping it.

Thank you.

tpanzarella, 2019-05-03:
@clementlermurier The `ifm3d-pcl-viewer` does not accept the `--ip=` command line argument. You will have to use the `IFM3D_IP` environment variable in your situation. I am not sure how that works on Windows (setting an environment variable) but on Linux, you could:

```
$ IFM3D_IP=169.254.208.219 ifm3d-pcl-viewer
```

HTH.

clementlermurier, 2019-05-03:
Thank you very much
I can’t test it now so I will answers on monday

Arag24, 2019-05-05:
@clementlermurier , @tpanzarella .

As Tom wrote, you may need to add `IFM3D_IP` to your environment variable.
To do so you may use the `set` Function, inside command terminal (Windows Powershell, CMD,...)

Should look like that : 

`set IFM3D_IP=169.254.208.219`
and launch `ifm3d-pcl-viewer` cmd.

I recommend you to write it  in batch file `install.bat`, it would be effective for both ifm3d tools command + ifm3d-pcl-viewer.
PS: If you want to permanently fill these environnement variables, just replace `set` with `setx`

clementlermurier, 2019-05-06:
Thank you for your help
Now it works perfectly

commit 0f2e878e (2022-09-26): Merge branch 'feature/dist_noise_image' into 'main'

add support for distance noise image

Closes #142

See merge request syntron/support/csr/ifm3d/ifm3d!158
closed 2019-05-06 by clementlermurier