ifm/ifm3d #143: Building 0.12.0 fails, boost not found, windows

pfrydlewicz, 2019-05-06:
I'm having problems building v0.12.0 on windows.
Using VS 2017 (toolset 14.1) cmake cannot find the boost binaries.

I downloaded the binaries as described in the doc page. Also building boost from scratch doesn't solve the problem.
The static libs are 100% in the given folder. Also already crosschecked the env variables like BOOST_LIBRARYDIR, BOOST_ROOT, BOOST_INCLUDEDIR, BOOST_INSTALL_DIR.
Already tried different prefixes for the static libs (just in case), no results.

Any ideas???

Thanks in advance,
Paul

-------------------------------------
Here is the output (CMake v3.13):

  Boost version: 1.64.0

  Boost include path: C:\local\boost_1_64_0

  Could not find the following static Boost libraries:

          boost_system

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
  Boost.
Call Stack (most recent call first):
  modules/framegrabber/src/libifm3d_framegrabber/CMakeLists.txt:5 (find_package)

Arag24, 2019-05-31:
@pfrydlewicz : 2 questions. 
Q1 : Is the binaries sent to you last week are running fine ? 

Q2 : I remembered they had some terrific complications with CMake + boost (I spent lot of time with 1.66 getting same issues with BOOST_LIBRARYDIR & BOOST_ROOT question).
Please have a look here : [https://github.com/PointCloudLibrary/pcl/issues/2201](https://github.com/PointCloudLibrary/pcl/issues/2201)
I used to do binaries CMake 3.10.2 [here](https://cmake.org/files/v3.10/cmake-3.10.2-win64-x64.msi) and Boost 1.64 for MSVC14.1 [here](https://dl.bintray.com/boostorg/release/1.64.0/binaries/boost_1_64_0-msvc-14.1-64.exe).
Are you using boost for other projects with other version ? Would it be acceptable to uninstall to check again with version I used to determine if this could be a root cause ?

Other check (even if I'm pretty sure i's not linked, used CMD as admin when building, sometimes we can find very strange behaviour otherwise ...

Last, in the CMD can you make : 
`echo %BOOST_LIBRARYDIR% ` and  
`echo %BOOST_ROOT% ` to besure PATH are corrects.

theseankelly, 2020-03-31:
Hi @pfrydlewicz  -- I have refreshed our Windows build instructions to use binaries for PCL/OpenCV, and to target the version of `boost` that's distributed along with PCL. Please try out the instructions in the `develop` branch and open a new issue if you run into additional challenges.

https://github.com/ifm/ifm3d/blob/develop/doc/windows.md

closed 2020-03-31 by theseankelly