ifm/ifm3d #144: PCICClient Call hangs

Unmoon, 2019-05-13:
I have this piece of code that is ran right after (successfully) connecting to the camera, which sometimes, after repeated forced disconnections (where camera waits for heartbeats to time out) hangs forever. 

Reconnecting to the camera of course fixes it, but I would like to have a timeout or something similar for it. Is that possible?

```C++
    pcicclient->Call("a" + std::to_string(index));
```
where `index = 1`.

graugans, 2019-05-13:
@Unmoon the issue you describe is caused by this [loop](https://github.com/ifm/ifm3d/blob/master/modules/pcicclient/src/libifm3d_pcicclient/pcicclient_impl.hpp#L517):
```cpp
	while (!has_result.load())
	{
		this->in_cv_.wait(lock);
}
```
From a quick glance over the code it looks like the return type of the inner call to ``Call`` is no checked

```cpp
std::string
ifm3d::PCICClient::Impl::Call(const std::string& request)
{
	std::atomic_bool has_result(false);
	std::string result;

	Call(request, [&](const std::string& content)
	{
		// Copy content, notify and leave callback
		result = content;
		std::unique_lock<std::mutex> lock(this->in_mutex_);
		has_result.store(true);
		this->in_cv_.notify_all();
	});

	std::unique_lock<std::mutex> lock(this->in_mutex_);
	while (!has_result.load())
	{
		this->in_cv_.wait(lock);
	}
	lock.unlock();

	return result;
}
```

By adding a check for the return code before entering the endless loop. The described behaviour should not happen in an regular use additionally we should add a timeout like we do in the framegrabber's  ``WaitForFrame``

```cpp
  try
    {
      if (timeout_millis <= 0)
        {
          this->in_cv_.wait(lock);
        }
      else
        {
          if (this->in_cv_.wait_for(
                lock, std::chrono::milliseconds(timeout_millis)) ==
              std::cv_status::timeout)
            {
              VLOG(IFM3D_TRACE)
                << "Timeout waiting for PCIC call from camera";
              return "";
            }
        }
    }
  catch (const std::system_error& ex)
    {
      LOG(WARNING) << "PCICClient::Call: " << ex.what();
      return "";
}
```

In my opinion to handle this in an appropriate way I would suggest to add a new method:
```cpp
bool
ifm3d::PCICClient::Impl::Call(const std::string& request,std::string& response,long timeout_millis )
```

But in the end @tpanzarella the maintainer has to decide what to do with this.

tpanzarella, 2019-05-13:
@graugans I am fine with your suggestion as you and your team are the authors/maintainers of the PCIC module. I'm happy to merge a suitable PR with associated tests.

tpanzarella, 2019-07-26:
Fix released in `0.15.1`.

commit d8d69deb (2023-01-26): Merge branch '144-viewer-example-link-should-point-to-released-example' into 'main'

Resolve "Viewer example link should point to released example"

Closes #144

See merge request syntron/support/csr/ifm3d/ifm3d!209
closed 2019-07-26 by tpanzarella