ifm/ifm3d #145: Smallest Object detection size

ShettyHarapanahalli, 2019-05-15:
Hi ,

I have an IFM 3D camera, I am planning to use it to detect below-sized objects can you please let me know what is the smallest object that can be detected by IFM 3d camera. 

1. 6.6 cm (2.6 in.) long and 2.9 cm (1.1 in.) in diameter,
2. 5.2 cm (2.1 in.) long and 2.6 cm (1.0 in.) in diameter.  

I am planning to detect this from a min distance, can I keep the camera at 350 mm or nearer. 

Best Regards,
Suman

tpanzarella, 2019-05-15:
This  will be more related to your object segmentation algorithm. The best resolution that we can do on an O3D is ~100k pixels. This is controlled via the `Resolution` parameter of the imager settings for the active application. A setting of `0` is 23k pixels, a setting of `1` is 100k pixels. Assuming the first application on the camera is your application of interest, you can:

Check the current resolution setting:
```
$ ifm3d dump | jq .ifm3d.Apps[0].Imager.Resolution
"0"
```

In this case, it is `0` which maps to 23k pixels. You can change this to 100k with:
```
 $ ifm3d dump | jq '.ifm3d.Apps[0].Imager.Resolution="1"' | ifm3d config
```

You can validate that the setting was accepted by the camera with:
```
$ ifm3d dump | jq .ifm3d.Apps[0].Imager.Resolution
"1"
```

Hopefully this helps.

ShettyHarapanahalli, 2019-05-15:
I am using Mask R-CNN for Object Detection and Segmentation. To get the Depth of the object at a distance of 350 mm, assume an eraser or nut and bolt is my object of interest. Can I get a 3D shape of this at 100k pixels? 

If there are any reference implementation for this kind of application.

graugans, 2019-05-15:
So you might also have to tune the exposure time and mode. I recommend to use the vision assistant for this to have direct feedback. Otherwise your object might have holes

ShettyHarapanahalli, 2019-05-23:
Sure I will use vision assistance, I see the below specification for the camera from the website- 

object size: 200 x 200 mm
reflectivity: 18 %

commit 181874c1 (2022-10-14): Merge branch '1.0.0-doc-fixes' into 'main'

1.0.0 doc fixes

Closes #145, #147, #149, #118, #124, and #152

See merge request syntron/support/csr/ifm3d/ifm3d!160
closed 2019-05-23 by ShettyHarapanahalli