ifm/ifm3d #147: problem with DistanceImage();

clementlermurier, 2019-05-29:
Hello,

I'm currently working with O3D and more especially to work with Distance image and opencv.

All of my workspace work correctly with amplitude image.
When I display the amplitude image, it's work normally.

`
auto cam = ifm3d::Camera::MakeShared();
 auto fg = std::make_shared<ifm3d::FrameGrabber>(cam);
 auto im = std::make_shared<ifm3d::ImageBuffer>();

cv::Mat amp;

  while (true)
  {
    if (! fg->WaitForFrame(im.get(), 1000))
    {
      std::cerr << "Timeout waiting for camera!" << std::endl;
      return -1;
    }

    amp = im->AmplitudeImage();

    amp.convertTo(amp, CV_8U);
    cv::imshow("amp", amp);
    cv::waitKey(0);
  }
`
But when I try the same program with DistanceImage(), this program send me an error of assertion.

I would like to understand why it does not work.

thanks for your help

tpanzarella, 2019-05-29:
By default, `ifm3d` does not stream back the radial distance image rather just the amplitude and the on-camera computed Cartesian data (i.e., the _point cloud_). The data that are streamed back by the camera, and thus populated in the `ImageBuffer` is controlled by the schema mask passed to the `FrameGrabber` ctor. In your call:

```
auto fg = std::make_shared<ifm3d::FrameGrabber>(cam);
```

You are using the `ifm3d::DEFAULT_SCHEMA_MASK`, which as I stated above includes the (normalized) amplitude data and the point cloud. If you would like to, for example, stream back the radial distance image and amplitude (as your code above implies), you can instantiate your `FrameGrabber` like:

```
auto fg = std::make_shared<ifm3d::FrameGrabber>(cam, ifm3d::IMG_AMP|ifm3d::IMG_RDIS);
```

This will make your above code work. Alternatively, if you do not want to modify your code above, you can set the environment variable `IFM3D_MASK` to `3` which will do the same thing. So, assuming your application is in a binary called `a.out`, you can run:

```
$ IFM3D_MASK=3 ./a.out
```

and your code should work as you expect without modification. The magical number `3` above was inferred by looking at the schema bitmask constants [here](https://github.com/ifm/ifm3d/blob/master/modules/framegrabber/src/libifm3d_framegrabber/schema.cpp) There is also a command line tool that you can use to help with schema masks. For example, to discover the magical number `3` above, you can run:

```
$ ifm3d schema --str='IMG_AMP|IMG_RDIS'
mask=3, str=IMG_AMP|IMG_RDIS
---
PCIC (O3D-compatible): 

      {
        "layouter": "flexible",
        "format"  : {"dataencoding":"ascii"},
        "elements":
         [
           {"type":"string", "value":"star", "id":"start_string"},
           {"type":"blob", "id":"distance_image"},
           {"type":"blob", "id":"normalized_amplitude_image"},
           {"type":"blob", "id":"confidence_image"},
           {"type":"blob", "id":"extrinsic_calibration"},
           {"type":"string", "value":"stop", "id":"end_string"}
         ]
      }
   
---
XML-RPC (O3X-compatible): 

      {
         "Apps":
         [
           {
             "Index":"1",
             "OutputDistanceImage":"true",
             "OutputAmplitudeImage":"true",
             "OutputGrayscaleImage":"false",
             "OutputXYZImage":"false",
             "OutputConfidenceImage":"true"
            }
         ]
      }

```

Hopefully this gets you started.

clementlermurier, 2019-05-29:
Thanks you very much for this answer.
Now it's works perfectly.

closed 2019-05-29 by clementlermurier