ifm/ifm3d #148: CMake build PCL1.8.1 faild with VS2015 x64 on Win7

GMSLLB, 2019-06-04:
Now, it always exited that:
CMake Error at C:/Program Files/CMake/share/cmake-3.13/Modules/FindBoost.cmake:2064 :
Unable to find the requested Boost libraries.
Unable to find the Boost header files. Please set BOOST_ROOT to the root directory containing Boost or
BOOST_INCLUDEDIR to the directory containing Boost's headers.
Call Stack :
cmake/pcl_find_boost.cmake:43 <find_package>
CMakeLists.txt:418

My boost version is Boost_1_64_0, installing in the path: D:/boost_1_64_0. 
I have a question of it: How do I install boost?

inbangsa, 2019-06-04:
Please provide details  of your.

Compiler : 
Windows Version : 

Also check the instruction for installing boost  our document for [windows install](https://github.com/ifm/ifm3d/blob/master/doc/windows.md#prepare-dependent-tools)

graugans, 2019-06-04:
The Windows version can be queried like described in this [SO](https://stackoverflow.com/a/42778990/695442) article. Unfortunately I do have found any solution how to query the Visual Studio Version from command line.

Arag24, 2019-06-05:
You can have your compiler info  in such way :

- Win click
- Type 'VS201**X** x64 Native Tools Command Prompt.
- Type `cl`  and then you would have Compiler Version.

On my own I did compiling of PCL 1.8.1 on VS2017 as described [here](https://github.com/ifm/ifm3d/blob/master/doc/windows.md). Compiler is 

```
**********************************************************************
** Visual Studio 2017 Developer Command Prompt v15.5.2
** Copyright (c) 2017 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Program Files (x86)\Microsoft Visual Studio\2017\Community>cl
Microsoft (R) C/C++ Optimizing Compiler Version 19.12.25831 for x64
```
@GMSLLB Do you have capability to compile with such version ?

inbangsa, 2019-06-06:
Thanks @GMSLLB for the information

I tried to reproduce the issue but with for Win7 X64 VS2015 everythings seems to be fine.  

Can you please provide a cmake log. with additional following flags   
```
 --debug-output  -DBoost_DEBUG=1 . 
```
this will give us insight what library version and what type of libArch_Compiler_tools  cmake is trying to search.

theseankelly, 2020-01-12:
Hi @GMSLLB -- have you been able to resolve your issue? If not, can you please provide the logs asked for by @inbangsa ?

theseankelly, 2020-03-31:
Hi @GMSLLB -- I have refreshed our Windows build instructions to use binaries for PCL/OpenCV, and to target the version of `boost` that's distributed along with PCL. Please try out the instructions in the `develop` branch and open a new issue if you run into additional challenges.

https://github.com/ifm/ifm3d/blob/develop/doc/windows.md

commit 48b937de (2022-11-02): Merge branch 'document/update_linux_build_instruction' into 'main'

Restructured build and install instruction for ifm3d

Closes #148

See merge request syntron/support/csr/ifm3d/ifm3d!170
closed 2020-03-31 by theseankelly