ifm/ifm3d #149: How get dynamic IP address of camera

ShettyHarapanahalli, 2019-06-12:
Hi,

I am trying to use the IFM camera on a network where static is not allowed, every time the IP gets updated in this setup. To get the list of IP on the network I have been using "arp-scan -I " command on Ubuntu 18., but it does not list the IFM camera. After connecting the camera with vision assistance, I see the IP list updated output will have this "<IP Address>	00:02:01:40:2f:75	IFM Electronic gmbh".  

Is there any way to get the devices IP updated on camera every time after a reboot, instead of connecting to vision assistance. 

BR,
SSH

graugans, 2019-06-12:
Hey @ShettyHarapanahalli have you tried this: https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md#dhcp-enabled-device-on-dhcp-network if this works I can share some WIP ifm3d discovery code based on https://github.com/HBPVIS/Servus

ShettyHarapanahalli, 2019-06-12:
Ni it is not working, below is the output - 
$avahi-browse -arp 2>/dev/null | grep  ^=.*O3D.*Workstation
=;enp59s0;IPv4;O3D313-40-2F-75\032\09100\05802\05801\05840\0582f\05875\093;Workstation;local;O3D313-40-2F-75.local;192.168.0.69;9;

^Z
[3]+  Stopped                 avahi-browse -arp 2> /dev/null | grep --color=auto ^=.*O3D.*Workstation

$ ifm3d --ip=192.168.0.69 ls
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?

graugans, 2019-06-12:
In your case it looks like the camera IP does not mach your network configuration. From your description I thought you are using DHCP for IP address configuration and you want to get the IP address of the camera.

ShettyHarapanahalli, 2019-06-12:
After using (Just connect camera and disconnect) vision assistance now the command output is - 
s$ avahi-browse -arp 2>/dev/null | grep  ^=.*O3D.*Workstation
=;enp59s0;IPv4;O3D313-40-2F-75\032\09100\05802\05801\05840\0582f\05875\093;Workstation;local;O3D313-40-2F-75.local;10.50.102.128;9; 

ifm3d --ip=10.50.102.128 ls
[
  {
    "Active": false,
    "Description": "",
    "Id": 1034160761,
    "Index": 1,
    "Name": "Sample Application"
  },
  {
    "Active": true,
    "Description": "Trying best config",
    "Id": 1034160762,
    "Index": 2,
    "Name": "Try1"
  }
]

I want to avoid using vision assistance every time of camera reboot.

graugans, 2019-06-12:
I guess the Vision Assistant informs you about the fact that the IP address of the camera does not match your network and asks if it shall change this for you isn't it?

ShettyHarapanahalli, 2019-06-12:
> I guess the Vision Assistant informs you about the fact that the IP address of the camera does not match your network and asks if it shall change this for you isn't it?

Yes that's true, I get a pop up for it .

graugans, 2019-06-12:
The recommended way is to change the IP set-up which fits your network config. If your network is managed by DHCP please use DHCP for the camera otherwise you may run in IP address conflicts.

ShettyHarapanahalli, 2019-06-17:
After I activated DHCP option I did not get any errors but need to find the sensor every time on boot, please do share the code to find IP of the camera, python code will be of good help as well.

theseankelly, 2020-01-31:
FYI we'll be addressing this as a new feature in a coming milestone in #194

closed 2020-01-31 by theseankelly

theseankelly, 2020-02-13:
FYI @ShettyHarapanahalli we've just opened sourced a python implementation of this functionality here: https://github.com/ifm/python-ipv4-discovery/blob/master/discovery.py