ifm/ifm3d #155: O3D303 camera for outdoor use

derektan95, 2019-07-08:
Hi, 
I am reading through the Operating Manual of the O3D303 camera which I recently purchased and I found out about these camera constraints. 

[image]

Since I will be using this camera for outdoor use as well, may I know if that is possible? What will the repercussions be? Thanks!

graugans, 2019-07-08:
Hey @derektan95 the O3D3xx is intended for indoor use. It can work outdoors but this depends on the use case you are planning to solve. If you provide us more details we may can assist you in tuning the camera for your use case

derektan95, 2019-07-08:
Hi, 
Thank you for your prompt response. We intend to use this camera to obtain the point cloud within its field of view in real-time. With this, we intend to perform object detection, specifically, to detect pallets in warehouses. There are some cases where we have to detect pallets in outdoor conditions. 

The current settings are as follows: 
	Trigger Source: Continuous
        Max Background Distance: Up to 30m (2 frequencies)
	Exposure: Low exposure (1 exposure)
        Frame rate: 8Hz 
	Exposure: 2500 μs
	3x3 Median Filter: Active
	X-axis clipping: Everything beyond 5m from camera lens

Let me know how I can optimize the settings. Thanks!

derektan95, 2019-07-08:
Hi @graugans,
Is it possible to discuss this offline? Thanks.

derektan95, 2019-07-10:
Hi @graugans,
Thank you for your prompt response. We intend to use this camera to obtain the point cloud within its field of view. With this, we intend to perform object detection, specifically, to detect pallets in warehouses. There are some cases where we have to detect pallets in outdoor conditions. Let me know if there are better settings for this, thanks!

derektan95, 2019-07-10:
I performed a simple test comparing the point cloud of indoor and outdoor conditions. 

Indoor: 
[image]

Outdoor: 
[image]

It seems like the surface of the object becomes less defined and more fuzzy. I there a setting to adjust for this, or is there another type of 3D camera that can accomodate for this? Why is outdoor light causing such an interference? Thanks!

graugans, 2019-07-10:
@derektan95 this is signal to noise ratio. The sunlight does have a lot energy in the infrared band. Which causes the pixels to saturate. You can try to fine tune your settings or work with a lense hood. Due to the fact this is not related to the software interface I highly recommend that you get in touch with our support team: "Support Syntron" <support.syntron@ifm.com>

derektan95, 2019-07-16:
Thank you very much for your help! I'll be sure to ask around.

Derek

closed 2019-07-16 by derektan95