ifm/ifm3d #157: 03D303 Not showing image output from the Camera on Ubuntu 18.04

srane96, 2019-07-09:
So I installed ifm3d v0.14.0 using apt on my Ubuntu OS.
Following is the configuration of my laptop:
`Linux srane-p5530.local 4.18.0-24-generic #25~18.04.1-Ubuntu SMP Thu Jun 20 11:13:08 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`
I also downloaded the ifm3d-examples repository. I am able to ping my camera but whenever I try to run basic examples, I get the following error:
`Timeout waiting for camera!`
Also, I tried running ifm3d-pcl-viewer, I got a completely blank screen.
Whenever I run `ifm3d dump ` command, it successfully shows the configuration (see attachment
[camera_config.txt](https://github.com/ifm/ifm3d/files/3374604/camera_config.txt)
. Also, I am able to create and delete new applications. 
Following is the firmware version of my device : **1.23.2848**

Opencv version I have is 4.1.0

Can anyone please tell me where the problem might be lying?

srane96, 2019-07-10:
Sorry, it was because it already had two applications preinstalled that were affecting the firmware settings. So after running the command  `$ ifm3d reset` it started working.

closed 2019-07-10 by srane96

graugans, 2019-07-10:
This is most probably due to the fact that the active application was set to hardware or software trigger.