ifm/ifm3d #163: Lens distortion.

srane96, 2019-07-19:
I am trying to run ifmO3D303 camera on Ubuntu 18.04. I am currently using ifm3d-pcl-viewer to visualize the point cloud. I have noticed that the output I am getting for a plane flat surface is not rectangular. It seems distorted. I have seen that you have mentioned lens distortion in documentation, but I could not find any resource which explains how to remove it. Can anyone help me?

tpanzarella, 2019-07-20:
You can turn on the rectification flags for either or both of the amplitude and the distance image. This can be done via `ifm3d dump > camera.json` (to get your current configuration). Modify the output JSON (the key names to turn on rectification should be “obvious”) and then use `ifm3d config < camera.json` to load in the modified settings.

srane96, 2019-07-22:
Thank you so much! Before I close this issue, I would like to ask one more doubt. Does FrameGrabber's `WaitForFrame()` method always returns the latest data from the camera?. For example, I am reading the point cloud data from the camera in a loop and sending it for the processing in each iteration which takes some time. So in the next iteration, when I come back and request next point cloud, is it going to be the latest point cloud or I am gonna receive the old one from some buffer?

tpanzarella, 2019-07-22:
It’s always the latest frame.

srane96, 2019-07-22:
Alright, thank you so much!

srane96, 2019-07-22:
Sorry for bothering you again. But this is the output I get whenever I keep my camera in front of a plane surface, even after setting rectification flags true. Is this the way it should look with curved edges or should be more rectangular with straight edges?
[image]

graugans, 2019-07-22:
@srane96 could you please be so kind and attach the output of the ifm3d dump command?

srane96, 2019-07-22:
Here it is. 
[config.txt](https://github.com/ifm/ifm3d/files/3417792/config.txt)

srane96, 2019-07-22:
I am so sorry for the confusion, I double checked it, and realized that I had set flag true for different application.

closed 2019-07-22 by srane96