ifm/ifm3d #164: [ WARN] [1563824664.818563389]: -100001: Lib: XMLRPC Timeout - can you `ping' the sensor?

saberclaw1213, 2019-07-22:
I am running O3D200 with DHCP enabled. I have fixed the ip to 192.168.4.180
when I ping that ip:
$ ping 192.168.4.180
PING 192.168.4.180 (192.168.4.180) 56(84) bytes of data.
64 bytes from 192.168.4.180: icmp_seq=1 ttl=64 time=0.195 ms
64 bytes from 192.168.4.180: icmp_seq=2 ttl=64 time=0.237 ms
64 bytes from 192.168.4.180: icmp_seq=3 ttl=64 time=0.238 ms
64 bytes from 192.168.4.180: icmp_seq=4 ttl=64 time=0.235 ms
^C
--- 192.168.4.180 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3049ms
rtt min/avg/max/mdev = 0.195/0.226/0.238/0.020 ms

But when I am running roslaunch ifm3d camera.launch

[ INFO] [1563824663.668365487]: Loading nodelet /ifm3d/camera of type ifm3d_ros/camera_nodelet to manager camera_standalone_nodelet with the following remappings:
[ INFO] [1563824663.672303377]: waitForService: Service [/ifm3d/camera_standalone_nodelet/load_nodelet] has not been advertised, waiting...
[ INFO] [1563824663.678883848]: Initializing nodelet with 12 worker threads.
[ INFO] [1563824663.693207047]: waitForService: Service [/ifm3d/camera_standalone_nodelet/load_nodelet] is now available.
[ INFO] [1563824663.759678966]: onInit(): /ifm3d/camera
[ INFO] [1563824663.815896252]: Camera clock will not be sync'd to system clock
[ INFO] [1563824663.815934316]: Running dtors...
[ INFO] [1563824663.816012471]: Initializing camera...
[ INFO] [1563824664.817218002]: Initializing framegrabber...
[ WARN] [1563824664.818563389]: -100001: Lib: XMLRPC Timeout - can you `ping' the sensor?
[ WARN] [1563824664.818752803]: Could not initialize pixel stream!

So i tried to troubleshoot:
I changed (no_proxy="127.0.0.1, localhost, 192.168.0.69, 192.168.0.70, 192.168.4.180, 192.168.1.*")

$ ifm3d ls
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?

but no change.

On the otherhand I am able to use IFM efector to connect device with 192.168.4.180

Can you please suggest what am I doing wrong here.

Thanks

tpanzarella, 2019-07-22:
Did you pass the ‘ip:=<your ip>’ argument on your roslaunch line?

inbangsa, 2019-07-23:
@saberclaw1213 please check this [troubleshoot guide](https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md#select-the-troubleshooting-options)

graugans, 2019-07-23:
@saberclaw1213 just for verification. Are you really trying to connect a [O3D2xx](https://www.ifm.com/at/en/product/O3D200) unfortunately this type of cameras are not supported by ifm3d.

saberclaw1213, 2019-07-23:
Well that make sense now! I had 03D200 with me since a long time so I was just wondering to do something with it. Is that any other way I can make use of it?
@graugans

closed 2019-12-12 by theseankelly

graugans, 2019-12-13:
@saberclaw1213 sorry for the delay, there is some very very old code floating around: https://github.com/christianege/O3DCapsicum which deals with the O3D2xx.