ifm/ifm3d #166: Connection problem with Ifm3d

matteofiorini, 2019-07-25:
I am stucked with a problem regarding ifm3d since 3 days.
I can ping the camera but cannot run the tests. 
In particular I get these after fixing my ipv4 to(192.168.0.69 255.255.255.0 192.168.0.201):
$ ping 192.168.0.69
PING 192.168.0.69 (192.168.0.69) 56(84) bytes of data.
64 bytes from 192.168.0.69: icmp_seq=1 ttl=64 time=0.043 ms
64 bytes from 192.168.0.69: icmp_seq=2 ttl=64 time=0.042 ms
64 bytes from 192.168.0.69: icmp_seq=3 ttl=64 time=0.044 ms
^C
--- 192.168.0.69 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2025ms
rtt min/avg/max/mdev = 0.042/0.043/0.044/0.000 ms

But When I run the command
$ ifm3d ls
 
The output is:
ifm3d error: -100001
Lib: XMLRPC Timeout - can you `ping' the sensor?

I already followed your troubleshooting guide and add this to ~/.bashrc:
no_proxy="127.0.0.1, localhost,  192.168.0.69, 192.168.0.70, 192.168.1.* "

I tried also with Ubuntu 18.04 and Ros melodic but nothing changed.
Thanks for the attention,
Best regards

tpanzarella, 2019-07-25:
Please post the output of running: `ifconfig` and `netstat -r`. Also, is there anything special about your environment? Is this running on the metal? In a VM? In a container? **How did you change the IP of the camera if you cannot connect to it?** Are you sure it is the camera that is responding to your `ping`? When you say "after fixing my ipv4..." I just want to be clear that you are talking about the camera IP address and not your host IP address. If you changed your host IP, of course the ping will reply and of course `ifm3d` cannot connect. Please give us more information about your set up and provide the details I ask about above. Thanks.

matteofiorini, 2019-07-26:
Hello tom, thanks for the answer.
I am running Ubuntu 18.04 in a dual boot computer not virtual machine or anything else. Just installed everything yesterday to be sure of having a clean installation. These are the outputs you asked for:
$ ifconfig
enp24s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.69  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::846c:8cea:fe82:4d1  prefixlen 64  scopeid 0x20<link>
        ether 4c:cc:6a:fd:ec:42  txqueuelen 1000  (Ethernet)
        RX packets 7633  bytes 6552837 (6.5 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 5612  bytes 691081 (691.0 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 2067  bytes 193110 (193.1 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2067  bytes 193110 (193.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

$ netstat -r
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
default         _gateway        0.0.0.0         UG        0 0          0 enp24s0
link-local      0.0.0.0         255.255.0.0     U         0 0          0 enp24s0
192.168.0.0     0.0.0.0         255.255.255.0   U         0 0          0 enp24s0

When I say fixing my ipv4 I meant that I went on the setting of the wired connection and changed my ipv4 address from automatic to manual and I inserted these settings:
192.168.0.69
255.255.255.0
192.168.0.201
I usually do these passage in order to communicate through ethernet for example with a PLC and I did the same passage in Windows when I used the ifm software. 
Thanks for the attention, just tell if you need other informations.

graugans, 2019-07-26:
Hey @matteofiorini you have assigned the PC and the camera to the same IPV4 address. You have to change the IP of the PC to something else than ``192.168.0.69`` for example ``192.168.0.10``. Otherwise the PC can`t route the Ethernet traffic to the camera. The proposed setting on the ifm software are related to the camera not the PC. Another option is to change the IP of the camera. Which needs extra handling of the IP address for ``ifm3d`` https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md#connecting-the-camera

matteofiorini, 2019-07-26:
@graugans Thanks for the answer, I tried your suggestion but this doesn't work. I assigned my ipv4 to: 192.168.0.10 but now I get these outputs:
$ ping 192.168.0.69
PING 192.168.0.69 (192.168.0.69) 56(84) bytes of data.
64 bytes from 192.168.0.69: icmp_seq=1 ttl=64 time=0.275 ms
64 bytes from 192.168.0.69: icmp_seq=2 ttl=64 time=0.155 ms
64 bytes from 192.168.0.69: icmp_seq=3 ttl=64 time=0.222 ms
^C
--- 192.168.0.69 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2028ms
rtt min/avg/max/mdev = 0.155/0.217/0.275/0.050 ms

So the ping seems to work. 

$ ifm3d ls
null

 But he can't find the sensor even if the error has changed!
I tried also the test and I get these outputs:

$ catkin_make run_tests
Base path: /home/matteo/catkin/ifm3d
Source space: /home/matteo/catkin/ifm3d/src
Build space: /home/matteo/catkin/ifm3d/build
Devel space: /home/matteo/catkin/ifm3d/devel
Install space: /home/matteo/catkin/ifm3d/install
####
#### Running command: "make cmake_check_build_system" in "/home/matteo/catkin/ifm3d/build"
####
####
#### Running command: "make run_tests -j8 -l8" in "/home/matteo/catkin/ifm3d/build"
####
Built target tests
Removing test result files from '/home/matteo/catkin/ifm3d/build/test_results/ifm3d'
- removing '/home/matteo/catkin/ifm3d/build/test_results/ifm3d/rostest-test_ifm3d.xml'
- removing '/home/matteo/catkin/ifm3d/build/test_results/ifm3d/rosunit-test_camera.xml'
- removing '/home/matteo/catkin/ifm3d/build/test_results/ifm3d/nosetests-test.xml'
Built target clean_test_results_ifm3d
-- run_tests.py: execute commands
  /opt/ros/melodic/share/rostest/cmake/../../../bin/rostest --pkgdir=/home/matteo/catkin/ifm3d/src/ifm3d --package=ifm3d --results-filename test_ifm3d.xml --results-base-dir "/home/matteo/catkin/ifm3d/build/test_results" /home/matteo/catkin/ifm3d/src/ifm3d/test/ifm3d.test 
-- run_tests.py: execute commands
  "/usr/bin/cmake" -E make_directory /home/matteo/catkin/ifm3d/build/test_results/ifm3d
  /usr/bin/nosetests-2.7 -P --process-timeout=60 --where=/home/matteo/catkin/ifm3d/src/ifm3d/test --with-xunit --xunit-file=/home/matteo/catkin/ifm3d/build/test_results/ifm3d/nosetests-test.xml

----------------------------------------------------------------------
Ran 0 tests in 0.000s

OK
-- run_tests.py: verify result "/home/matteo/catkin/ifm3d/build/test_results/ifm3d/nosetests-test.xml"
... logging to /home/matteo/.ros/log/rostest-matteo-MS-7A37-3403.log
Built target _run_tests_ifm3d_nosetests_test
Built target _run_tests_ifm3d_nosetests
[ROSUNIT] Outputting test results to /home/matteo/catkin/ifm3d/build/test_results/ifm3d/rostest-test_ifm3d.xml
[ INFO] [1564123310.945932194]: Loading nodelet /ifm3d/camera of type ifm3d_ros/camera_nodelet to manager camera_standalone_nodelet with the following remappings:
[ INFO] [1564123310.951787716]: waitForService: Service [/ifm3d/camera_standalone_nodelet/load_nodelet] has not been advertised, waiting...
[ INFO] [1564123310.958186799]: Initializing nodelet with 8 worker threads.
[ INFO] [1564123310.973134544]: waitForService: Service [/ifm3d/camera_standalone_nodelet/load_nodelet] is now available.
[ INFO] [1564123311.035479666]: onInit(): /ifm3d/camera
[ INFO] [1564123311.076651781]: Camera clock will not be sync'd to system clock
[ INFO] [1564123311.076753361]: Running dtors...
[ INFO] [1564123311.076860851]: Initializing camera...
[ INFO] [1564123312.147840464]: Initializing framegrabber...
[ INFO] [1564123312.158553438]: Initializing image buffer...
[ WARN] [1564123312.659295257]: Timeout waiting for camera!
[ WARN] [1564123313.159923456]: Timeout waiting for camera!
[ WARN] [1564123313.660613265]: Timeout waiting for camera!
[ WARN] [1564123314.161072494]: Timeout waiting for camera!
[ WARN] [1564123314.661479293]: Timeout waiting for camera!
[ WARN] [1564123315.162014722]: Timeout waiting for camera!
[ WARN] [1564123315.662400981]: Timeout waiting for camera!
[ WARN] [1564123316.162940120]: Timeout waiting for camera!
[... 30 lines cut ...]

[ROSTEST]-----------------------------------------------------------------------

[ifm3d.rosunit-test_camera/test_camera][FAILURE]--------------------------------
False is not true
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/home/matteo/catkin/ifm3d/src/ifm3d/test/test_camera.py", line 117, in test_camera
    self.assertTrue(self.success_)
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 1

rostest log file is in /home/matteo/.ros/log/rostest-matteo-MS-7A37-3403.log
-- run_tests.py: verify result "/home/matteo/catkin/ifm3d/build/test_results/ifm3d/rostest-test_ifm3d.xml"
Built target _run_tests_ifm3d_rostest_test_ifm3d.test
Built target _run_tests_ifm3d_rostest
Built target _run_tests_ifm3d
Built target run_tests

matteofiorini, 2019-07-26:
Update:
After creating an application in Windows now these are the outputs I get:
$ ifm3d ls
[
  {
    "Active": true,
    "Description": "",
    "Id": 557985589,
    "Index": 1,
    "Name": "Misurazione oggetto (oggetti parallelepipedi)"
  }
]

So the sensor now works and indeed:
$ ifm3d hz
FrameGrabber running at: 3.2711 Hz
10 frames captured, over 1 runs

But If I run the code in ros and launch $roscore 
then in another terminal  $ roslaunch ifm3d camera.launch
and in another one $ roslaunch ifm3d rviz.launch
I can see in rviz all the images but in the terminal of camera.launch I get this output:

$ roslaunch ifm3d camera.launch
... logging to /home/matteo/.ros/log/f4cea386-af90-11e9-8b3e-4ccc6afdec42/roslaunch-matteo-MS-7A37-14357.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://matteo-MS-7A37:32851/

SUMMARY
========

PARAMETERS
 * /ifm3d/camera/amplitude/disable_pub_plugins: ['image_transport...
 * /ifm3d/camera/assume_sw_triggered: False
 * /ifm3d/camera/confidence/disable_pub_plugins: ['image_transport...
 * /ifm3d/camera/distance/disable_pub_plugins: ['image_transport...
 * /ifm3d/camera/frame_id_base: ifm3d/camera
 * /ifm3d/camera/frame_latency_thresh: 60.0
 * /ifm3d/camera/good_bad_pixels/disable_pub_plugins: ['image_transport...
 * /ifm3d/camera/ip: 192.168.0.69
 * /ifm3d/camera/password: 
 * /ifm3d/camera/raw_amplitude/disable_pub_plugins: ['image_transport...
 * /ifm3d/camera/schema_mask: 15
 * /ifm3d/camera/soft_off_timeout_millis: 500
 * /ifm3d/camera/soft_off_timeout_tolerance_secs: 600.0
 * /ifm3d/camera/soft_on_timeout_millis: 500
 * /ifm3d/camera/soft_on_timeout_tolerance_secs: 5.0
 * /ifm3d/camera/sync_clocks: False
 * /ifm3d/camera/timeout_millis: 500
 * /ifm3d/camera/timeout_tolerance_secs: 5.0
 * /ifm3d/camera/xmlrpc_port: 80
 * /ifm3d/camera/xyz_image/disable_pub_plugins: ['image_transport...
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /ifm3d/
    camera (nodelet/nodelet)
    camera_standalone_nodelet (nodelet/nodelet)
    camera_tf (tf2_ros/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

process[ifm3d/camera_standalone_nodelet-1]: started with pid [14372]
process[ifm3d/camera-2]: started with pid [14373]
process[ifm3d/camera_tf-3]: started with pid [14374]
[ INFO] [1564137283.812351415]: Loading nodelet /ifm3d/camera of type ifm3d_ros/camera_nodelet to manager camera_standalone_nodelet with the following remappings:
[ INFO] [1564137283.814809864]: waitForService: Service [/ifm3d/camera_standalone_nodelet/load_nodelet] has not been advertised, waiting...
[ INFO] [1564137283.827431501]: Initializing nodelet with 8 worker threads.
[ INFO] [1564137283.836005303]: waitForService: Service [/ifm3d/camera_standalone_nodelet/load_nodelet] is now available.
[ INFO] [1564137283.894540572]: onInit(): /ifm3d/camera
[ INFO] [1564137283.944523009]: Camera clock will not be sync'd to system clock
[ INFO] [1564137283.944600249]: Running dtors...
[ INFO] [1564137283.944697750]: Initializing camera...
[ INFO] [1564137284.958698310]: Initializing framegrabber...
[ INFO] [1564137284.970464464]: Initializing image buffer...
[ INFO] [1564137285.074023901]: Got unit vectors, restarting framegrabber with mask: 15
[ INFO] [1564137285.074075971]: Running dtors...
[ INFO] [1564137285.074609003]: Initializing camera...
[ INFO] [1564137286.085475926]: Initializing framegrabber...
[ INFO] [1564137286.096472427]: Initializing image buffer...
[ WARN] [1564137335.274014374]: Timeout waiting for camera!
[ WARN] [1564137342.629679463]: Timeout waiting for camera!
[ WARN] [1564137355.881443418]: Timeout waiting for camera!
[ WARN] [1564137356.973603497]: Timeout waiting for camera!

What is this timeout waiting for camera ? Thanks

tpanzarella, 2019-07-26:
This output:
```
$ ifm3d ls
null
```
Indicates to me that you are connecting to the camera, however there are no applications to list. Can you attach the output of `$ ifm3d dump`?

IIRC, you are running an O3D302 which is the so called "smart sensor" and not the O3D303 which is the "camera". @graugans can an O3D302 run a "camera application"? If the problem is simply to create a new application, you can:

```
$ echo '{"Apps":[{}]}' | ifm3d config
```

Then see if it is there with:

```
$ ifm3d ls
```

You may also want to try

```
$ ifm3d reset
```

However, I do not know what the hard reset state is on an O3D302. For an O3D303, it will give you one usable camera application with the factory default settings.

tpanzarella, 2019-07-26:
Sorry, I missed your output above that you now have this working. Regarding the timeout in ROS, can you send the output of `$ ifm3d dump`? Given that the `hz` application reported 3 Hz, you are likely at the edge of your configured frame rate and the default timeout of the ROS node. You can either, increase your frame rate on the camera itself, or, if you are happy with 3 Hz, increase the ROS node timeout settings to something like 1000 ms. You can do this on the `roslaunch` command line or edit the launch file directly. The parameter you want is: `timeout_millis`.

graugans, 2019-07-26:
@matteofiorini and @tpanzarella the smartsensor by default has rectification enabled. There is also a different default setting. I highly recommend to post the output of ``ifm3d dump``

matteofiorini, 2019-07-26:
Thanks for the answers. This is the output:
```
$ ifm3d dump
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "557985589",
        "Imager": {
          "Channel": "0",
          "ClippingBottom": "131",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
          "ClippingRight": "175",
          "ClippingTop": "0",
          "ContinuousAutoExposure": "false",
          "EnableAmplitudeCorrection": "true",
          "EnableFastFrequency": "false",
          "EnableFilterAmplitudeImage": "true",
          "EnableFilterDistanceImage": "true",
          "EnableRectificationAmplitudeImage": "true",
          "EnableRectificationDistanceImage": "true",
          "ExposureTimeList": "100;1000;5000",
          "FrameRate": "3",
          "MaxAllowedLEDFrameRate": "19.6",
          "MinimumAmplitude": "42",
          "Resolution": "0",
          "SpatialFilter": {},
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {},
          "TemporalFilterType": "0",
          "ThreeFreqMax2FLineDistPercentage": "80",
          "ThreeFreqMax3FLineDistPercentage": "80",
          "TwoFreqMaxLineDistPercentage": "80",
          "Type": "under5m_high",
          "UseSimpleBinning": "false"
        },
        "Index": "1",
        "LogicGraph": "{\n    \"IOMap\": {\n        \"OUT1\": \"USEROUT\",\n        \"OUT2\": \"NONE\",\n        \"OUT3\": \"RFT\"\n    },\n    \"blocks\": {\n        \"B00001\": {\n            \"pos\": {\n                \"x\": 50,\n                \"y\": 200\n            },\n            \"properties\": {\n            },\n            \"type\": \"MODEL_RESULT_DIMENSIONING\"\n        },\n        \"B00002\": {\n            \"pos\": {\n                \"x\": 1300,\n                \"y\": 1000\n            },\n            \"properties\": {\n            },\n            \"type\": \"STATE_DEFINITION_PASS_FAIL\"\n        },\n        \"B00003\": {\n            \"pos\": {\n                \"x\": 1050,\n                \"y\": 1000\n            },\n            \"properties\": {\n            },\n            \"type\": \"AND\"\n        },\n        \"B00004\": {\n            \"pos\": {\n                \"x\": 1300,\n                \"y\": 100\n            },\n            \"properties\": {\n                \"pulse_duration\": 0\n            },\n            \"type\": \"DIGITAL_OUT1\"\n        },\n        \"B00005\": {\n            \"pos\": {\n                \"x\": 50,\n                \"y\": 800\n            },\n            \"properties\": {\n            },\n            \"type\": \"PIN_EVENT_READY_FOR_TRIGGER\"\n        },\n        \"B00006\": {\n            \"pos\": {\n                \"x\": 1300,\n                \"y\": 800\n            },\n            \"properties\": {\n                \"pulse_duration\": 0\n            },\n            \"type\": \"DIGITAL_OUT3\"\n        }\n    },\n    \"connectors\": {\n        \"C00001\": {\n            \"dst\": \"B00002\",\n            \"dstEP\": 0,\n            \"src\": \"B00003\",\n            \"srcEP\": 0\n        },\n        \"C00002\": {\n            \"dst\": \"B00004\",\n            \"dstEP\": 0,\n            \"src\": \"B00001\",\n            \"srcEP\": 0\n        },\n        \"C00003\": {\n            \"dst\": \"B00003\",\n            \"dstEP\": 0,\n            \"src\": \"B00001\",\n            \"srcEP\": 0\n        },\n        \"C00004\": {\n            \"dst\": \"B00006\",\n            \"dstEP\": 0,\n            \"src\": \"B00005\",\n            \"srcEP\": 0\n        }\n    }\n}\n",
        "Name": "Misurazione oggetto (oggetti parallelepipedi)",
        "PcicEipResultSchema": "{\"layouter\" : \"flexible\",\"format\": { \"dataencoding\": \"binary\", \"order\": \"busdepending\" },\"elements\" : [{ \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" },{ \"type\": \"records\", \"id\": \"models\",\"elements\": [{ \"type\": \"int16\", \"id\": \"boxFound\" },{ \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } },{ \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } },{ \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } },{ \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } },{ \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } },{ \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } },{ \"type\": \"int16\", \"id\": \"yawAngle\" },{ \"type\": \"int16\", \"id\": \"qualityWidth\" },{ \"type\": \"int16\", \"id\": \"qualityHeight\" },{ \"type\": \"int16\", \"id\": \"qualityLength\" }]},{ \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" }]}",
        "PcicPnioResultSchema": "{\"layouter\" : \"flexible\",\"format\": { \"dataencoding\": \"binary\", \"order\": \"busdepending\" },\"elements\" : [{ \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" },{ \"type\": \"records\", \"id\": \"models\",\"elements\": [{ \"type\": \"int16\", \"id\": \"boxFound\" },{ \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } },{ \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } },{ \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } },{ \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } },{ \"type\": \"int1 [... cut ...]
        "Main_Application": "0.4.222"
      },
      "ifm3d_version": 1501
    }
  }
}
```
I am trying to use software triggered mode but it is not working. It works in continuous mode even if I type:
$ GLOG_minloglevel=3 roslaunch ifm3d camera.launch assume_sw_triggered:=true
I also changed the parameter in the camera.launch file but in rviz I can still see continuous images.

Another question:
Do you know which is the best software for recognition using either the point cloud image or the xyz image on ros?

Thanks

tpanzarella, 2019-07-26:
For the camera to run in software triggered mode, you need to set that parameter on the camera itself for the particular application in question. Once you have done that, you invoke the ROS node as you have done above. To that end, for your configuration, to set the active application into software triggered mode, you can:

```
$ ifm3d dump | jq '.ifm3d.Apps[0].TriggerMode="2"' | ifm3d config
```
This assumes that you have the `jq` command available. If you do not have it, you can get it with:

```
$ sudo apt install jq
```

Per your second question regarding "recognition". 1) It is under-specified; 2) it is outside of the scope of support we can offer for the sensor interface.

HTH.

tpanzarella, 2019-07-26:
I am going to mark this issue as closed as your connection problems are solved. If other issues come up, please open a new issue specific to it. Thank you.

closed 2019-07-26 by tpanzarella