ifm/ifm3d #167: Pointcloud noise near the origin of the ifm o3d303 camera

srane96, 2019-07-29:
Whenever I point my camera in the room where there are many objects, I get this kind of pyramid shaped noise near the origin.
[image].
However when I point my camera in the direction of the wall or the ceiling I do not get any outliers/noise between origin and the point clouds associated with the wall/ceiling.

I am wondering what could be the reason of this pyramid shaped noise?  I am guessing it's due to reflections from the other objects in the surrounding. Am I right ? or is there any other reason that's causing it?

tpanzarella, 2019-07-29:
This looks like ambiguity range issues (standard ToF artifact). Likely, you are using an imager setting with a single frequency and increasing it to two frequencies may fix this issue. Please post the output of `$ ifm3d dump` and we can take a look to validate. If this is the case, we can give you a setting to eliminate this artifact.

srane96, 2019-07-30:
Just clarifying. RAA is caused when 1st pulse reaches receiver after 2nd is transmitted, and sensor thinks that it is from the 2nd emitted pulse, right? So is this why you are suggesting me to increase the frequency?
Btw here is the config file.
[config.txt](https://github.com/ifm/ifm3d/files/3446828/config.txt)

tpanzarella, 2019-07-30:
You are using a single frequency. The following command should fix your issue:

```
$ ifm3d dump | jq '.ifm3d.Apps[0].Imager.Type="upto30m_moderate"' | ifm3d config
```

This assumes you have `jq`. If you do not, you can `$ sudo apt install jq` or just edit your config.txt directly and `$ ifm3d config < config.txt`.

HTH.

srane96, 2019-07-30:
Thank you so much!

commit 12e5dfa3 (2022-11-30): Merge branch 'bugfix/#167-ifm3dpy_dependency_issue' into 'main'

Add numpy as dependency for ifm3dpy while installation

Closes #167

See merge request syntron/support/csr/ifm3d/ifm3d!178
closed 2019-07-30 by srane96