ifm/ifm3d #168: basic_usage.md does not compile

lueiropablo, 2019-08-02:
When I try to compile the ifm3d / doc / basic_usage.md example, it does not compile, throwing the error: fatal error: opencv2 / core / core.hpp: File or directory does not exist.

The ifm3d Library does not contain the opencv2 module. I bet that in some recent version of ifm3d said faith module replaced by the new opencv module.

tpanzarella, 2019-08-02:
How do you try to compile it? Are your compiler flags set appropriately? Can you paste in: 1) your code; 2) how your building -- i.e., either the compiler line you are using or your `CMakeLists.txt` file if you are using CMake.

lueiropablo, 2019-08-05:
I compile and building from Geany. 

I have only installed the binaries for ifm3d, following the instructions here https://github.com/ifm/ifm3d

Should I install all the Boost, Curl, Glog dependencies ... as explained here https://github.com/ifm/ifm3d/blob/master/doc/source_build.md?

The compiler complains in this line:  #include <opencv2 / core / core.hpp>

[image]

tpanzarella, 2019-08-05:
You will need to install the opencv development files. In this case: `$ sudo apt install libopencv-core-dev`. You should be OK without Boost. You may need the Glog development files though: `$ sudo apt install libgoogle-glog-dev`.

lueiropablo, 2019-08-05:
I have added all the dependencies:
```
sudo apt-get update && apt-get install -y libboost-all-dev git jq libcurl4-openssl-dev \
                                            libgtest-dev libgoogle-glog-dev  \
                                            libxmlrpc-c++8-dev libopencv-dev \
                                            libpcl-dev libproj-dev \
                                            build-essential coreutils cmake
```
almost ready .... but again ...

`/usr/include/ifm3d/image/image_buffer.h:25:10: fatal error: pcl/point_cloud.h: No such file or directory`
[image]

pcl / point_tipes.h does not complain about the directory. The directory does exist ...
`/usr/include/pcl-1.8/pcl`

[image]

tpanzarella, 2019-08-06:
Looks like you need to add `/usr/include/pcl-1.8` to your compiler's include path, typically something like `-I/usr/include/pcl-1.8`.

As a general comment, you may want to consider using CMake as a lot of this type of configuration is taken care of for you. YMMV.

theseankelly, 2019-10-09:
Hi @lueiropablo, were you able to work through these issues or are you still having trouble compiling?

closed 2019-12-12 by theseankelly