ifm/ifm3d #171: Where i can get the example code about the Integrating PDS ?

Silent-Souls, 2019-08-26:
I found it in the link below, but I can't find additional information about this system.
https://www.ifm.com/us/en/us/robotics/software-solutions/pds/integrating-pds/integrating-Thanks!

Silent-Souls, 2019-08-28:
> Hi -- happy to help but we're currently handling PDS support internally rather than on GitHub. Could you shoot a mail to [Sean.Kelly@ifm.com](mailto:Sean.Kelly@ifm.com)?

Thank you so much,I will send you a mail soon.

closed 2019-08-28 by Silent-Souls

josephchen5, 2021-10-07:
Hello, I have an ifm TOF camera, the model is O3D303, I also cannot use PDS, how can I get support
https://www.ifm.com/us/en/us/robotics/software-solutions/pds/integrating-pds/integrating-pds
Thanks!

graugans, 2021-10-07:
Hey @josephchen5 , you need a special device for PDS support. Please get in touch with @GarrettPlace