ifm/ifm3d #173: Nexus down

faev999, 2019-09-03:
Hi,

I was trying to use the docker but apparently the Nexus repository is down

theseankelly, 2019-09-09:
Apologies for the delay. I'll have our IT department take a look right away and get back to you with the result.

theseankelly, 2019-09-10:
We're still working to diagnose and fix this issue I apologize for the inconvenience.

In case it helps, I've attached zips of the debians for the four apt flavors we offer (16.04/18.04 ros/no-ros) to our latest release (0.15.1)

https://github.com/ifm/ifm3d/releases/tag/v0.15.1

theseankelly, 2019-09-11:
All, our IT department discovered a hardware problem with the NIC associated with our nexus server. The issue should be resolved; please reactivate this issue if you are still encountering issues.

Again, apologies, and thanks for your patience.

closed 2019-09-11 by theseankelly