ifm/ifm3d #175: O3D303: Sensor Timeout, Cannot Retrieve Frame

bryanIkeNextStep, 2019-09-09:
Hello,

Our team is attempting to understand what is causing an O3D303's imager to become locked up.  We're unable to grab an image using the FrameGrabber and consequently receiving a sensor timeout.  When logging into the sensor via Vision Assistant, we're unable to view a live stream from the sensor (see image below).

[image]

We're running IFM3D 0.8.3 and using firmware 1.30.4123.  Below is output when running the IFM3D Trace command through Windows' command line.

C:\ifm3d\install\bin>ifm3d trace --ip=192.168.0.167 --limit=100
T: 187964.5280, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 187966.7194, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 187966.7199, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 144
2 has not been locked ?!
T: 187966.7203, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem
T: 187966.9278, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 187969.1198, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 187969.1204, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 144
4 has not been locked ?!
T: 187969.1207, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem
T: 187969.3278, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 187971.5198, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 187971.5204, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 144
6 has not been locked ?!
T: 187971.5207, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem
T: 187971.7869, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 187973.9172, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 187973.9177, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 144
8 has not been locked ?!
T: 187973.9180, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem
T: 187974.1277, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 187976.3197, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 187976.3203, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 145
0 has not been locked ?!
T: 187976.3206, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem
T: 187976.5278, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 187978.7184, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 187978.7189, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 145
2 has not been locked ?!
T: 187978.7192, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem
T: 187978.9278, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 187981.1204, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 187981.1209, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 145
4 has not been locked ?!
T: 187981.1212, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem
T: 187981.3278, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 187983.5179, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 187983.5184, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 145
6 has not been locked ?!
T: 187983.5187, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem
T: 187983.7279, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 187985.9176, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 187985.9181, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 145
[... 91 lines cut ...]
4 has not been locked ?!
T: 188017.1182, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem
T: 188017.3277, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 188019.5193, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 188019.5198, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 148
6 has not been locked ?!
T: 188019.5202, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem
T: 188019.7278, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 188021.9176, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 188021.9181, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 148
8 has not been locked ?!
T: 188021.9184, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem
T: 188022.1278, P: 436,  C: ERROR,  S: imager,  M: virtual void ifm::RawDataIntegrityCheck::execute(ifm::Phase&) "Sequence index in pseudo d
ata (2) is not equal to expected sequence index (1)"
T: 188024.3176, P: 436,  C: ERROR,  S: imager,  M: void ifm::TriggerStateMachine::onFrameCompletionTimeout(): sequence did not complete afte
r waiting 2000 ms -> cancel acquisition
T: 188024.3181, P: 436,  C: WARN,  S: imager,  M: bool ifm::FilesystemRingBuffer::producerUnlock(ifm::FilesystemRingBuffer::Item&): Slot 149
0 has not been locked ?!
T: 188024.3185, P: 436,  C: WARN,  S: imager,  M: void ifm::FilesystemRingBuffer::releaseSlot(const QString&): Could not unlockItem

Please let us know if there is anything additional information we can provide to assist in troubleshooting this issue.  Thank you for your time.

Arag24, 2019-09-09:
@bryanIkeNextStep  : looking at the the Vision Assistant [V]A screenshot, it looks like you are in Process Interface trigger.
Is it expected ? I mean, do you send any PCIC command (TCP trigger) to make snapshot, or are you expected a "live" freerun camera?
Could you please change this inside VA (for diagnostics or your purpose directly) ? just switch to Freerun mode and save.

You can also have a look / change this mode within ifm3d:
[https://github.com/ifm/ifm3d/blob/master/doc/configuring.md](https://github.com/ifm/ifm3d/blob/master/doc/configuring.md)

Please have a look to 

> `"Triggermode"`

 parameter.

They are defined here  : [camera.h](https://github.com/ifm/ifm3d/blob/05762f59c91f8b59375d602a0c9c5c5705b88679/modules/camera/include/ifm3d/camera/camera.h)
```
/**
* Image acquisition trigger modes
*/
    enum class trigger_mode : int { FREE_RUN = 1, SW = 2 };
```

bryanIkeNextStep, 2019-09-09:
Yes, we're expecting to be in Process Interface mode.  We send a PCIC command to grab a snapshot.

inbangsa, 2019-09-10:
@bryanIkeNextStep  you might be missing a fg->SWTrigger(); call before fg->WaitForFrame(...) call. 
as camera is in trigger mode camera is waiting for trigger to send the data. and fg->SWTrigger() call will send the trigger to the camera.

@bryanIkeNextStep  can you share the  code snippet with which you are trying to grab the image.  from IVA it seems your device in trigger mode but for more insight please  send the out of dump command from ifm3d tool. 
if  ifm3d is installed, following will create  dump of the camera configuration.
```
> ifm3d --ip=192.168.0.70 dump 
```

bryanIkeNextStep, 2019-09-10:
@inbangsa below is the source code for how we're grabbing the camera's frame.

std::vector<float> xyz3Dcoordinate;
cameras.at(index).cam = ifm3d::Camera::MakeShared(cameras.at(index).ip);
cameras.at(index).img = std::make_shared<ifm3d::ImageBuffer>();
cameras.at(index).fg = std::make_shared<ifm3d::FrameGrabber>(cameras.at(index).cam);

cameras.at(index).fg->SWTrigger();//TRIGGER
if (!cameras.at(index).fg->WaitForFrame(cameras.at(index).img.get(), 3000)) //WAIT FOR FRAME
{
      std::cerr << "Timeout waiting for camera(" << index << ") !" << std::endl; trayLogging << 
      "FS_Basecode: Camera " << index << " Sensor Timeout, Please Scan Again" << std::endl;
      return 405;
}
else
{
      trayLogging << "FS_Basecode GetFrame: Camera " << index << " Frame Captured" << std::endl;
      cameras.at(index).fullCloud = cameras.at(index).img->Cloud();
}

Sleep(100);

I rebooted the sensor and we're now able to grab a frame.  However, I'm curious what behavior caused the sensor to end up in the previously mentioned state.  Below is output from the sensor using the dump command after rebooting (in other words, the sensor is in a functional state).

C:\ifm3d\install\bin>ifm3d --ip=192.168.0.167 dump
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "476707713",
        "Imager": {
          "AutoExposureMaxExposureTime": "10000",
          "AutoExposureReferencePointX": "88",
          "AutoExposureReferencePointY": "66",
          "AutoExposureReferenceROI": "{\"ROIs\":[{\"id\":0,\"group\":0,\"type\":\"Rect\",\"width\":130,\"height\":100,\"angle\":0,\"center_
x\":88,\"center_y\":66}]}",
          "AutoExposureReferenceType": "0",
          "Channel": "0",
          "ClippingBottom": "131",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3
.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
          "ClippingRight": "175",
          "ClippingTop": "0",
          "ContinuousAutoExposure": "false",
          "ContinuousUserFrameCalibration": "false",
          "EnableAmplitudeCorrection": "true",
          "EnableFastFrequency": "false",
          "EnableFilterAmplitudeImage": "false",
          "EnableFilterDistanceImage": "true",
          "EnableRectificationAmplitudeImage": "true",
          "EnableRectificationDistanceImage": "true",
          "ExposureTime": "10000",
          "ExposureTimeList": "5000;10000",
          "ExposureTimeRatio": "2",
          "FlyingPixelEraser": "3",
          "FrameRate": "5",
          "MaxAllowedLEDFrameRate": "4.1",
          "MinimumAmplitude": "42",
          "Resolution": "0",
          "SpatialFilter": {},
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {},
          "TemporalFilterType": "0",
          "ThreeFreqMax2FLineDistPercentage": "80",
          "ThreeFreqMax3FLineDistPercentage": "80",
          "TwoFreqMaxLineDistPercentage": "80",
          "Type": "upto30m_moderate",
[... 95 lines cut ...]
      "Date": "Tue Sep 10 09:43:13 2019",
      "HWInfo": {
        "Connector": "#!02_A300_C40_03831786_008023176",
        "Diagnose": "#!02_D322_C34_03995634_008026824",
        "Frontend": "#!02_F342_C34_19_00057_008023607",
        "Illumination": "#!02_I300_001_03978946_008001175",
        "MACAddress": "00:02:01:41:0A:88",
        "Mainboard": "#!02_M381_C41_04054010_008023690",
        "MiraSerial": "04f0-59ae-0f13-06b0"
      },
      "SWVersion": {
        "Algorithm_Version": "2.2.6",
        "Calibration_Device": "00:02:01:41:0a:88",
        "Calibration_Version": "0.9.0",
        "Diagnostic_Controller": "v1.0.30-73a66baf54-dirty",
        "IFM_Recovery": "unversioned",
        "IFM_Software": "1.30.4123",
        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00048-gc8112a91aeb2-dirty (jenkins@dettlx190) (gcc version 4.9.2 (GCC) ) #1 SMP
PREEMPT RT Wed Mar 27 18:25:10 CET 2019",
        "Main_Application": "1.0.30"
      },
      "ifm3d_version": 803
    }
  }
}

graugans, 2019-09-11:
@bryanIkeNextStep I am glad the camera is working now. @inbangsa pointed out that your setting can cause overheat on the camera. Depending what you did during your test it could be the cause that the camera went into an over temperature state. Based on the trace log you have send in the initial issue this could be a possibility. You can find more information about possible ``WaitForFrame`` timeouts in our [troubleshooting guide](https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md#waitforframe-timeout)

If you trigger the cameras in a tight loop it might be better to reuse the objects and not recreate them in each iteration.

Based on the code snippet I do expect you run multiple cameras in a row. You'll might find this example useful ([ifm3d-examples/ex-multiple-camera-parallel-grabber](https://github.com/inbangsa/ifm3d-examples/tree/feature/ex-multiple-camera-parallel-grabber/multi_camera_parallel_grabber)). This is a work in progress Pull-Request, but hopefully gives you an idea how to trigger multiple cameras in a fast way.

It might also worth to take a look at the examples repository: [ifm3d-examples](https://github.com/ifm/ifm3d-examples).

closed 2019-09-19 by bryanIkeNextStep