ifm/ifm3d #177: FrameGrabber WaitForFrame returns true on Windows

lpc921, 2019-09-27:
I compiled ifm3d in Windows from source by following the procedure. The WaitForFrame() function sometimes returns true when a frame received is incomplete. It only happen when timeout_millis is set to a small number.

Output:
```
Exception thrown at 0x00007FFE0DD55299 in ifm3d-o3d-viewer.exe: Microsoft C++ exception: ifm3d::error_t at memory location 0x000000702E2FFB80.
```

Code to reproduce the issue:
```
#include <iostream>
#include <ifm3d/camera.h>
#include <ifm3d/fg.h>
#include <ifm3d/simpleimage.h>

int main(int argc, char *argv[]) {
	auto cam = ifm3d::Camera::MakeShared();

	ifm3d::SimpleImageBuffer::Ptr img = std::make_shared<ifm3d::SimpleImageBuffer>();
	ifm3d::FrameGrabber::Ptr fg =
		std::make_shared<ifm3d::FrameGrabber>(
			cam, ifm3d::IMG_AMP | ifm3d::IMG_CART | ifm3d::IMG_RDIS);

	while (!fg->WaitForFrame(img.get(), 1));

	std::cout << "Got image" << img->Bytes().size() << std::endl;

	return 0;
}
```

graugans, 2019-09-27:
@lpc921 this highly depends on the used Software Version on the camera, your configuration and the environment. Please check our [troubleshooting](https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md) guide for more information. I also recommend to use the latest software version on the camera.

lpc921, 2019-09-27:
Thanks for your quick replay. I am using the latest firmware on the camera and the recommended versions of software in the Readme.

The Organize() function is called by default when the function returns true, and hence exception is thrown sometimes. Other times it just produces a shaky/broken visual.

I will do more digging next week and let you know.

lpc921, 2019-09-30:
I've traced the issue to this line. The wait_for function sometimes returns timeout normally, but sometimes does not return timeout when front_buffer_ has size zero.
https://github.com/ifm/ifm3d/blob/05762f59c91f8b59375d602a0c9c5c5705b88679/modules/framegrabber/src/libifm3d_framegrabber/frame_grabber_impl.hpp#L339-L341

It might be related to a Visual C++ 2017 bug: [https://developercommunity.visualstudio.com/content/problem/193041/the-visual-c-2017-condition-variable-implementatio.html](url)

I compiled ifm3d with VS2019 but the issue persists.

The problem does not happen on Linux (Ubuntu).

I added a condition to check for empty buffer and it seems to prevent the problem from happening. I will open a pull request.

theseankelly, 2019-09-30:
@lpc921 thanks for digging in and for providing a PR! I'd like to spend some time digging in to the issue (and that Visual C++ bug you referenced) before proceeding with your PR, if that's alright.

lpc921, 2019-09-30:
I totally agree with you. My PR doesn't get to the bottom of the issue. I still get frames jumping backwards and forwards when timeout_millis is set to a small number. My C++ skill is limited so I appreciate any help from you.

theseankelly, 2019-10-04:
Hey @lpc921 -- I've finally gotten to this item today. I'm certainly able to reproduce your issue (thanks very much for the minimal example) and I am chasing down some solid leads.

Curious if you could explain more about "I still get frames jumping backwards and forwards" -- do you mean you're dropping frames, getting an inconsistent frame rate, or frames out of order? How are you determining this?

Finally -- while indeed you have pointed out a real bug that I shall fix, I'd also suggest that your implementation isn't optimal. The timeout parameter on `WaitForFrame` is intended for usage with software triggering (where triggering happens as a separate API call) and with fairly large values (our samples all use 1s or more). By providing a very small timeout value, you're interrupting the CPU quite often and not taking advantage of the underlying condition variable synchronization. If you pass a value of '0' to the timeout, `WaitForFrame` will block indefinitely and the thread will sleep until a frame is available. Does that fit better with your use case?

lpc921, 2019-10-06:
Hi Sean, thanks for looking into this. The frame jumping forward and backward can be seen when I move something quickly under the camera. Say normal you get frame 1,2,3. But sometimes when the problem occur I get 1,2,1,2,3. For me it happens quite rarely, maybe 1 out of 10 times.

Thanks for your suggestion and explaination on the implementation. I changed my code recently to run the camera io and visualisation in 2 threads. I will make sure to use a relatively large timeout value.

graugans, 2019-10-07:
@lpc921 I a not sure if you are aware of the fact that the [ImageBuffer](https://github.com/ifm/ifm3d-examples/pull/11#discussion_r288636147) is not thread safe. This might causes your jumps in the two thread use-case.

lpc921, 2019-10-07:
The frame jump back and forth problem happens to me with the standard usage of the library, without additional threads. Even with long timeout, if I disconnect the Ethernet or power to the camera, there is a chance that this jump happens before an exception is thrown. I can take a deeper look once I am back at my PC.

In my implementation, in one thread the data is copied from ImageBuffer to a pcl pointcloud using the pcl::copyPointCloud function (guarded by a mutex). Then in another threaded I run the pcl visualizer. I haven't observed any issue so far but please let me know if this could cause a problem. I am on mobile and sorry for the brevity. I can provide more details when I can access my PC.

theseankelly, 2019-10-07:
The root cause of this issue is that the `wait` and `wait_for` calls here:
https://github.com/ifm/ifm3d/blob/05762f59c91f8b59375d602a0c9c5c5705b88679/modules/framegrabber/src/libifm3d_framegrabber/frame_grabber_impl.hpp#L333-L347
Do not handle [spurious wakeups](https://en.wikipedia.org/wiki/Spurious_wakeup) -- that is, when the waiting thread unblocks for reasons other than timeout or being signaled.

This hasn't been much of an issue to date because spurious wakeups are quite rare. However, the Visual C++ 2017 bug you linked above exacerbates the issue. Essentially, the bug in Visual C++ 2017 causes false spurious wakeups when the wait_for value is very small, which triggers the unhandled code path in ifm3d::FrameGrabber.

When a spurious wakeup occurs in the code today, one of the previously delivered buffers is delivered again. Depending on your usage of the copy_buff flag, this can manifest as one of:
1) The previous frame (frame n-1)
2) Frame n-2
3) If it happens on one of the first frames, you end up with a pointer to an empty buffer, triggering your original crash (again, this is far more likely on Windows due to the Visual C++ issue).

So, even if you work around 3 (as you did in your PR) you still see instances of 1/2 -- your 'jumping around' symptom.

A fix is forthcoming, but in the meantime I've verified that a longer timeout (>=500ms) gets around the Visual Studio C++ 2017 bug, and you should not see crashes any longer. Hopefully that can unblock your progress?

lpc921, 2019-10-08:
Hi Sean, thanks for your detailed explanation and all the help so far. I also thought that the frame jumping is a related issues. Looking forward to the fix. I've closed my PR. It's not blocking my progress so take your time.

theseankelly, 2019-10-09:
Hey @lpc921 I just submitted a PR which fixes this -- I hope to have it reviewed and merged by tomorrow. This will prevent the empty buffers and out-of-order buffers.

My previous recommendation still holds, though -- recommended usage involves a sufficiently large timeout (at least a frame period). Otherwise you'll be utilizing CPU time on timeouts and possibly missing frames (refer to the race condition mentioned in the commit message on my PR).

I'll close this issue out once the PR goes through!

commit 202b72c3 (2022-11-22): Merge branch 'python_311_ubuntu_2204' into 'main'

Add python 3.11 & ubuntu 22.04 support

Closes #177 and #117

See merge request syntron/support/csr/ifm3d/ifm3d!182
closed 2019-10-11 by theseankelly