ifm/ifm3d #179: Build under windows ignores -std=c++14

bashar-tom, 2019-10-09:
Hi, 
I have tried to build the Library in windows and followed every detail in the windows.md
I have compiled it just with the OpenCV Module and the build process gives no error, just some warnings. 
The problem is whenever I use the library in my project it throws the same failure: 
cl : Command line warning D9002 : ignoring unknown option '-std=c++14' which causes a lot of syntax errors. 
I have double-checked it and it seems coming from the ifm library.

I have also tried to build it with VS 2019 with Boost 1_71_0 msv14.2-64 and build the other library using the appropriate versions, which leads to seeing this error in the building process itself. 
Can you please help me with this issue?
For more details, I will gladly provide them.

graugans, 2019-10-09:
Hey @bashar-tom,

as far as I  know the Windows compilers do not use ``-std=c++14`` they have a [different way](https://docs.microsoft.com/de-de/cpp/build/reference/std-specify-language-standard-version?view=vs-2019) of defining the C++ standard. From the text linked I would expect that the default is C++14 already. Could you please be so kind and paste the error output here?

bashar-tom, 2019-10-09:
Here are the default options in the CMakeLists.txt provided by the ifm library.
[https://github.com/ifm/ifm3d/blob/05762f59c91f8b59375d602a0c9c5c5705b88679/CMakeLists.txt#L46](url)
Also, I have used the same way to define the c++ standard. I also have tried to define it in the same way stated in the link you gave me.

Also to mention, that I have the same failure in the build process when building with VS 2019

Leo16112, 2019-10-09:
Hello Graugans,
this is my error output:

cl : befehlszeile warning D9002: Unbekannte Option "-std=c++11" wird ignoriert. [C:\ifm3d\ifm3d\build\modules\camera\sr
c\libifm3d_camera\ifm3d_camera.vcxproj]
C:\ifm3d\ifm3d\modules\camera\src\libifm3d_camera\camera.cpp(640,13): warning C4018: ">": Konflikt zwischen "signed" un
d "unsigned" [C:\ifm3d\ifm3d\build\modules\camera\src\libifm3d_camera\ifm3d_camera.vcxproj]
C:\ifm3d\ifm3d\modules\camera\src\libifm3d_camera\camera.cpp(646,17): warning C4018: ">": Konflikt zwischen "signed" un
d "unsigned" [C:\ifm3d\ifm3d\build\modules\camera\src\libifm3d_camera\ifm3d_camera.vcxproj]

I did compilation according manual with difference, that I use boost 1_71_0, opencv_3.4.7, VisualStudio Professional 2019, My CMake is 3.14.6 ( with 3.15.4 it was the same)

> Hey @bashar-tom,
>
> as far as I know the Windows compilers do not use `-std=c++14` they have a [different [...]

graugans, 2019-10-09:
Hi @Leo16112,

In the original issue @bashar-tom mentioned he has successfully build the lib. The error output indicates that this is not the case. Is it possible for you to check if this works on VS2017?

Leo16112, 2019-10-09:
> 
> 
> Hi @Leo16112,
> 
> In the original issue @bashar-tom mentioned he has successfully build the lib. The error output indicates that this is not the case. Is it possible for you to check if this works on VS2017?

He could compile the library, but then got the same warning with c++ standart and errors with compilation of our programm. I will install VS2017 now and try.

graugans, 2019-10-09:
Hi @Leo16112,

how do you include the lib in your project? 
You can take a look at our [ifm3d-examples](https://github.com/ifm/ifm3d-examples) project for reference
```cmake
find_package(ifm3d 0.11.0 CONFIG
    REQUIRED COMPONENTS camera framegrabber
    OPTIONAL_COMPONENTS image pcicclient
    )
```

Leo16112, 2019-10-09:
Hello again @graugans 

thank you for the advice about vs17. I also used exactly same versions of other libs provided in manual, and i could compile ifm3d. But then i got problems with my application. Compiler did not accept "or", "and" and needed everywhere header "string" if i used "getline".  After i changed code, there is linking problem with opencv arised. I tried to use opencv, that was compiled with IFM-Manual.  After I used another one prebuild opencv version, evrething worked. I could even change my code back from && to "and".  I am going to try to get image from the camera tomorrow and if it works, i consider the issue closed. 

Kind Regards,
Anton.

theseankelly, 2019-10-09:
Hi @Leo16112 and @bashar-tom,

I have a whole bunch of questions about how you're including/linking against ifm3d, how your VS project is configured so forth. Could you provide a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) Visual Studio solution that reproduces the issues?

Please ensure it's a basic example depending only on ifm3d (and dependencies of ifm3d) that compiles and demonstrates the warning and the syntax errors you speak of? Don't worry about relative paths to libraries in the project; I can fix them up on my machine to reproduce your issues.

Best,
Sean

Leo16112, 2019-10-10:
Hello @theseankelly  and @graugans ,
after we finish the current project, we make an example. We are running out of time unfortunitely.

We have another problem today.  The programm crushes on this line:
` if (!this->fg->WaitForFrame(this->img.get(), 10000)) {`
Evrething was compiled with msvs 2017, but our software with msvs 2019.

We are using opencv container. On linux evrething works perfectly. If we start IFMAsssistent, we also can get stream. Firewall was switched off in both cases.

Regards,

bashar-tom, 2019-10-10:
Hello @theseankelly and @graugans ,
I have used the following example:
`#include <iostream>
#include <memory>
#include <opencv2/core/core.hpp>
#include <ifm3d/camera.h>
#include <ifm3d/fg.h>
#include <ifm3d/opencv.h>

int main(int argc, const char **argv)
{
    auto cam = ifm3d::Camera::MakeShared();
    auto fg = std::make_shared<ifm3d::FrameGrabber>(cam);
    auto im = std::make_shared<ifm3d::OpenCVBuffer>();

    cv::Mat amp;
    cv::Mat xyz;

    while (true)
    {
        //if (! fg->WaitForFrame(im.get(), 1000))
        //{
         //   std::cout<< "Timeout waiting for camera!" << std::endl;
          //  return -1;
        //}
        im.get();
        amp = im->AmplitudeImage();
        //xyz = im->XYZImage();
        std::cout<<amp<<std::endl;
        std::cout<<"after amp , xyz"<<std::endl;
        // now do something with `amp` and `xyz`
    }

    return 0;
}`

cmakelists file:
`cmake_minimum_required(VERSION 3.14)
project(test)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} /std:c++11")
set(OpenCV_DIR "D:\\Workspace\\StartUp\\windows\\opencv\\build\\x64\\vc15\\lib")
find_package(OpenCV REQUIRED)# COMPONENTS core imgproc highgui)
#include_directories(${OpenCV_INCLUDE_DIRS})
##############################################################
## finding the ifm3d lib.
##############################################################
find_package(ifm3d
        REQUIRED COMPONENTS camera framegrabber
       # OPTIONAL_COMPONENTS image pcicclient
        )

include_directories(${OpenCV_INCLUDE_DIRS})
add_executable(test main.cpp)
target_link_libraries(test ifm3d::camera
        ifm3d::framegrabber
        ${OpenCV_LIBS})`

which only meant to test the im.get() function with the OpenCVBuffer. The im.get() always returns an empty array.

Otherwise, when building the project, the compiler still giving the following warning cl : Command line warning D9002 : ignoring unknown option '/std:c++11'
cl : Command line warning D9002 : ignoring unknown option '-std=c++14'
Thanks in advance

graugans, 2019-10-10:
Hey @bashar-tom,

the ``ifm3d::OpenCVBuffer`` is just a buffer as its name imposes.  The call ``fg->WaitForFrame(im.get(), 1000))`` fills this buffer with the latest data from the camera.

The ifm3d::FrameGrabber handles the communication with the camera and stores the data into the ``ImageBuffer`` if there is no call to ``WaitForFrame`` it will never filled with real data.

The call [im.get()](https://de.cppreference.com/w/cpp/memory/shared_ptr/get) is just returning the pointer from the ``std::shared_ptr`` it does not include any interaction with the camera.

Leo16112, 2019-10-10:
Hello @graugans ,

Bashar commented out the place, where we have the truble.
The problem is right now in my and in Bashars case, that the program crashes on this line: `(!this->fg->WaitForFrame(this->img.get(), 10000))` by me or
`if (! fg->WaitForFrame(im.get(), 1000))` by Bashar.

inbangsa, 2019-10-10:
Hello @Leo16112  @bashar-tom 

I am able to run your example code  on my test machine which is configured with VS2017 and all recommended libraries. I have uncommented  fg->WaitForFrame call for the correct execution. 

can you use ifm3d tools and send us dump command output this will help us to understand the camera configuration.  tools [doc](https://github.com/ifm/ifm3d/blob/master/doc/cmdline.md)

graugans, 2019-10-10:
Hi @Leo16112 and @bashar-tom,

in c++ mixing compiler versions is kind of dangerous. There is no standard how an object is represented in Memory. This layout can change from version to version. My recommendation is if you are using C++ compile everything with the same Toolchain (Compiler). Maybe this because I am old and I had some real bad experience in the past.

Somewhere you mentioned you are using two different OpenCV libs. The one you compiled ifm3d with and the other one which is pre compiled.

Leo16112, 2019-10-10:
Hello @inbangsa ,

that is my dump
`{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "1095301150",
        "Imager": {
          "AutoExposureMaxExposureTime": "10000",
          "AutoExposureReferencePointX": "88",
          "AutoExposureReferencePointY": "66",
          "AutoExposureReferenceROI": "{\"ROIs\":[{\"id\":0,\"group\":0,\"type\":\"Rect\",\"width\":130,\"height\":100,\"angle\":0,\"center_x\":88,\"center_y\":66}]}",
          "AutoExposureReferenceType": "0",
          "Channel": "0",
          "ClippingBottom": "131",
          "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
          "ClippingLeft": "0",
          "ClippingRight": "175",
          "ClippingTop": "0",
          "ContinuousAutoExposure": "false",
          "ContinuousUserFrameCalibration": "false",
          "EnableAmplitudeCorrection": "true",
          "EnableFastFrequency": "false",
          "EnableFilterAmplitudeImage": "true",
          "EnableFilterDistanceImage": "true",
          "EnableRectificationAmplitudeImage": "false",
          "EnableRectificationDistanceImage": "false",
          "ExposureTime": "5000",
          "ExposureTimeList": "125;5000",
          "ExposureTimeRatio": "40",
          "FlyingPixelEraser": "0",
          "FrameRate": "5",
          "MaxAllowedLEDFrameRate": "23.2",
          "MinimumAmplitude": "42",
          "Resolution": "0",
          "SpatialFilter": {},
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {},
          "TemporalFilterType": "0",
          "ThreeFreqMax2FLineDistPercentage": "80",
          "ThreeFreqMax3FLineDistPercentage": "80",
          "TwoFreqMaxLineDistPercentage": "80",
          "Type": "under5m_moderate",
          "UseSimpleBinning": "false"
        },
        "Index": "1",
        "LogicGraph": "{\"IOMap\": {\"OUT1\": \"RFT\",\"OUT2\": \"AQUFIN\"},\"blocks\": {\"B00001\": {\"pos\": {\"x\": 200,\"y\": 200},\"properties\": {},\"type\": \"PIN_EVENT_IMAGE_ACQUISITION_FINISHED\"},\"B00002\": {\"pos\": {\"x\": 200,\"y\": 75},\"properties\": {},\"type\": \"PIN_EVENT_READY_FOR_TRIGGER\"},\"B00003\": {\"pos\": {\"x\": 600,\"y\": 75},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT1\"},\"B00005\": {\"pos\": {\"x\": 600,\"y\": 200},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT2\"}},\"connectors\": {\"C00000\": {\"dst\": \"B00003\",\"dstEP\": 0,\"src\": \"B00002\",\"srcEP\": 0},\"C00001\": {\"dst\": \"B00005\",\"dstEP\": 0,\"src\": \"B00001\",\"srcEP\": 0}}}",
        "Name": "Sample Application",
        "PcicEipResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicPnioResultSchema": "{\"layouter\" : \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
        "PcicTcpResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"ascii\" }, \"elements\": [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"blob\", \"id\": \"normalized_amplitude_image\" }, { \"type\": \"blob\", \"id\": \"distance_image\" }, { \"type\": \"blob\", \"id\": \"x_image\" }, { \"type\": \"blob\", \"id\": \"y_image\" }, { \"type\": \"blob\", \"id\": \"z_image\" }, { \"type\": \"blob\", \"id\": \"confidence_image\" }, { \"type\": \"blob\", \"id\": \"diagnostic_data\" }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
[... cut ...]
        "Algorithm_Version": "2.2.6",
        "Calibration_Device": "00:02:01:41:38:cb",
        "Calibration_Version": "1.0.2",
        "Diagnostic_Controller": "v1.0.30-73a66baf54-dirty",
        "IFM_Recovery": "unversioned",
        "IFM_Software": "1.30.4123",
        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00048-gc8112a91aeb2-dirty (jenkins@dettlx190) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Wed Mar 27 18:25:10 CET 2019",
        "Main_Application": "1.0.30"
      },
      "ifm3d_version": 1501
    }
  }
}`

Leo16112, 2019-10-11:
Hallo @theseankelly ,
unfortunitely not. The original warning about str 14, and error that makes to change all "and" and "or" to && and || still there. But after first compilation I can change them back to and/or and it compiles.

theseankelly, 2019-10-11:
Hi @Leo16112 -- I have determined your syntax issues are not related to the ifm3d library and reproduce in simple MSVC projects.

The "and"/"or" syntax (rarely used alternate tokens called 'digraphs') are not supported by MSVC unless you use the '/Za' flag to disable MSVC C++ language extensions.

Consider this project which does not depend on or use `ifm3d` in any way:

CMakeLists.txt:
```cmake
cmake_minimum_required(VERSION 3.14)
project(test)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} /std:c++14")
add_executable(test main.cpp)
```

main.cpp (I use rand to ensure optimization doesn't remove the `if` block)
```cpp
#include <iostream>
#include <cstdlib>
int main()
{
  int x = rand();
  int y = rand();
  if (x > 20 and y < 100)
  {
    std::cout << "Condition met" << std::endl;
  }
  return 0;
}
```
Command line:
```
$ md build && cd build
$ cmake -G "Visual Studio 16 2019 Win64" ..
$ cmake --build . --config Release
```

This fails to compile and complains about an undeclared identifier `and`.

Two things fix this:
1) Changing `and` to &&
2) Adding /Za to the CMAKE_CXX_FLAGS property (`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} /std:c++14 /Za"`)

I'd strongly suggest sanitizing the digraphs from your codebase as they're not widely known or used and may be on a path towards deprecation in the standard. Microsoft's documentation also recommends against the /Za flag for C++ codebases.

Some background:
https://en.cppreference.com/w/cpp/language/operator_alternative
https://github.com/PixarAnimationStudios/OpenSubdiv/issues/779
https://docs.microsoft.com/en-us/cpp/build/reference/za-ze-disable-language-extensions?view=vs-2019

Leo16112, 2019-12-02:
@theseankelly  Hello Sean. 
Sorry for late reply. 
Big thanks to you for your explanation. We changed everething to && || and added (set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} /std:c++14 /Za") to code.
It still ignors c++14, but evrething is compiled and works pretty. Thank you for help, I consider issue closed, but do note have rights for it.
Kind Regards,
Anton.

theseankelly, 2019-12-02:
Thanks for the follow-up, and glad to hear it's working!

closed 2019-12-02 by theseankelly

inbangsa, 2019-12-03:
I think with cmake 3.8 and later we can use 

set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
And for correct linking you will also need 

set(CMAKE_CXX_EXTENSIONS OFF)

Above three flags should link to C++14 standard.