ifm/ifm3d #183: Synchronization service call has undefined behavior

JD-ETH, 2019-10-28:
We observe strong non-deterministic behavior when trying to trigger the synchronization. The resulting time delay is much more than expected (~0.7 s instead of ~0.1s). 

 **Hardware**: 
Linux Ubuntu 18.04, ifm o3x100. 

**On sensor configuration**:
The relevant setting here is _SynchronizationActivated_, and is set to "false". This conflicts otherwise with the software disable. _sync_clocks_ flag is enabled in the ros launch file.  

**Problem description**
The synchronization seems to give reasonable result when the sensor boots up. however, when for some reason the `ifm3d_ros` interface is relaunched, or when the `syncClocks` is called as a service, the time stamp starts to jump around. We examine the delay between system time (ROS time) and sensor time (message time) using the ros diagnostics tool. Videos and small descriptions are attached under the following [link](https://drive.google.com/drive/folders/1ge3LaaDVePxYGK5orBM7tck2aVVola5d?usp=sharing). 

The issue is posted here as it more closely relates to the underlying ifm3d library then the ros wrapper.

JD-ETH, 2019-10-30:
@theseankelly thanks for the quick response!

We tested the hardware synchronization first, and couldn't get it to work reliably. I'm describing our procedure here and please let me know if anything's wrong: 
1. setup chrony for ntp server on local network at the host. 
2. set configuration to use _SynchronizationActivated_, wait until 5 tries. 
We then observe the sensor syncing status via `ifm3d dump`. When booting sensor and host at the same time, the synchronization doesn't seem to run and status keep showing _server not available_.
If however `ifm3d reboot` is called, and one wait for several minutes, essentially the result of `ifm3d dump` shows synchronized status. 

We are sure the ntp server is available, and the host has access to internet. Could you give some suggestions? 

I will be closing this thread afterwards, as the nondeterminism seems inevitable and shouldn't be viewed as an issue.

graugans, 2019-10-31:
As I described in #96 the NTP server needs to be available when the camera is booting up. Otherwise the camera does not sync due to the risk of time jump effects. I wonder why you do have a sync time of several minutes. This is not intended behaviour. I can check this on monday when I am in the office.

JD-ETH, 2019-11-01:
I have found a fix to the problem. 

Indeed, the syncing fails if the ntp server is unavailable. Which is to be expected on a mobile robot platform by the way, how would your PC boots up earlier than your sensor? 

The interesting and unexpected of the ifms is however, when rebooting the sensor after the NTP server is available, it doesn't attempt or does not immediately attempt to reconnect with the server. Seems like a "number of sync tries" was somehow saved and the syncing was skipped. Sometimes I observe syncing however after a long time after the reboot, which is unreliable for our purposes. 

I can confirm however, by deactivating and reactivating the SynchronizationActivated flag the ifms can immediately be synchronized with the server. The command I ended up using was: 

`ifm3d dump | jq '.ifm3d.time.SynchronizationActivated="false"' | ifm3d config
ifm3d dump | jq '.ifm3d.time.SynchronizationActivated="true"' | ifm3d config
`
For all our ifm sensors. 

Feel free to close this MR, as I have found my fix. But there is obviously something unexpected going on, so I will let you decide whether you want to close this thread here. @graugans .

commit 5aba6852 (2022-11-30): Merge branch '183-add-o3r-diagnostic-xmlrpc-methods' into 'main'

Support for retrieving O3R Diagnostics over XMLRPC

Closes #183

See merge request syntron/support/csr/ifm3d/ifm3d!181
closed 2019-11-06 by JD-ETH