ifm/ifm3d #184: IFM 03dxxx ip adress reset's to default

apr8, 2019-10-30:
We are facing this issue where the ip adress of the ifm gets reset to default 192.168.0.69, when we periodically reboot ifm, as a way to handle slow down in transmission and other unknown latency issues in SW trigger mode. And with the inability to configure the ip online as the gateway is also reset, we are able to change this only though windows application. 
FW version:1.30.4123
Driver version: 0.11.2
How do I solve this issue?
Thanks,
Akshay

theseankelly, 2019-10-31:
Hi @apr8 -- are you executing a hard reboot (by pulling the power supply) or soft reboot (via ifm3d either programmatically or via the command line tool)?

theseankelly, 2019-11-04:
Hey @apr8 -- I've set up a long-haul reboot test (reboot over and over) on my end using your FW and ifm3d version. Unfortunately I have not reproduced the issue; which likely means there's something else triggering your issue besides just a reboot.

Some asks for you:
- Can you please provide a camera dump of your running (working) system before the IP changes via the `ifm3d dump` tool?
- How often are you triggering the sensor in SW trigger mode?
- Can you provide more details about the slow-down and latency issues you allude to -- how do you detect this and how do you measure latency?

apr8, 2019-11-04:
1. Camera dump:

[ifm3d_j65_config.json.tar.gz](https://github.com/ifm/ifm3d/files/3804693/ifm3d_j65_config.json.tar.gz)

2. The camera is set to SW trigger mode through out and we trigger it at 5Hz, during specific operational state of our system.
3. We use ROS interface to publish the ifm sensor data to a topic. We have a a script that monitors this frequency and  a slowdown in this frequency would trigger a reboot. This issue was found when we were doing some stress test, by constantly rebooting the ifm3d, which we would not do ideally. But it could be possible if people from different ssh sessions try to reboot simultaneously multiple times (this would include the script trying to reboot the ifm as well).

theseankelly, 2019-11-04:
Thanks! Is it possible to share your stress test script/code? If so I can use it (and your config) to try for a repro.

theseankelly, 2019-11-04:
Thanks for the additional details, @apr8. My previous test slept 45 seconds between reboots, but since you stated you're doing a reboot stress test, I tried issuing the reboot command in a tighter loop and I was able to reproduce the issue.

Repro:

1. echo '{"ifm3d": {"Net": {"StaticIPv4Address": "192.168.0.170"}}}' | ifm3d config
2. while true; do ifm3d --ip=192.168.0.170 reboot; sleep 1; done
3. Let the script run for a few reboot cycles (I haven't had to wait more than ~5 minutes)
4. ifm3d dump
5. observe the value of StaticIPv4Address has returned to default, 192.168.0.69

Other camera settings (e.g. application settings) remain intact through this repro.

@graugans can you pass this info along to the embedded team to investigate?

graugans, 2019-11-06:
@apr8 thanks for reporting this issue. It looks like this behaviour is highly depending on the camera. With my camera I cannot reproduce this issue. @theseankelly can reproduce it with his camera. Anyway we are already working on a fix.

theseankelly, 2020-01-21:
A fix for this issue has been identified. It will be included in the next O3D3xx firmware release, targeting late Q1 or early Q2.