ifm/ifm3d #185: How do I get the exact factorized intrinsic parameters for O3X100 camera?

JiongruiHu, 2019-11-14:
I need the intrinsic parameters for my experiments, where can I find those numbers?
"SWVersion": {
        "Algorithm_Version": "0.2.0",
        "Calibration_Device": "00:02:01:40:a2:49",
        "Calibration_Version": "1.0.0",
        "ELDK": "GOLDENEYE_YOCTO_HARDFP-371-06d9c894636352a6c93711c7284d02b0c794a527",
        "IFM_Software": "1.0.126",
        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00029-gdf05c54-dirty (jenkins@dettlx152) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Wed Jan 24 13:40:11 CET 2018",
        "Main_Application": "0.4.1483"
      },
      "ifm3d_version": 1200

graugans, 2019-11-14:
Hey @JiongruiHu,
have you already checked this document: https://github.com/ifm/ifm3d/blob/master/doc/O3X_rectExample_annotated.pdf unfortunately the O3X at the moment does not provide the intrinsic parameters like the O3D does. Can you please help me to understand what are you the data needing for?

Best,
Christian

JiongruiHu, 2019-11-14:
What I really need is the extrinsic parameters, or more exactly the translations in x and y. I was thinking to use the intrinsics to rectify the amplitude image( from a chessboard) and then get the positions of the corners in the rectified image, in such a way I can get the position of the optical center in the world frame. 

In the pdf file, are the unit vectors from unit vector topic?

Thanks for your help

theseankelly, 2019-12-12:
Hi @JiongruiHu -- sorry for the delay here. The unit vectors you've seen are indeed from the unit vector topic, but unfortunately as @graugans mentions the O3X camera does not expose intrinsic calibration values like the O3D303 does. I'm double checking internally to see about other ways to recover extrinsics.

theseankelly, 2020-02-03:
Hi @JiongruiHu -- after re-reading this thread I realized the document @graugans linked is probably what you're looking for. Sorry for missing that.

theseankelly, 2020-02-13:
Closing -- feel free to reopen if the linked document isn't sufficient for your needs.

closed 2020-02-13 by theseankelly