ifm/ifm3d #186: How to fix "[WARN]: Timeout waiting for camera! " after network interruption

zhaoyingdeveloper, 2019-12-03:
Hi guys,

when  the camera occurs network interruption, after I reconnecting the network,  **I can ping the sensor actually**, but the camera driver always throw out an warning "[WARN]: Timeout waiting for camera! "(shown as the following img) and not work anymore. My current solution is reboot the camera(i.e, cut down the power off and power on again), but I want to use this camera in some real product which may not allowed to take too much time on reboot. Is there anyway to fix this issues ? 

[image]

theseankelly, 2019-12-12:
Hi @zhaoyingdeveloper -- the need to reboot the camera is not expected. The ROS node is able to recover communication after a network dropout as well without restarting anything.

Have you tried stopping and restarting the ROS node (instead of the camera)?

Can you please run `ifm3d dump`  and share the configuration of your device?

theseankelly, 2020-01-12:
Migrating issue to the ifm3d-ros project