ifm/ifm3d #187: Crash while waiting external trigger

HEVA-sea, 2019-12-04:
Hi,

I'm working on a robotic application using your OD303 camera and I need to use a negative edge trigger. However, I don't know when I will receive an image (it's our API which manage it) so I create a thread to catch any trig like this : 

```
bool CameraTOF::StartContinuousAcquisition() {
	if (isInitialised) {
		try {
			if (acquisitionThread != nullptr || isStarted)
				StopContinuousAcquisition();

			stopAcquisitionThread = false;

			acquisitionThread = new std::thread(&CameraTOF::acquisition, this);
			acquisitionThread->detach();

			isStarted = true;

			return true;
		}
		catch (ifm3d::error_t & _exception) {
			lastError = "CameraTOF::StartContinuousAcquisition - ifm3D exception : ";
			lastError += _exception.what();
			return false;
		}
	}
	else {
		lastError = "CameraTOF::StartContinuousAcquisition - camera is not initialised";
		return false;
	}
}

void CameraTOF::StopContinuousAcquisition() {
	if (isStarted) {
		stopAcquisitionThread = true;
		if (acquisitionThread != nullptr) {
			if (acquisitionThread->joinable())
				acquisitionThread->join();
			delete acquisitionThread;
			acquisitionThread = nullptr;
		}
	}
}

void CameraTOF::acquisition(){
	try {
		while (!stopAcquisitionThread) {
			if (!externalTrigger) 
				frameGrabber->SWTrigger();
			if (frameGrabber->WaitForFrame(imgBuffer.get(), 10000)) 
				tryToComputeDatas();
			else
				lastError ="No Image received";

			std::this_thread::sleep_for(std::chrono::milliseconds(100));
		}
	}
	catch (const ifm3d::error_t & ex) {
		lastError = "Ifmd3D Exception";
		lastError += ex.what();
	}
	catch (std::exception const& ex) {
		lastError = "Exception : ";
		lastError += ex.what();
	}
}
```

My camera is initialised in a previous method like that :

```
bool CameraTOF::InitCamera() {
	cam = ifm3d::Camera::MakeShared();
	imgBuffer = std::make_shared<ifm3d::ImageBuffer>();
	frameGrabber = std::make_shared<ifm3d::FrameGrabber>(
		cam, ifm3d::DEFAULT_SCHEMA_MASK | ifm3d::IMG_AMP | ifm3d::IMG_CART | ifm3d::IMG_RDIS | ifm3d::IMG_GRAY | ifm3d::IMG_UVEC);
	isInitialised = true;
	return isInitialised;
}
```

The problem is that when I'm waiting an image and when I don't receive it during the 10000ms, my application crash without any exception catched. The problem doesn't always occure on the first step, sometimes the loop run several times before crashing. It seems that when I receive an image, it delays the problem.

I set the TriggerMode on "4". When I try to use the software trig (TriggerMode value is "2", and my property externalTrigger is true), there is no problem.

Can you help me with this issue?

Thanks!

graugans, 2019-12-04:
Hey @HEVA-sea, have you considered to use ``WaitForFrame`` in blocking mode ``timeout_millis=0``? This will wait until your external trigger has fired. I also recommend to run your application in a debugger so you see where exactly the crash happens. Note the ``ImageBuffer`` [is not thread safe!](https://github.com/ifm/ifm3d/blob/master/modules/image/include/ifm3d/image/image_buffer.h#L38) I am also missing some sort of mutex in your thread code.

HEVA-sea, 2019-12-04:
I didn't know that we can use `timeout_millis=0` and indeed it solves my problem.

Before seeing your answer, I managed to catch the exception using `catch(...)`. When I caught the exception, I saw that the imgBuffer was dirty while I never received any trig.

Indeed, I don't have any mutex, I will have a look at this.

Thank's for your answer.

closed 2019-12-12 by theseankelly