ifm/ifm3d #188: Sporadic crashes on reconnect after restarting ifm o3d3 camera

timoja45i, 2019-12-17:
For integration of your camera in our own proprietary framework, I implemented a driver using your library. As a special requirement for our use case we should restart the camera  after a framedrop. 

While this approach works most of the time (around 80%), sometimes an unhandled exception occurs leading to the crash of the overall application. Please find an exemplary stacktrace below.

I'm aware that this might be not enough information to reproduce the error. However I was wondering if you already encountered a similar error and if there are some measures preventing such behavior. 

As you might notice, this error occurs when trying to access the device temperature after a reconnect. However, a similar error can also occur, but it is always contains the message malloc(): smallbin double linked list corrupted: 

**Environment:**
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12) 
ifm3d-* debian packages, version 0.17

`
malloc(): smallbin double linked list corrupted: 0x00007f01c0073ef0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f0217f437e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x82651)[0x7f0217f4e651]
/lib/x86_64-linux-gnu/libc.so.6(__libc_malloc+0x54)[0x7f0217f50184]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_Znwm+0x18)[0x7f0218842e78]
/usr/lib/x86_64-linux-gnu/libxmlrpc_client++.so.8(_ZN8xmlrpc_c3rpcC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_9paramListE+0x34)[0x7f021769af74]
/usr/lib/x86_64-linux-gnu/libxmlrpc_client++.so.8(_ZN8xmlrpc_c6rpcPtrC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_9paramListE+0x5c)[0x7f021769b99c]
/usr/lib/libifm3d_camera.so.0.17.0(_ZN5ifm3d6Camera4Impl6_XCallIJPKcEEEKN8xmlrpc_c5valueERNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSD_DpT_+0x10e)[0x7f021a501d7e]
/usr/lib/libifm3d_camera.so.0.17.0(_ZN5ifm3d6Camera4Impl15DeviceParameterERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0xe5)[0x7f021a4d4085]
/usr/lib/libifm3d_camera.so.0.17.0(_ZN5ifm3d6Camera15DeviceParameterERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x21)[0x7f021a4d41b1]
/home/user/git/workspaces/build-user2.0-Desktop_Qt_5_12_1_GCC_64bit-Debug/apps/user2.0_navigation/user2.0_navigation(_ZN6inmach6sensor11ifm_3d_impl27async_take_temperature_implESt10shared_ptrIdESt8functionIFvRKNS0_13ifm3d_error_tEEE+0x175)[0x28e54d9]
`

graugans, 2019-12-17:
Hey @timoja45i,
could you please provide a small example to express this issue. I guess you know that you have to destroy the camera object after you ``restart`` the  camera. I assume by ``restart`` you mean to cut the power off or call the ``Reboot`` method? Otherwise the underlying XML-RPC connection is in an invalid state.

Could you also please explain what you mean by ``after a framedrop``

timoja45i, 2019-12-17:
Hey graugans,

thanks for the quick reply!.  _After a framedrop_ means that framegrabber->WaitForFrame() returns false, meaning frame grabbing is unsuccessful.
Yes, we are destroying the camera object after a restart.
We are restarting by just disconnecting power, as this is the use case that we need to cover.

**Update:** I was unable to reproduce this behavior if I build the current master manually instead of using your provided debian packages. Could you provide the flags that are used for building the packages? Also, are there any changes on master not included in the 0.17 debian package?

graugans, 2019-12-17:
Hey @timoja45i,

you can check the build parameters based on the Dockerfile used for your platform. For Ubuntu 18.04 this looks like this: https://github.com/ifm/ifm3d/blob/master/docker/build/amd64/ubuntu/18.04/Dockerfile

Related to the changes, using the GitHub compare features does state both are the same: https://github.com/ifm/ifm3d/compare/master...dev-0.17.0

Maybe @theseankelly can provide more insights here. He keeps track of the deb packages

theseankelly, 2019-12-18:
Hey @timoja45i -- this is puzzling. The debian packages are built on the v0.17.0 tag which is equivalent to `master`. The build flags can indeed be found in the dockerfiles (see the link @graugans shared ). The build steps in the docker file are identical to what we publish in the README.

If you're able to share a minimal example of how you've wrapped `ifm3d` that will be most useful.

theseankelly, 2020-01-12:
@timoja45i  -- In order for this to be actionable, we need a minimal working example that demonstrates the issue. Would you be able to provide such an example?

timoja45i, 2020-01-13:
@theseankelly -- Sorry to disappoint you, but currently we can't provide a minimal example demonstrating the issue due to tight integration with proprietary code. For us, this issue doesn't have a high priority anymore. The workaround (compiling the library ourselves) is good enough for us. If you want you can close the issue, as I couldn't provide a minimal example and I can understand if you can't investigate this further.

theseankelly, 2020-01-13:
@timoja45i no worries. It's worth attempting a repro anyway so I'll give some effort to reproducing before closing the issue.

Could you confirm whether you're using ROS, and whether you sourced the problematic binaries from the ROS or the non-ROS apt repository?

timoja45i, 2020-01-20:
@theseankelly I've been using the non-ROS apt repository, we don't have any ROS dependencies.

theseankelly, 2020-01-20:
@timoja45i -- I'll dig into it. Thanks for the report & information!

theseankelly, 2020-02-18:
Hey @timoja45i -- I've attempted to create a minimal example but have failed to reproduce the issue. Could you take a quick look at the code below and see if it accurately mimics your use case?

Couple specific questions:
1. Timing -- how long are you waiting between power cycling and creating the new camera object? Are you polling like below or sleeping for some fixed interval?
2. Object lifecycle -- how are you destroying and recreating the objects? Are you using shared_ptr::reset, or letting objects go out of scope, etc?

```cpp
#include <iostream>
#include <memory>
#include <thread>

#include <ifm3d/camera.h>
#include <ifm3d/fg.h>
#include <ifm3d/image.h>

int main()
{
  // Initial setup
  auto cam = ifm3d::Camera::MakeShared();
  auto fg = std::make_shared<ifm3d::FrameGrabber>(cam);
  auto im = std::make_shared<ifm3d::ImageBuffer>();

  // Try for a repro in a loop
  while (true)
  {
    // Poll camera temperature until we get a success...
    while (true)
    {
      std::cout << "Polling for camera temperature..." << std::endl;
      try
      {
        cam.reset(new ifm3d::Camera());

        // This is the call we expect to trigger a segfault
        std::cout << "Camera Temperature: "
                  << cam->DeviceParameter("TemperatureIllu")
                  << std::endl;
        break;
      }
      catch (const ifm3d::error_t& ex)
      {
        std::cout << "Error handled...retrying" << std::endl;
        // absorb errors -- this way we'll continually retry
      }
    }

    // Grab some frames with the newly created camera.
    fg.reset(new ifm3d::FrameGrabber(cam));
    for (std::uint8_t i = 0 ; i < 10; ++i)
    {
      if(fg->WaitForFrame(im.get(), 1000))
      {
        std::cout << "Got frame" << std::endl;
      }
      else
      {
        std::cout << "Timed out waiting for frame" << std::endl;
      }
    }

    // Simulate the power cycle by rebooting.
    // Also sleep a while to ensure we don't re-create the camera object
    // before the camera has actually gone down for reboot.
    std::cout << "Rebooting!" << std::endl;
    cam->Reboot();
    std::this_thread::sleep_for(std::chrono::seconds(20));
  }

  return 0;
}
```

timoja45i, 2020-02-19:
Hi Sean,

thanks for still investigating the issue!
regarding 1:  We are sleeping for more than 20 s after a reboot was 
triggered/connection was lost.
2: After reboot is triggered, we do a reset of the camera shared_ptr, 
but first to destroy the contained object (camera_.reset()). We reset it 
with an actual object when trying to reconnect.

> Hey @timoja45i <https://github.com/timoja45i> -- I've attempted to 
> create a minimal example but have failed to reproduce the issue. Could 
> you take a quick look at the code below and see if it accurately 
> mimics your use case?
>
> Couple specific questions:
>
>  1. Timing -- how long are you waiting between power cycling and
>     creating the new camera object? Are you polling like below or
>     sleeping for some fixed interval?
>  2. Object lifecycle -- how are you destroying and recreating the
>     objects? Are you using shared_ptr::reset, or letting objects go
>     out of scope, etc?
>
> #include  <iostream>
> #include  <memory>
> #include  <thread>
>
> #include  <ifm3d/camera.h>
> #include  <ifm3d/fg.h>
> #include  <ifm3d/image.h>
>
> int  main()
> {
>    // Initial setup
>    auto  cam =ifm3d::Camera::MakeShared();
>    auto  fg = std::make_shared<ifm3d::FrameGrabber>(cam);
>    auto  im = std::make_shared<ifm3d::ImageBuffer>();
>
>
>    // Try for a repro in a loop
>    while  (true)
>    {
>      // Poll camera temperature until we get a success...
>      while  (true)
>      {
>        std::cout <<"Polling for camera temperature..."  << std::endl;
>        try
>        {
>          cam.reset(new  ifm3d::Camera());
>
>          // This is the call we expect to trigger a segfault
>          std::cout <<"Camera Temperature: "
>                    << cam->DeviceParameter("TemperatureIllu")
>                    << std::endl;
>          break;
>        }
>        catch  (const  ifm3d::error_t& ex)
>        {
>          std::cout <<"Error handled...retrying"  << std::endl;
>          // absorb errors -- this way we'll continually retry
>        }
>      }
>
>      // Grab some frames with the newly created camera.
>      fg.reset(new  ifm3d::FrameGrabber(cam));
>      for  (std::uint8_t  i =0  ; i <10; ++i)
>      {
>        if(fg->WaitForFrame(im.get(),1000))
>        {
>          std::cout <<"Got frame"  << std::endl;
>        }
>        else
>        {
>          std::cout <<"Timed out waiting for frame"  << std::endl;
>        }
>      }
>
>      // Simulate the power cycle by rebooting.
>      // Also sleep a while to ensure we don't re-create the camera object
>      // before the camera has actually gone down for reboot.
>      std::cout <<"Rebooting!"  << std::endl;
>      cam->Reboot();
>      std::this_thread::sleep_for(std::chrono::seconds(20));
>    }
>
>    return  0;
> }
-- 
Timo Bressmer

Softwareentwicklung

*InMach Intelligente Maschinen GmbH*
Nicolaus-Otto-Str. 4
89079 Ulm

fon: +49 731 550166-69

e-mail: timo.bressmer@inmach.de <mailto:timo.bressmer@inmach.de>
web: www.inmach.de <http://www.inmach.de/>

Name und Sitz der Gesellschaft: InMach Intelligente Maschinen GmbH, Ulm
Handelsregister: Ulm, HRB 4829
Geschäftsführung: Matthias Strobel, Dr. Boris Kluge
USt-IdNr.: DE813776520
AGB: www.inmach.de/agb <http://www.inmach.de/agb/>

Diese E-Mail und alle mitgesendeten Dateien sind vertraulich und 
ausschließlich für den Gebrauch durch den Empfänger bestimmt.
This E-Mail and any files transmitted with it are confidential intended 
solely for the use of the addressee.

theseankelly, 2020-02-19:
Thanks @timoja45i -- I've modified my example per your feedback. Unfortunately I'm still not able to reproduce the issue.

It seems that the `smallbin double linked list corrupted` is most likely caused by writing to an invalid memory address/pointer at some point in the flow of execution. In this case, the site of the crash does not necessarily match the site of the invalid write. I recommend you run your application through [valgrind](https://valgrind.org) to check for any possible invalid memory accesses. I have run my sample (with a full a Debug build of ifm3d) through valgrind and nothing is flagged.

(NOTE: The ifm3d CMakeLists.txt isn't well set up to build debug binaries -- you need to manually change ln 45 from `Release` to `Debug`, and also remove the `-s` flag from `CMAKE_CXX_FLAGS` on line 53 to avoid stripping debug symbols.)

I'll close for now since I can't take this any farther unfortunately. Please do reach out again if you're still able to reproduce and can share any portion of your implementation, or if you have valgrind output that flags a problem in the ifm3d library.

closed 2020-02-19 by theseankelly