ifm/ifm3d #189: Max Framerate with O3X

pfrydlewicz, 2019-12-19:
Hi,
how is it possible to get the max framerate out of the O3X with ifm3d?

The spec gives 20fps. However, when setting all parameters to minimal settings, I cannot get more than 6fps. Even when only transferring the point-cloud data (no other devices on the network).

Is the performance of my machine (using virtual box) the problem??
A customer encounters a similar issue, but he gets 10fps.

Thanks,
Paul

pfrydlewicz, 2020-01-06:
An update on this.
By increasing the MTU from 1500 (default) to 9000 for my ethernet device, the framerate increases by around 30%.
Obviously the bottleneck is the network.
Does anyone know some other tweaks to increase the throughput?

Arag24, 2020-01-06:
@pfrydlewicz , I'm not really understanding totally your point here.
You used Jumbo Frame / MTU to 9k of your computer's NIC, and you have seen a difference in fps performance of your O3X?

It looks strange for me. Indeed, adding Jumbo frame is big help to decrease CPU ticks and so CPU load at the end to transfer with same quantity of data. (just less packets)
From what I had seen, I would say yes O3X could have some limitation embeeded (and yes could affect network performance) but it would require to change packet size inside O3X camera to get a benefit somewhere (waht is not allowed up to now I think).

To give som factual data on this topic : currently, on my computer I'm using this config. file:
```
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "1056751412",
        "Imager": {
          "ExposureTime": "1080",
          "ExposureTimeRatio": "28",
          "FrameRate": "20",
          "MaxAllowedFrameRate": "11.3636363636364",
          "MinimumAmplitude": "42",
          "SpatialFilter": {
            "MaskSize": "0"
          },
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {
            "": ""
          },
          "TemporalFilterType": "0",
          "Type": "upTo02m_moderate"
        },
        "Index": "1",
        "Name": "",
        "OutputAmplitudeImage": "true",
        "OutputConfidenceImage": "true",
        "OutputDistanceImage": "false",
        "OutputGrayscaleImage": "false",
        "OutputXYZImage": "true",
        "TriggerMode": "1",
        "Type": "Camera"
      }
    ],
    "Device": {
      "ArticleNumber": "O3X100",
      "ArticleStatus": "AA",
      "Description": "",
      "DeviceType": "1:512",
      "IPAddressConfig": "0",
      "ImageTimestampReference": "1516804772",
      "Name": "New sensor",
      "OperatingMode": "0",
      "PasswordActivated": "false",
      "SessionTimeout": "30",
      "TemperatureIllu": "42.125",
      "UpTime": "0.388333333333333"
    },
    "Net": {
      "MACAddress": "00:02:01:40:7A:29",
      "NetworkSpeed": "0",
      "StaticIPv4Address": "192.168.0.69",
      "StaticIPv4Gateway": "192.168.0.201",
      "StaticIPv4SubNetMask": "255.255.255.0",
      "UseDHCP": "false"
    },
    "Time": {
      "CurrentTime": "1516804772",
      "NTPServers": "",
      "StartingSynchronization": "false",
      "Stats": "",
      "SynchronizationActivated": "false",
      "Syncing": "false",
      "WaitSyncTries": "1"
    },
    "_": {
      "Date": "Mon Jan  6 12:45:19 2020",
      "HWInfo": {
        "MACAddress": "00:02:01:40:7A:29",
        "Mainboard": "#!02_M162_B30_12345678_008026357",
        "MiraSerial": "Not implemented"
      },
      "SWVersion": {
        "Algorithm_Version": "0.1.7",
        "Calibration_Device": "00:02:01:40:7a:29",
        "Calibration_Version": "1.0.0",
        "ELDK": "GOLDENEYE_YOCTO_HARDFP-371-06d9c894636352a6c93711c7284d02b0c794a527",
        "IFM_Software": "1.0.122",
        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00029-gdf05c54-dirty (jenkins@dettlx152) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Wed Jan 24 13:40:11 CET 2018",
        "Main_Application": "0.4.1479"
      },
      "ifm3d_version": 1700
    }
  }
}
```
with `ifm3d hz --nframes=50 --nruns=5`
I got the following results (MTU/jumbo frame change on my computer NIC)

- With MTU Disabled -> 10.9881 Hz / 10.9869 Hz / 10.9884 Hz (3 tests)
- With MTU = 9k  ->10.9882 Hz / 10.9873 Hz / 10.9892 Hz (3 tests)

Having a look on Wireshark with MTU at 9k : 
> Frame Length: 1514 bytes (12112 bits)
because I have no chance from camera directly...

One more thing, @tompanzarella has made a very deep analysis [https://github.com/ifm/ifm3d/blob/master/doc/jitter/jitter.ipynb](https://github.com/ifm/ifm3d/blob/master/doc/jitter/jitter.ipynb). Inded, in some configurations od O3X we could see some latensties appearing. This was due to calculation inside the camera that would impact limited resource inside camera. In such cas, we recommend using IFM3D_MASK = 3 that would lead to get only radial distnace information (instead of cartesian) and you have to make your calc with unit vectors ino your calculation unit. 
A good example is included inside test of `image `module. [here](https://github.com/ifm/ifm3d/blob/master/modules/image/test/ifm3d-image-tests.cpp)

It also has the benefit to reduce bandwith (1 info instead of 3) 👍 

Hope it could help...

graugans, 2020-01-07:
Hey @pfrydlewicz 

Could you please so kind and share your Firmware and ifm3d version and the configuration (``ifm3d dump``) you are using 

Can you please check what your `` "MaxAllowedFrameRate"`` value looks like? This value is typically limited by the eye-safety constraint coming from the laser class and the VCSEL used. The `` "MaxAllowedFrameRate"`` is the absolut maximum you could achieve. Typically your result is a little bit below this value. Like in the configuration @Arag24 posted above

 ``11.3636363636364`` vs. ``10.9881``

As @Arag24 already pointed out, increasing the MTU on the host side should not make any big difference because the camera still sends data in 1500 byte chunks.

In the past we had some bad experience with various USB 2.0 network interface cards. It also highly depends on your machine set-up and running ifm3d on a Virtual machine may introduce additionally latency.

pfrydlewicz, 2020-01-07:
Thanks for your responses.
Well, indeed it might make no sense, but on two of my machines (one virtual Box, one not, Intel i5) and on the machine of our customer it did make an improvement, all around 30% increase in framerate.

I tried to run iperf and tried playing around with various TCP settings. I could go up to over 90Mbit/s.
However I don't know if that is a valid test.
Can you acknowledge the use of iperf? Does it measure the correct data load?
The funny thing is, that the data bandwidth is higher than necessary for 20fps. That's why I'm not sure about its validity. 

In my tests I didn't focus on the data quality, only to get 20fps on my machine. All settings where at the lowest possible values.
I'll share the settings in the following comment.

pfrydlewicz, 2020-01-07:
Until now with either setting I couldn't get more than 12 fps...

pfrydlewicz, 2020-01-07:
This is my current config file:
------------------------------------------------
{
  "ifm3d": {
    "Apps": [
      {
        "Description": "",
        "Id": "1009285635",
        "Imager": {
          "AutoExposureMaxExposureTime": "10000",
          "AutoExposureMaxNumberOfSaturatedPixels": "20",
          "AutoExposureReferencePointX": "88",
          "AutoExposureReferencePointY": "66",
          "AutoExposureReferenceROI": "{\"ROIs\":[{\"id\":0,\"group\":0,\"type\":\"Rect\",\"width\":224,\"height\":172,\"angle\":0,\"center_x\":112,\"center_y\":86}]}",
          "AutoExposureReferenceType": "0",
          "ContinuousAutoExposure": "false",
          "ExposureTime": "50",
          "FrameRate": "20",
          "MaxAllowedFrameRate": "250",
          "MinimumAmplitude": "42",
          "SpatialFilter": {
            "MaskSize": "0"
          },
          "SpatialFilterType": "0",
          "SymmetryThreshold": "0.4",
          "TemporalFilter": {
            "": ""
          },
          "TemporalFilterType": "0",
          "Type": "upTo02m_low"
        },
        "Index": "1",
        "Name": "",
        "OutputAmplitudeImage": "true",
        "OutputConfidenceImage": "true",
        "OutputDistanceImage": "false",
        "OutputGrayscaleImage": "false",
        "OutputXYZImage": "true",
        "TriggerMode": "1",
        "Type": "Camera"
      }
    ],
    "Device": {
      "ArticleNumber": "O3X100",
      "ArticleStatus": "AA",
      "Description": "This is a O3X presetation device.",
      "DeviceType": "1:512",
      "IPAddressConfig": "0",
      "ImageTimestampReference": "1516803550",
      "Name": "My O3X",
      "OperatingMode": "0",
      "PasswordActivated": "false",
      "SessionTimeout": "30",
      "TemperatureIllu": "31.375",
      "UpTime": "0.0491666666666667"
    },
    "Net": {
      "MACAddress": "00:02:01:40:6B:98",
      "NetworkSpeed": "0",
      "StaticIPv4Address": "192.168.0.69",
      "StaticIPv4Gateway": "192.168.0.201",
      "StaticIPv4SubNetMask": "255.255.255.0",
      "UseDHCP": "false"
    },
    "Time": {
      "CurrentTime": "1516803550",
      "NTPServers": "",
      "StartingSynchronization": "false",
      "Stats": "",
      "SynchronizationActivated": "false",
      "Syncing": "false",
      "WaitSyncTries": "1"
    },
    "_": {
      "Date": "Tue Jan  7 15:39:11 2020",
      "HWInfo": {
        "MACAddress": "00:02:01:40:6B:98",
        "Mainboard": "#!02_M162_B30_12345678_008026357",
        "MiraSerial": "Not implemented"
      },
      "SWVersion": {
        "Algorithm_Version": "0.2.0",
        "Calibration_Device": "00:02:01:40:6b:98",
        "Calibration_Version": "1.0.0",
        "ELDK": "GOLDENEYE_YOCTO_HARDFP-371-06d9c894636352a6c93711c7284d02b0c794a527",
        "IFM_Software": "1.0.126",
        "Linux": "Linux version 3.14.34-rt31-yocto-standard-00029-gdf05c54-dirty (jenkins@dettlx152) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Wed Jan 24 13:40:11 CET 2018",
        "Main_Application": "0.4.1483"
      },
      "ifm3d_version": 1700
    }
  }
}

pfrydlewicz, 2020-01-07:
> Having a look on Wireshark with MTU at 9k :
> 
> > Frame Length: 1514 bytes (12112 bits)
> > because I have no chance from camera directly...
> 

Is 1514 not more than 1500?
What does it mean to the receiver, when its MTU is 1500 bytes and the sent sequence is 1514 bytes long?

pfrydlewicz, 2020-01-07:
> Having a look on Wireshark with MTU at 9k :
> 
> > Frame Length: 1514 bytes (12112 bits)

I am not good in the TCP details, but is it possible that when using a MTU of 9000, the buffer may hold multiple sequences of 1500 bytes?
A reason might be that the controller was too busy clearing the data.

In this case it may indeed increase the transfer speed because it reduces re-sending packages, am I correct?
This is the case if the CPU/controller is the bottleneck.

graugans, 2020-01-07:
A typical Ethernet Wire frame is 1514 bytes (14-byte Ethernet header plus 1500 bytes data) so the payload maximum transfer unit on a traditional **Fast Ethernet** (100 MBit/s) is 1500 bytes (MTU) The 9k is also known as Jumbo Frames which was introduced in Gigabit Ethernet. It is way too long since I left University and I am not 100% sure if TCP has some handshaking related to the MTU.

Things are getting harder when running Wireshark on the same Machine which captures the frames. There is a thing called  ``TCP Large Segment Offload`` which sends bigger packets to the hardware and the hardware takes care of the segmentation.

I expect that in your case you are sending Data in 9k Jumbo Frames some one within the routing needs to reassemble the packages in 1.5k packages to address the Fast Ethernet wire. The camera itself will send data in 1.5k frames which may be stuffed together to 9k Jumbo frames between the Windows host and the VM. It can make sense to use larger frames on the virtual interface to mitigate context changes during copy to kernel and copy to user space.

My expectation is that a decent modern PC should be capable in handling 100 MBit/s of data.

graugans, 2020-01-07:
> Thanks for your responses.
> Well, indeed it might make no sense, but on two of my machines (one virtual Box, one not, Intel i5) and on the machine of our customer it did make an improvement, all [...]
The usage of ``iperf`` will give you a good hint of your overall Ethernet throughput. So in my opinion it is a valid test. You might need to take care that you are using TCP and not UDP (what might be the default) There is also a lot of fine tuning possible in iperf.

> The funny thing is, that the data bandwidth is higher than necessary for 20fps. That's why I'm not sure about its validity.

My gut feeling tells me that we might have more of an CPU constrain on the camera than a Network bottleneck. IMHO the O3X is not capable of producing a higher framerate than the 12 FPS you measured.
> 
> In my tests I didn't focus on the data quality, only to get 20fps on my machine. All settings where at the lowest possible values.
What is the goal of having 20fps?
> I'll share the settings in the following comment.

pfrydlewicz, 2020-01-07:
Well, you might be right, the CPU seems to be the constraint here.

When you use my configuration above on your hardware, can you receive 20fps?
(with ifm3d)

Arag24, 2020-01-07:
@pfrydlewicz , @graugans 

Here is what I did, with your FW and config file:

`ifm3d jitter --nframes=500 --outfile=cycle.csv`

**Results are interesting (funny?):**

- With ifm3d::ByteBuffer implementation:

> Mean:   63.1018 ms
> Median:   62.1578 ms
> Stdev:  7.64836 ms
> Mad:    5.74176 ms

- With ifm3d::ImageBuffer implementation:

> Mean:   105.983 ms
> Median: 105.775 ms
> Stdev:  9.7569 ms
> Mad:    3.2268 ms

In a graph, it looks as following:
[image]

_I tried 5 times the same test. Sometimes Bytebuffer Mean is 100ms and ImageBuffer is 60 ms._
Example below:
[image]

It looks like it is not stable...

graugans, 2020-01-07:
Hey @Arag24 interesting output you get there. I had to reinstall my PC so some time was needed to set-up my Docker container again. This is what I do measure with the config from @pfrydlewicz this is running Ubuntu 18.04 and a Ubuntu 18.04 Docker container on top of this:

ifm@695599c14e1a:~$ ifm3d hz --nframes=100 --nruns=3
FrameGrabber running at: 10.3911 Hz
100 frames captured, over 3 runs
ifm@695599c14e1a:~$ IFM3D_MASK=3 ifm3d hz --nframes=100 --nruns=3
FrameGrabber running at: 19.2757 Hz
100 frames captured, over 3 runs

So the **19.2757 Hz** looks quite close to the goal @pfrydlewicz wants to achieve.

pfrydlewicz, 2020-01-07:
Thanks for all your work. The results make sense. And it makes me better understand the problem.

When I see @Arag24 and @graugans results, there is indeed a problem in the framerates which is obviously caused by the limitations in the choice of 100Mbit ethernet.

With the configuration @Arag24 only gets 15fps, @graugans even only 10fps.
Only when @graugans switches off almost all data he gets the 20fps.

Measured with iperf I get less than 90Mbit/s on the connection, that is 11 MB/s.
When doing a rough estimate, one frame has:
- XYZ: 39000 x 3 x 4 bytes
- Amplitude: 39000 x 4 bytes
- Confidence: 39000 x 2 bytes

Put all together it’s 700,000 bytes per frame or 7 MB per second at 10 fps.
There will be some overhead, that means when using the camera, the network always limits it to a bit more than 10fps (when downloading an useful amount of data).

Is ifm aware that the camera performance could be better if the bandwidth was higher?
When talking with prospects in robotics I am often asked, why the camera runs only at 10fps.

obviously this is not necessarily an issue with the imager settings, but mainly due to bandwidth limitations.

Are there plans to use 1000bit ethernet?

pfrydlewicz, 2020-01-07:
@Arag24 , in your opinion what is the explanation that in one test the ImageBuffer needs twice the time, in the other example vice versa?

Arag24, 2020-01-07:
@graugans , @pfrydlewicz 

Could you please use `ifm3d jitter` command to see variability of the framerate, please ?
And here :

- using `IFM3D_MASK=3`, I got very stable results with your config @pfrydlewicz . (500 frames & > 5 tests)

> Mean:   52.9651 ms
> Median: 53.1998 ms
> Stdev:  1.40917 ms
> Mad:    0.986984 ms

FrameGrabber running at: 18.863 Hz
500 frames captured, over 1 runs

- using `IFM3D_MASK=10`, I got unstable results:

Capturing frame data for ifm3d::ByteBuffer...

I0107 17:59:35.097549 14312 frame_grabber_impl.hpp:596] FrameGrabber thread done.
Mean:   70.6636 ms
Median: 66.3116 ms
Stdev:  17.9467 ms
Mad:    5.81287 ms

Capturing frame data for ifm3d::ImageBuffer...

> Mean:   63.7276 ms
> Median: 61.8337 ms
> Stdev:  11.3688 ms
> Mad:    5.26823 ms

From what I think @pfrydlewicz , and what I already saw in the past, the root cause is CPU calculation inside the camera. 
We are CPU constrain, and so calculation are "not free". As consequence, we can reach some "unstable" answers and miss ome frames... I often recommend to have a look on `jitter` command to see what's going back. When this behavior occur, I also observed, NIC interruption is becoming mad, and this is a ....!

So for me : Limit of embedded computation -> Frame Not Computed/or delayed/Next frame sent -> Erratic FrameRate.
And so, even if I would lov to get MTU/jumboFrame option inside camera, I don't think this would help (except if the extra amount of bytes on the wire transfer is leading to some issue [instead of extra computation] and this would have been to test)

graugans, 2020-01-08:
``<spoiler>@pfrydlewicz the next 3D camera platform will come with Gigabit Ethernet..... `<\spoiler>``

closed 2020-01-12 by theseankelly