ifm/ifm3d #191: Make check issue

NKKD, 2020-01-13:
Hi All,
I'm new to IFM cameras, trying to build ifm3d library and get some point clouds directly from the camera. Unfortunately I'm not able to build the library successfully. I attached the errors below. Any idea how to fix it?

Thanks in advance.

IFM3D_IP=169.254.36.192 make check
[ 18%] Built target ifm3d_camera
[ 31%] Built target ifm3d_framegrabber
[ 37%] Built target ifm3d_image
[ 43%] Built target gtest
[ 50%] Built target gtest_main
[ 59%] Built target ifm3d-image-tests
[==========] Running 10 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 10 tests from Image
[ RUN      ] Image.MoveCtor
[       OK ] Image.MoveCtor (249 ms)
[ RUN      ] Image.MoveAssignmentOperator
[       OK ] Image.MoveAssignmentOperator (175 ms)
[ RUN      ] Image.CopyCtor
[       OK ] Image.CopyCtor (200 ms)
[ RUN      ] Image.CopyAssignmentOperator
[       OK ] Image.CopyAssignmentOperator (193 ms)
[ RUN      ] Image.References
[       OK ] Image.References (200 ms)
[ RUN      ] Image.CloudMechanics
[       OK ] Image.CloudMechanics (399 ms)
[ RUN      ] Image.XYZImage
[       OK ] Image.XYZImage (205 ms)
[ RUN      ] Image.ComputeCartesian
[       OK ] Image.ComputeCartesian (401 ms)
[ RUN      ] Image.TimeStamp
[       OK ] Image.TimeStamp (1468 ms)
[ RUN      ] Image.IlluTemp
[       OK ] Image.IlluTemp (202 ms)
[----------] 10 tests from Image (3693 ms total)

[----------] Global test environment tear-down
[==========] 10 tests from 1 test case ran. (3693 ms total)
[  PASSED  ] 10 tests.
[ 59%] Built target check_image
[ 75%] Built target ifm3d-camera-tests
[==========] Running 19 tests from 3 test cases.
[----------] Global test environment set-up.
[----------] 16 tests from CameraTest
[ RUN      ] CameraTest.FactoryDefaults
[       OK ] CameraTest.FactoryDefaults (6807 ms)
[ RUN      ] CameraTest.DefaultCredentials
[       OK ] CameraTest.DefaultCredentials (17 ms)
[ RUN      ] CameraTest.ApplicationList
[       OK ] CameraTest.ApplicationList (39 ms)
[ RUN      ] CameraTest.SessionManagement
[       OK ] CameraTest.SessionManagement (200 ms)
[ RUN      ] CameraTest.CopyDeleteApplication
[       OK ] CameraTest.CopyDeleteApplication (918 ms)
[ RUN      ] CameraTest.CopyDeleteExceptions
[       OK ] CameraTest.CopyDeleteExceptions (719 ms)
[ RUN      ] CameraTest.CreateDeleteApplication
[       OK ] CameraTest.CreateDeleteApplication (1239 ms)
[ RUN      ] CameraTest.CreateApplicationException
[       OK ] CameraTest.CreateApplicationException (366 ms)
[ RUN      ] CameraTest.ImportExportApplication
[       OK ] CameraTest.ImportExportApplication (1324 ms)
[ RUN      ] CameraTest.ImportExportConfig
[       OK ] CameraTest.ImportExportConfig (1146 ms)
[ RUN      ] CameraTest.ActiveApplication
[       OK ] CameraTest.ActiveApplication (6076 ms)
[ RUN      ] CameraTest.ImagerTypes
[       OK ] CameraTest.ImagerTypes (32718 ms)
[ RUN      ] CameraTest.Filters
[       OK ] CameraTest.Filters (8816 ms)
[ RUN      ] CameraTest.JSON
[       OK ] CameraTest.JSON (5349 ms)
[ RUN      ] CameraTest.Time
[       OK ] CameraTest.Time (976 ms)
[ RUN      ] CameraTest.TemporaryParameters
[       OK ] CameraTest.TemporaryParameters (69 ms)
[----------] 16 tests from CameraTest (66779 ms total)

[----------] 2 tests from Err
[ RUN      ] Err.LibraryError
[       OK ] Err.LibraryError (0 ms)
[ RUN      ] Err.SystemError
[       OK ] Err.SystemError (0 ms)
[----------] 2 tests from Err (0 ms total)

[----------] 1 test from Version
[ RUN      ] Version.Version
[       OK ] Version.Version (0 ms)
[----------] 1 test from Version (0 ms total)

[----------] Global test environment tear-down
[==========] 19 tests from 3 test cases ran. (66779 ms total)
[  PASSED  ] 19 tests.
[ 75%] Built target check_camera
[ 84%] Built target ifm3d-fg-tests
[==========] Running 14 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 14 tests from FrameGrabber
[ RUN      ] FrameGrabber.FactoryDefaults
[       OK ] FrameGrabber.FactoryDefaults (6988 ms)
[ RUN      ] FrameGrabber.ByteBufferBasics
[       OK ] FrameGrabber.ByteBufferBasics (0 ms)
[ RUN      ] FrameGrabber.WaitForFrame
[       OK ] FrameGrabber.WaitForFrame (1869 ms)
[ RUN      ] FrameGrabber.CustomSchema
[       OK ] FrameGrabber.CustomSchema (222 ms)
[ RUN      ] FrameGrabber.IntrinsicParamSchema
[       OK ] FrameGrabber.IntrinsicParamSchema (37 ms)
[ RUN      ] FrameGrabber.InverseIntrinsicParamSchema
[       OK ] FrameGrabber.InverseIntrinsicParamSchema (37 ms)
[ RUN      ] FrameGrabber.ByteBufferMoveCtor
[       OK ] FrameGrabber.ByteBufferMoveCtor (106 ms)
[ RUN      ] FrameGrabber.ByteBufferMoveAssignmentOperator
[       OK ] FrameGrabber.ByteBufferMoveAssignmentOperator (198 ms)
[ RUN      ] FrameGrabber.ByteBufferCopyCtor
[       OK ] FrameGrabber.ByteBufferCopyCtor (200 ms)
[ RUN      ] FrameGrabber.ByteBufferCopyAssignmentOperator
[       OK ] FrameGrabber.ByteBufferCopyAssignmentOperator (201 ms)
[ RUN      ] FrameGrabber.FrameGrabberRecycling
[       OK ] FrameGrabber.FrameGrabberRecycling (1999 ms)
[ RUN      ] FrameGrabber.SoftwareTrigger
[       OK ] FrameGrabber.SoftwareTrigger (7827 ms)
[ RUN      ] FrameGrabber.SWTriggerMultipleClients
[       OK ] FrameGrabber.SWTriggerMultipleClients (6003 ms)
[ RUN      ] FrameGrabber.JSON_model
[       OK ] FrameGrabber.JSON_model (95 ms)
[----------] 14 tests from FrameGrabber (25782 ms total)

[----------] Global test environment tear-down
[==========] 14 tests from 1 test case ran. (25782 ms total)
[  PASSED  ] 14 tests.
[ 84%] Built target check_fg
[ 90%] Built target ifm3d_swupdater
[100%] Built target ifm3d-swupdater-tests
[==========] Running 3 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 3 tests from SWUpdater
[ RUN      ] SWUpdater.FactoryDefaults
[       OK ] SWUpdater.FactoryDefaults (16955 ms)
[ RUN      ] SWUpdater.DetectBootMode
/home/j/Documents/ifm3d-master/modules/swupdater/test/ifm3d-swupdater-tests.cpp:47: Failure
Value of: swu->WaitForRecovery(60000)
  Actual: false
Expected: true
/home/j/Documents/ifm3d-master/modules/swupdater/test/ifm3d-swupdater-tests.cpp:50: Failure
Value of: swu->WaitForRecovery(-1)
  Actual: false
Expected: true
unknown file: Failure
C++ exception with description "Lib: Couldn't connect to the device (make sure the device is in Recovery Mode)" thrown in the test body.
[  FAILED  ] SWUpdater.DetectBootMode (84372 ms)
[ RUN      ] SWUpdater.FlashEmptyFile
/home/j/Documents/ifm3d-master/modules/swupdater/test/ifm3d-swupdater-tests.cpp:65: Failure
Value of: swu->WaitForProductive(-1)
  Actual: false
Expected: true
unknown file: Failure
C++ exception with description "Lib: XMLRPC Timeout - can you `ping' the sensor?" thrown in the test body.
[  FAILED  ] SWUpdater.FlashEmptyFile (18363 ms)
[----------] 3 tests from SWUpdater (119690 ms total)

[----------] Global test environment tear-down
[==========] 3 tests from 1 test case ran. (119690 ms total)
[  PASSED  ] 1 test.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] SWUpdater.DetectBootMode
[  FAILED  ] SWUpdater.FlashEmptyFile

 2 FAILED TESTS
modules/swupdater/test/CMakeFiles/check_swupdater.dir/build.make:57: recipe for target 'modules/swupdater/test/CMakeFiles/check_swupdater' failed
make[3]: *** [modules/swupdater/test/CMakeFiles/check_swupdater] Error 1
CMakeFiles/Makefile2:700: recipe for target 'modules/swupdater/test/CMakeFiles/check_swupdater.dir/all' failed
make[2]: *** [modules/swupdater/test/CMakeFiles/check_swupdater.dir/all] Error 2
CMakeFiles/Makefile2:109: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2

theseankelly, 2020-01-13:
Hi @NKKD -- I've never tried running the unit tests with a non-default IP address. I wonder if that's related. I'll try it out in the next couple days.

In the meantime, the unit tests are not mandatory to finish building. Simply omit the `make check` step and proceed to `make package`.

NKKD, 2020-01-13:
Thank you so much @theseankelly.
Another issue I'm facing right now is that I'm trying to run the ifm3d examples. But it gives error like this:
~/Documents/ifm3d-examples-master/build/file_io$ IFM3D_IP=169.254.36.192 ./ex-file_io 
terminate called after throwing an instance of 'ifm3d::error_t'
  what():  Lib: XMLRPC Timeout - can you `ping' the sensor?
Aborted (core dumped)

Any idea how to fix it?

graugans, 2020-01-14:
Hey @NKKD, have you already checked https://github.com/ifm/ifm3d/blob/master/doc/troubleshoot.md Is there any specific reason why you are using the link local address of the camera? Depending on your Network set-up this might change during address conflicts and automatic address assignment.

As stated in the error message have you tried to ping the camera? you can also check with

```
$ avahi-browse -arp 2>/dev/null | grep  ^=.*O3D.*Workstation
```
If the same address is reported.

theseankelly, 2020-01-14:
Opened #192 related to unit tests

NKKD, 2020-01-14:
@graugans Thank you for getting back to me. My camera is directly connected to my PC so I just use link local address.
I can ping the camera and I made the example code worked properly, but the wired thing is sometime it work sometime it doesn't. Don't really know what happened in it.

theseankelly, 2020-01-14:
@NKKD for initial evaluation, I recommend you leave the camera IP at the default (192.168.0.69) and configure your NIC for a static IP in the camera's subnet (I typically use 192.168.0.2).

closed 2020-01-15 by NKKD