ifm/ifm3d #194: Method for auto-discovery of cameras on network

theseankelly, 2020-01-31:
Create an API to scan the network and auto-discover all connected O3D/O3X cameras

theseankelly, 2020-02-13:
FYI we've just open sourced our python implementation here: https://github.com/ifm/python-ipv4-discovery/blob/master/discovery.py

theseankelly, 2020-05-22:
Updated issue description to include some design notes. @graugans any thoughts?

graugans, 2020-05-25:
We should consider to use the idea of an ``URI`` instead of the IPv4 address. There might be the case when host names are more useful than raw IP addresses. Related to the additional information. We have the Article number and the MAC address which comes handy. There is also the idea of the vendor and device type which should be used to distinguish the type of the device I would not recommend to use the article number to classify the type of camera (O3X vs. O3D).

closed 2022-12-13 by github-actions[bot]